lumapu/ahoy #435: keine Verbindung / Daten vom Microinverter

Loetking, 2022-11-17:
## Hardware
  * [ x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

### nRF24L01+ Module
* [ x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ x ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

## Software
* [ x ] AhoyDTU
* [ ] OpenDTU

### Build & Flash Method:
* [ ] Arduino
* [ x ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ x ] Windows
* [ ] Mac OS

Hallo, 

ich bin auf dieses kleine, aber feine Projekt gestoßen und musste dies auch gleich realisieren.
Als Inverter läuft ein TSOL-M800(DE), der auf 600W kastriert ist, jedoch von diesem erhalte ich keine Daten.
Die 12 stellige Seriennummer habe ich richtig eingegeben.
In der Ahoy Übersicht wird angezeigt : 

Uptime: 0 Days, 01:19:41
ESP-Time: 17.11.2022, 20:48:59
WiFi RSSI: -64 dBm
Statistics:

RX success: 0
RX fail: 0
RX no answer: 160
Frames received: 0
TX cnt: 955

Inverter #0: Boss (v0) is not available and is not producing

Verstehe ich  das richtig, dass "RX no answer" empfangene Pakete sind, mit der die DTU nichts anfangen kann ?
Gibt es eine Möglichkeit der Fehlersuche für den Laien ?
In der Beschreibung steht, das die TSOL-M800 Firmwareabhängig funktionieren, könnte dies das Problem sein ?

Würde mich über jegliche Info freuen.

lumapu, 2022-11-17:
Hallo ganz kurz: es ist schon dunkel

Die Wechselrichter antworten nur, wenn DC Spannung anliegt, dh. wenn es draußen hell ist. Um deine Frage zu beantworten: Nein "RX no answer" heißt, dass Ahoy nichts empfangen hat, also nicht mal ein Fragment empfangen hat. (Zur Info: eine Antwort vom Wechselrichter besteht aus mehreren Fragmenten)

Loetking, 2022-11-17:
@lumapu

das wäre ja zu einfach, 
danke für Deine Info, zumindest zerbreche ich mir heute Nacht nicht den Kopf darüber, ich werde morgen berichten.

RadivSyrbe, 2022-11-18:
Benutze auch ESP8266 und nRF24L01+ und hatte mit der vorgeschlagenen Belegung das gleiche Problem. Habe dann umgesteckt; CE auf D3, IRQ auf D4. Danach auf Weboberfläche unter Setting geändert, dann Reboot und es funktioniert. Vielleicht hilft mein Vorgehen.

Loetking, 2022-11-18:
Hallo,

Danke für die Unterstützung, es geht, und es lag wirkich daran, dass es gestern schon dunkel war ...
Ich suche irgendwie immer Fehler, die es nicht gibt ...

closed 2022-11-18 by lumapu