lumapu/ahoy #438: nach flash - AP vergibt keine IP Adresse

BW-TH, 2022-11-19:
## Hardware
  * [ x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0_5.28
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [x ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [x ] Windows
* [x ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)
mit der V 0.5.17 kann ich mich zumindest nach dem flashen über den AP vom 8266 verbinden. leider geht das umstellen auf mein eigenes Wlan nicht. das log sagt es wäre connected, ist aber am router nicht angemeldet. in der V 0.5.28 und 0.5.40 kann ich mich gar nicht auf den esp 8266 verbinden, da ich vom "AHOY-DTU" Lan keine IP Adresse bekomme. Habe gelesen das man die WLan settings hard coden kann, allerdings habe ich kene Ahnung wie ich das machen kann.

BW-TH, 2022-11-19:
mit der V 0.5.17 kann ich mich zumindest nach dem flashen über den AP vom 8266 verbinden. leider geht das umstellen auf mein eigenes Wlan nicht. das log sagt es wäre connected, ist aber am router nicht angemeldet. in der V 0.5.28 und 0.5.40 kann ich mich gar nicht auf den esp 8266 verbinden, da ich vom "AHOY-DTU" Lan keine IP Adresse bekomme. Habe gelesen das man die WLan settings hard coden kann, allerdings habe ich kene Ahnung wie ich das machen kann.

stefan123t, 2022-11-19:
Die IP Adresse des AccessPoints hat sich geändert probiere mal http://192.168.4.1/setup im AP-MODE vorher war es 192.168.1.1 was häufig Probleme mit dem Router verursacht hat.

BW-TH, 2022-11-19:
Komme ja nicht drauf weder auf 192.168.4.1 noch auf 192.168.1.1 da ich ja keine ip zugewiesen bekomme. Mein wlan hat dann fallback in Adresse

lumapu, 2022-11-19:
würdest du das Serielle Log während dem Start (im AP Mode) hier posten?

BW-TH, 2022-11-20:
Using 'COM3' as serial port.
Unexpected error: could not open port 'COM3': PermissionError(13, 'Zugriff verweigert', None, 5)
[12:18:14]........................................
[12:18:18]........................................
[12:18:19]W: your settings are not valid! check [IP]/setup
[12:18:19]I: 
[12:18:19]---------
[12:18:19]AP MODE
[12:18:19]SSID: AHOY-DTU
[12:18:19]PWD: esp_8266
[12:18:19]Active for: 60 seconds
[12:18:19]---------
[12:18:19]
[12:18:20]I: RF24 Amp Pwr: RF24_PA_I: LOW
[12:18:20]I: Radio Config:
[12:18:20]SPI Frequency		= 1 Mhz
[12:18:20]Channel			= 3 (~ 2403 MHz)
[12:18:20]Model			= nRF24L01+
[12:18:20]RF Data Rate		= 250 KBPS
[12:18:20]RF Power Amplifier	= PA_LOW
[12:18:20]RF Low Noise Amplifier	= Enabled
[12:18:20]CRC Length		= 16 bits
[12:18:20]Address Length		= 5 bytes
[12:18:20]Static Payload Length	= 32 bytes
[12:18:20]Auto Retry Delay	= 250 microseconds
[12:18:20]Auto Retry Attempts	= 0 maximum
[12:18:20]Packets lost on
[12:18:20]    current channel	= 0
[12:18:20]Retry attempts made for
[12:18:20]    last transmission	= 0
[12:18:20]Multicast		= Disabled
[12:18:20]Custom ACK Payload	= Disabled
[12:18:20]Dynamic Payloads	= Enabled
[12:18:20]Auto Acknowledgment	= Disabled
[12:18:20]Primary Mode		= RX
[12:18:20]TX address		= 0xe7e7e7e7e7
[12:18:20]pipe 0 (closed) bound	= 0xe7e7e7e7e7
[12:18:20]pipe 1 ( open ) bound	= 0x1301448101
[12:18:20]pipe 2 (closed) bound	= 0xc3
[12:18:20]pipe 3 (closed) bound	= 0xc4
[12:18:20]pipe 4 (closed) bound	= 0xc5
[12:18:20]pipe 5 (closed) bound	= 0xc6
[12:18:29]I: AP will be closed in 49 seconds

lumapu, 2022-11-20:
sieht für mich ganz normal aus, habe aktuell keine Idee warum es nicht klappen sollte

BW-TH, 2022-11-21:
Leider auch kein Erfolg, bestelle mal ein neues 8266 Board

stefan123t, 2022-11-21:
@lumapu können wir nach AP Mode bzw STA Mode connected nicht einfach immer die jeweilige IP Adresse auch auf der Serial Console ausgeben. Das würde dieses Problem leichter debugbar machen. Danke!

lumapu, 2022-11-22:
ja ist sinnvoll, baue ich ein

BW-TH, 2022-11-25:
so, heute neuen ESP 8266 bekommen und mit der 0.5.41 funktioniert es. jetzt nur noch morgen auf den sonnenaufgang warten und hoffentlich daten vom Wechselrichter empfangen

BW-TH, 2022-11-27:
jezt ist genug sonne am himmel doch leider habe ich folgendes Problem: 
wechselrichter kann nicht erkannt werden (inverter type can't be detected)
ist ein HM-1500
an was kann das liegen, anbei die Logs

Using '/dev/cu.SLAB_USBtoUART' as serial port.
Showing logs:
Serial port closed!
[08:18:39]r$[08:18:39]I:  .. done
[08:18:39]I: loadDefaults
[08:18:39]I: Settings valid: true
[08:18:39]I: RF24 Amp Pwr: RF24_PA_I: LOW
[08:18:39]I: Radio Config:
[08:18:39]SPI Frequency		= 1 Mhz
[08:18:39]Channel			= 3 (~ 2403 MHz)
[08:18:39]Model			= nRF24L01+
[08:18:39]RF Data Rate		= 250 KBPS
[08:18:39]RF Power Amplifier	= PA_LOW
[08:18:39]RF Low Noise Amplifier	= Enabled
[08:18:39]CRC Length		= 16 bits
[08:18:39]Address Length		= 5 bytes
[08:18:39]Static Payload Length	= 32 bytes
[08:18:39]Auto Retry Delay	= 250 microseconds
[08:18:39]Auto Retry Attempts	= 0 maximum
[08:18:39]Packets lost on
[08:18:39]    current channel	= 0
[08:18:39]Retry attempts made for
[08:18:39]    last transmission	= 15
[08:18:39]Multicast		= Disabled
[08:18:39]Custom ACK Payload	= Disabled
[08:18:39]Dynamic Payloads	= Enabled
[08:18:39]Auto Acknowledgment	= Disabled
[08:18:39]Primary Mode		= RX
[08:18:39]TX address		= 0xdeadbeef01
[08:18:39]pipe 0 (closed) bound	= 0xdeadbeef01
[08:18:39]pipe 1 ( open ) bound	= 0x4374348201
[08:18:39]pipe 2 (closed) bound	= 0xc3
[08:18:39]pipe 3 (closed) bound	= 0xc4
[08:18:39]pipe 4 (closed) bound	= 0xc5
[08:18:39]pipe 5 (closed) bound	= 0xc6
[08:18:39]I: added inverter 116181664431
[08:18:39]E: inverter type can't be detected!
[08:18:39]E: inverter type can't be detected!
[08:18:39]E: inverter type can't be detected!
[08:18:39]E: inverter type can't be detected!
[08:18:39]E: inverter type can't be detected!
[08:18:39]E: inverter type can't be detected!
[08:18:39]E: inverter type can't be detected!
[08:18:39]E: inverter type can't be detected!
[08:18:39]E: inverter type can't be detected!
[08:18:41]connect to network 'BWTH' ...
[08:18:43]...........
[08:18:44]I: 
[08:18:44]
[08:18:44]----------------------------------------
[08:18:44]I: Welcome to AHOY!
[08:18:44]I: 
[08:18:44]point your browser to http://192.168.0.153
[08:18:44]I: to configure your device
[08:18:44]I: ----------------------------------------
[08:18:44]
[08:18:44]I: MQTT connected
[08:19:46]I: [NTP]: 2022-11-27 07:19:40 UTC
[08:19:50]I: resetPayload: id: 0
[08:19:50]I: (#0) enqueuedCmd: 11
[08:19:50]I: (#0) enqueuedCmd: 1
[08:19:50]I: (#0) enqueuedCmd: 5
[08:19:50]I: (#0) sendTimePacket
[08:19:50]I: TX 27B Ch40 | 15 81 66 44 31 82 34 74 43 80 0B 00 63 83 0F 96 00 00 00 00 00 00 00 00 5E 34 9E 
[08:19:51]W: while retrieving data: last frame missing: Request Retransmit
[... 30 lines cut ...]
[08:20:29]I: (#0) sendTimePacket
[08:20:29]I: TX 27B Ch61 | 15 81 66 44 31 82 34 74 43 80 01 00 63 83 0F B4 00 00 00 00 00 00 00 00 F5 BE 97 
[08:20:50]I: resetPayload: id: 0
[08:20:50]I: (#0) sendTimePacket
[08:20:50]I: TX 27B Ch75 | 15 81 66 44 31 82 34 74 43 80 05 00 63 83 0F D2 00 00 00 00 00 00 00 00 93 38 15 
[08:20:51]W: while retrieving data: last frame missing: Request Retransmit
[08:20:51]I: (#0) sendTimePacket
[08:20:51]I: TX 27B Ch3 | 15 81 66 44 31 82 34 74 43 80 05 00 63 83 0F D2 00 00 00 00 00 00 00 00 93 38 15 
[08:20:53]W: while retrieving data: last frame missing: Request Retransmit
[08:20:53]I: (#0) sendTimePacket
[08:20:53]I: TX 27B Ch23 | 15 81 66 44 31 82 34 74 43 80 05 00 63 83 0F D2 00 00 00 00 00 00 00 00 93 38 15 
[08:20:55]W: while retrieving data: last frame missing: Request Retransmit
[08:20:55]I: (#0) sendTimePacket
[08:20:55]I: TX 27B Ch40 | 15 81 66 44 31 82 34 74 43 80 05 00 63 83 0F D2 00 00 00 00 00 00 00 00 93 38 15 
[08:20:57]W: while retrieving data: last frame missing: Request Retransmit
[08:20:57]I: (#0) sendTimePacket
[08:20:57]I: TX 27B Ch61 | 15 81 66 44 31 82 34 74 43 80 05 00 63 83 0F D2 00 00 00 00 00 00 00 00 93 38 15 
[08:20:59]W: while retrieving data: last frame missing: Request Retransmit
[08:20:59]I: (#0) sendTimePacket
[08:20:59]I: TX 27B Ch75 | 15 81 66 44 31 82 34 74 43 80 05 00 63 83 0F D2 00 00 00 00 00 00 00 00 93 38 15 
[08:21:20]I: resetPayload: id: 0
[08:21:20]I: (#0) enqueuedCmd: 11
[08:21:20]I: (#0) enqueuedCmd: 1
[08:21:20]I: (#0) enqueuedCmd: 5
[08:21:20]I: (#0) sendTimePacket

lumapu, 2022-11-27:
das passt, in log findet man auch diese Zeile: [08:18:39]I: added inverter 116181664431

Die anderen Inverter kann er nicht finden, da sie null sind, hier müssen wir die Ausgabe noch verbessern / verschönern

stefan123t, 2022-11-27:
Hallo @BW-TH was für ein NRF Modul hast Du denn (externe Antenne oder Leiterbahn) ?
Dein aktueller PA_Level ist auf LOW eingestellt, das reicht i.d.R. für ein Modul mit externer Antenne und PA+LAN.
Für ein Modul ohne Antenne nur mit Leiterbahn braucht man m.E. mindesten PA HIGH oder sogar MAX 
und einen 47-100uF Elko an Pin 1 & 2 (GND & +3,3V) des NRF24 Moduls gelötet um die Spannung beim Senden zu stabilisieren.
Er versucht zwar ständig Deinen Wechselrichter zu erreichen (TX Meldungen), empfängt aber umgekehrt keine Antworten (RX Meldungen) vom Wechselrichter.
Kannst Du ggf. unter Settings > System Config > [x] Serial Debug aktivieren und nochmal bei Tage im Serial Log nach schauen ?

BW-TH, 2022-11-28:
Habe mit externer Antenne, Abstand zum wechselrichter max 5 m. Auf dem Nrf-Modul sind einige pins nicht eindeutig lesbar. Kann es sein das ein Kabel verkehrt ist?

BW-TH, 2022-11-29:
habe das Amplifier Power Level auf max gestellt, leider noch keine daten vom wechselrichter. anbei das log mit serial debug und print inverter data

08:39:44 I: enqueued cmd failed/timeout
08:39:44 I: (#0) I: no Payload received! (retransmits: 5)
08:39:44 I: resetPayload: id: 0
08:39:44 I: (#0) Requesting Inv SN 116181664431
08:39:44 I: (#0) sendTimePacket
08:39:44 I: TX 27B Ch40 | 15 81 66 44 31 82 34 74 43 80 05 00 63 85 c5 50 00 00 00 00 00 00 00 00 a1 7b 2a 
08:39:45 W: while retrieving data: last frame missing: Request Retransmit
08:39:45 I: (#0) sendTimePacket
08:39:45 I: TX 27B Ch61 | 15 81 66 44 31 82 34 74 43 80 05 00 63 85 c5 50 00 00 00 00 00 00 00 00 a1 7b 2a 
08:39:47 W: while retrieving data: last frame missing: Request Retransmit
08:39:47 I: (#0) sendTimePacket
08:39:47 I: TX 27B Ch75 | 15 81 66 44 31 82 34 74 43 80 05 00 63 85 c5 50 00 00 00 00 00 00 00 00 a1 7b 2a 
08:39:49 W: while retrieving data: last frame missing: Request Retransmit
08:39:49 I: (#0) sendTimePacket
08:39:49 I: TX 27B Ch3 | 15 81 66 44 31 82 34 74 43 80 05 00 63 85 c5 50 00 00 00 00 00 00 00 00 a1 7b 2a 
08:39:51 W: while retrieving data: last frame missing: Request Retransmit
08:39:51 I: (#0) sendTimePacket
08:39:51 I: TX 27B Ch23 | 15 81 66 44 31 82 34 74 43 80 05 00 63 85 c5 50 00 00 00 00 00 00 00 00 a1 7b 2a 
08:39:53 W: while retrieving data: last frame missing: Request Retransmit
08:39:53 I: (#0) sendTimePacket
08:39:53 I: TX 27B Ch40 | 15 81 66 44 31 82 34 74 43 80 05 00 63 85 c5 50 00 00 00 00 00 00 00 00 a1 7b 2a

stefan123t, 2022-11-30:
@BW-TH vielleicht füllst Du mal das Connection Diagram im ersten Post ganz oben einmal aus und hängst ein paar Bilder von Deinem Aufbau an. Der IRQ Pin kann nicht Software-seitig geprüft werden, hier kommt es manchmal (z.B. bei Wemos D1 mini Modulen von Makershop) zu Problemen an GPIO0. Man kann ihn aber mit den anderen Pins z.B. GPIO2 tauschen und die Konfiguration unter Settings > System Config > Wemos Pinout anpassen, speichern und neu starten. Haken bei [x] restart after config save setzen.

BW-TH, 2022-11-30:
[image]

BW-TH, 2022-11-30:
[image]

BW-TH, 2022-11-30:
[image]

BW-TH, 2022-11-30:
[image]

BW-TH, 2022-11-30:
[image]

closed 2022-12-30 by lumapu

stefan123t, 2023-01-06:
@BW-TH funktioniert denn Dein Setup jetzt ?

Eventuell kannst Du die Spannungs / Stromversorgung verbessern indem Du einen Elko (richtige Polung +/- beachten) mit ca. 10-100uF an die Pins 1&2 (GND & +3.3V) des NRF24 Moduls lötest. Andere haben auch das Netzteil oder das USB Kabel gegen ein hochwertiges / neues getauscht. Bei den Wemos D1 mini v3 half auch vielfach die Anschlüsse D3 / D4 für IRQ und CE sowohl per Kabel als auch in den Settings zu tauschen um eine bessere und verlässliche Verbindung zu bekommen.

BW-TH, 2023-01-06:
leider nein, habe jetzt noch 2 anderer boards versucht (Wemos mini pro, Node mcu v3)

 

leider immer der gleiche fehler:

I: TX 27B Ch23 | 15 81 66 44 31 81 70 45 96 80 0B 00 63 B8 07 E3 00 00 00 00 00 00 00 00 02 35 26 W: while retrieving data: last frame missing: Request Retransmit

 

 
 

Gesendet: Freitag, 06. Januar 2023 um 01:44 Uhr

stefan123t, 2023-01-07:
Elko anlöten ?