lumapu/ahoy #441: ESP32 v0.5.41 - falsche uptime offline Meldungen über MQTT

knickohr, 2022-11-20:
### Platform

ESP32

### Model name

DevKit C von AZ

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO16 IRQ |

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.41

### Github Hash

https://github.com/lumapu/ahoy/commits/dec333f

### Build & Flash Method

ESP Tools (flash)

### Desktop

Mac OS

### Setup

{"menu":{"name":["Live","Serial Console","Settings","-","REST API","-","Update","System"],"link":["/live","/serial","/setup",null,"/api",null,"/update","/system"],"trgt":[null,null,null,null,"_blank"]},"system":{"ssid":"Odyssee2001","device_name":"AHOY-DTU","version":"0.5.39","build":"69855b1","ts_uptime":151594,"ts_now":1668933681,"ts_sunrise":1668925650,"ts_sunset":1668959228,"ts_sun_upd":1668898800,"wifi_rssi":-64,"pwd_set":false,"hostname":"AHOY-DTU","sdk_version":"v4.4.2","cpu_freq":240,"heap_total":316784,"heap_used":122824,"chip_revision":1,"chip_model":"ESP32-D0WDQ6","chip_cores":2,"sketch_total":1310720,"sketch_used":974496,"esp_type":"ESP32"},"inverter":{"inverter":[{"id":0,"name":"PV1","serial":"112182841769","channels":1,"version":"10014","ch_max_power":[360],"ch_name":["Links"]},{"id":1,"name":"PV2","serial":"112182844440","channels":1,"version":"10014","ch_max_power":[360],"ch_name":["Rechts"]}],"interval":"30","retries":"5","max_num_inverters":10},"mqtt":{"broker":"192.168.100.80","port":"1883","user":"mqttuser","pwd":"{PWD}","topic":"AhoyDTU"},"ntp":{"addr":"192.168.100.1","port":"123"},"sun":{"lat":"48.44000","lon":"9.89000","disnightcom":false},"pinout":{"cs":5,"ce":4,"irq":16,"led0":25,"led1":26},"radio":{"power_level":1},"serial":{"interval":5,"show_live_data":false,"debug":false},"static_ip":{"ip":"192.168.100.79","mask":"255.255.255.0","dns1":"192.168.100.1","dns2":"0.0.0.0","gateway":"192.12.100.1"}}

### Debug Serial Log output

```shell
version: 0.5.41
wifi_rssi: -64
mac: 24:62:AB:F3:9F:E8
sdk: v4.4.2
cpu_freq: 240
heap_free: 191252
sketch_used: 952
heap_total: 316560
chip_revision: 1
chip_model: ESP32-D0WDQ6
chip_cores: 2
esp_type: ESP32
```

### Error description

Über MQTT werden immer wieder uptime offline gesendet obwohl die WRs alle online sind, available and producing.

ist sei der DEV 40++

[image]

lumapu, 2022-11-20:
mit welchem Tool trust du die MQTT Nachrichten an?

knickohr, 2022-11-20:
Na, alte Linux-Depp-Methode:

ssh-Konsole und grappa
[image]

lumapu, 2022-11-20:
Kann es nicht navollziehen.

mit ESP8266 und v0.5.41
[image]

mit ESP32 und v0.5.41
[image]

das `offline` kommt von hier:
https://github.com/lumapu/ahoy/blob/dec333fdf904a857fdf1873167803388a81af34e/src/publisher/pubMqtt.h#L200

knickohr, 2022-11-20:
Es kommt komischerweise auch nur beim ESP32 🤔

Du mußt länger warten, manchmal dauert es Minuten, dann aber geballte Ladung 😲

womoak75, 2022-11-21:
Wenn das die 'last will message' ist, dann hat der Server einen Verbindungsabbruch erkannt,
d.h. der Client hat nicht innerhalb des erwarteten Zeitraums seinen 'Ping' gesendet.
Der Server schliesst dann die Verbindung und published die 'last will message'.

Aber auf der Clientseite sollte man dann ein 'connection lost' bzw. reconnect sehen.

knickohr, 2022-11-21:
Das Problem tritt nur beim ESP32 zu Tage.

ich habe gestern wieder den 8266 in Betrieb genommen, da kommen die Meldungen nicht. Allgemein läuft der 8266 besser, schneller, flüssiger und ohne hängende Webseite und gelegentliche Reboot. Das Problem scheint wohl eher generell an der Hardware mit dem Programm zu sein. Da es in früheren Versionen (<30) nicht auftrat, vermute ich ein Softwareproblem, speziell beim ESP32.

stefan123t, 2022-11-21:
ME handelt es sich dabei um eine Offline Meldung dass aktuelle keine Verbindung zum MQTT broker besteht nachdem ich mit den von @lumapu genannten Code bzw die folgenden Zeilen angesehen habe.

lumapu, 2022-11-21:
ich glaube eher eine "was offline" Meldung beim Versuch des reconnect.

Wir sollten hier testweise die Lib https://github.com/bertmelis/espMqttClient einbauen, die verwendet @tbnobody auch.

Wenn man sich das Readme durchliest, geht es auch um den TCP Stack, der wohl in anderen Libs veraltet ist.

lumapu, 2022-11-21:
@knickohr ich habe testweise auf eine andere Lib umgestellt, kannst gerne testen ob der ESP32 jetzt stabiler ist:

https://github.com/lumapu/ahoy/tree/feature/espMqttClient

knickohr, 2022-11-22:
Ähhh, jetzt haste mich erwischt 😵 Wo bekomme ich das Binary ohne selbst zu compilieren ? 🤔

lumapu, 2022-11-22:
@knickohr ich hoffe der dev Link auf https://ahoydtu.de funktioniert wieder.

knickohr, 2022-11-22:
Yupp, funzt wieder 😉

knickohr, 2022-11-22:
Tja, ernüchternde Feststellung 😞

es ist keinen Deut besser, es ist sogar so schlimm das er mir den Broker blockiert 😵 Nichts kommt mehr an von anderen Clients - inakzeptabel !

lumapu, 2022-11-22:
Schade

lumapu, 2022-11-22:
Schade

knickohr, 2022-11-23:
Stecke da nicht zu viel Energie rein. Solange das mit dem 8266 noch keine Engpässe oder Limits gibt, macht mir das wenig aus.

knickohr, 2022-11-24:
Die update offline Meldungen kommen ja jetzt nicht mehr an der 43, oder ich habe sie bis jetzt übersehen. Statt dessen kommt jetzt Status online/offline. Das ist schon Buckel besser als das unter die uptime (Uhrzeit) zu knallen 👍

Gerri1, 2022-11-28:
Ich habe mir heute die dev_v0.5.44 auf den ESP8266 zum Testen geladen, hier ist uptime Meldung über MQTT (ioBroker) ständig auf 0.
Auch auf der Ahoy Startseite bleibt die  Anzeige "Every 10 seconds the values are updated" so stehen, egal was man eingestellt hat!

lumapu, 2022-11-28:
@Gerri1 ja auf der Index Seite habe ich das Intervall auf 10s standartmäßig eingestellt, da ich nicht nur alle 30s ein Lebenszeichen haben will

Die uptime sollte schon größer werden, ich nutze auch ioBroker, aktuell habe ich 6000s

Gerri1, 2022-11-28:
[image]
Ich habe den ganzen Objektbaum schon gelöscht und dann neu Anlegen lassen, die 0 bleibt!

Gerri1, 2022-11-28:
... nein, eine NodeMCU v3 ESP8266, auf einem Wemos habe ich die v0.5.28 drauf.

knickohr, 2022-11-29:
Moment mal, 28 ?  War das nicht so das die Nachts immer offline ausgegeben haben ?  Mach mal. Eine 41+ drauf. Die 41 ist als stable gekennzeichnet.

Gerri1, 2022-11-29:
@knickohr
Die 41er hatte ich auch schon am Laufen, da diese aber über den Tag hinweg viele Neustarts hatte, habe ich die 28 wieder aufgespielt.

knickohr, 2022-11-30:
Ich bin mit den ESP32, speziell mit MQTT auch nicht wirklich glücklich geworden.  Zu viele Hänger und Reboots, die Webseite lahmte und es kamen teilweise verwirrende Meldungen über MQTT.

Die 44 ist nach der 28 bis jetzt noch am stabilsten.

Gerri1, 2022-12-01:
[image]

Kann es evtl. daran liegen, das die uptime Meldung über MQTT (ioBroker) ständig auf 0 ist?

lumapu, 2022-12-01:
das würde aber heißen, dass er ständig neu connected, aber der Hinweis ist gut, wir sollten an der Stelle nicht pauschal 0 senden

Gerri1, 2022-12-01:
So wie das aussieht, connected er wirklich ständig neu!

[image]
[image]
[image]

Gerri1, 2022-12-05:
Bei der 0.5.47 bekomme ich jetzt im ioBroker unter mqtt "connected" angezeit und bleibt auch so! 
Auf der Index- Seite und in der Konsole ist es unverändert.

[image]

knickohr, 2022-12-06:
Korrekt !

Siehe #468 😉

Gerri1, 2022-12-09:
Mit der 0.5.50 bekomme ich jetzt wieder die uptime in ioBroker angezeigt. 👍👍😁
Auf der Ahoy-DTU index- Seite und Konsole werden nacheinander immernoch die nachfolgenden Meldungen angezeigt!

[image]
[image]

closed 2022-12-17 by knickohr