lumapu/ahoy #444: NRF24 module can't be reached

MartinL96, 2022-11-21:
## Hardware
  * [ ] ESP8266 D1 Mini

### nRF24L01+ Antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

## Software
* [ ] AhoyDTU

### Version / Git SHA: 
[ahoy_v0.5.41]

### Build & Flash Method:
* [ ] Platform IO

### Desktop OS:
* [ ] Windows

[image]
____________________________________
Hi! 

Erstmal vielen Dank für dieses tolle Projekt :)

Leider bin ich noch ein Anfänger in Sachen Mikrocontroller und kann den Fehler nicht finden, weshalb AHOY die NRF24 Antenne nicht erkennt. 

Versucht habe ich es mit und ohne Kondensator und die Verdrahtung mehrmals auf Richtigkeit geprüft. Auch das NRF Modul selbst habe ich gegen ein anderes Ausgetauscht, aber auch hier das selbe Ergebnis. Kann mir hier jemand weiterhelfen die Ursache des Problems zu finden?

lumapu, 2022-11-21:
sind die Kabel verlötet? Das muss mit der Hardware zusammenhängen. Für diesen Test wird die SPI Verbindung benötigt.
Wurde von der richtigen Seite auf die Module geschaut?
Ein Bild des Aufbaus könnte evtl auch weiterhelfen.

MartinL96, 2022-11-21:
Hallo @stefan123t und @lumapu vielen Dank für euere raschen Antworten!

@stefan123t die Pins zu tauschen und in der Konfiguration zu ändern war scheinbar die Lösung. Die Fehlermeldung ist weg. 
Heute ist die Sonne leider auch schon weg, aber dann bin ich gespannt ob´s morgen dann klappt mit dem Daten auslesen.

stefan123t, 2022-11-21:
@MartinL96 Wir brauchen dennoch ein Foto von Deinem ESP8266 Board für unsere Beweissammlung :) wie gesagt am Besten mit Händler Link.

MartinL96, 2022-11-23:
Um das Projekt zu unterstützen möchte ich gerne auch die 3D-Druck Daten meines Gehäuses mit euch teilen :)
https://www.printables.com/de/model/323500-ahoy-dtu-housing

lumapu, 2022-11-24:
@MartinL96 du kannst den Link auch gerne per pull request in unsere Dokumentation einfließen lassen

closed 2022-12-08 by lumapu

lumapu, 2023-01-05:
Verlöten würde auf jeden Fall eine große Fehlerquelle ausschließen.
Komisch finde ich allerdings, das 3 Versuche nicht klappen. Vom NRF müssen alle 8 Pins angeschlossen sein.
Wie lang sind die Drähte und gibt es eine Möglichkeit die verbundenen durchzuklingeln?

frankhoerper, 2023-01-06:
Danke für die Hinweise. Die Bauteile sind alle neu, aber ich werde bei Gelegenheit nochmal eine Version verlöten. Habe jetzt einen alten nodemcu8266 an den verkabelten rf24 gehängt und damit lief alles direkt wie gewünscht. Somit weiss ich auf jeden Fall, dass das rf24 Modul nicht defekt ist.

stefan123t, 2023-01-07:
@frankhoerper dann liegt es also nicht am NRF24 Modul das ist schon mal gut und erwartet. Wenn Du noch Zeit und Lust hast kannst Du gerne mal mit verschiedenen Belegungen con D1..D4 und IRQ bzw CE probieren. M.W. haben die Wemos D1 Module da irgendeine spezielle Reset/Flash/Boot Schaltung die das Problem verursacht.
Wäre schön wenn wir mal so einen Wemos D1 hätten der das Problem zeigt um es ggf. mit dem Oszi / Logikanalyzer auszumessen.

frankhoerper, 2023-01-13:
@stefan123t , werde das in den nächsten Tagen nochmal mit einem Breadboard durchtesten. Die D1-D4 Variante hatte ich aber auch bereits getestet , ebenso  den IRQ/CE Tausch. Zwischenzeitlich habe ich den gleichen Effekt auch mit openDTU und zwei verschiedenen ESP32 Modellen. Die nodecmu Variante , die in der Doku angegeben ist funktioniert, eine Dev Kit Variante mit ESP32 Chipset/38 Pins nicht. Auf irgendeinem PIN funkt scheinbar irgendein Signal je nach Modell nicht wie erforderlich. Auf jeden Fall habe ich gelernt , dass bei den Modellen bzgl Kompatibiltät eine ganz schöne Varianz besteht. 
Die ESP8266nodecmu / ahoyDTU Konfig läuft seit einigen Tagen problemlos. Vielleicht sollte das als Empfehlung mit in die Doku, falls es mit den D1 Modellen nicht auf Anhieb funktioniert und man nicht x Modelle durchtesten möchte.