lumapu/ahoy #446: Ständige Reboots beim ESP8266

van-shorty, 2022-11-22:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: NodeMCU Lua Lolin V3
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.41

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [x] Platform IO

### Desktop OS:
* [ ] Linux
* [x] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [x] Setup settings (use our templates ... to be added)

_{"menu":{"name":["Live","Serial Console","Settings","-","REST API","-","Update","System"],"link":["/live","/serial","/setup",null,"/api",null,"/update","/system"],"trgt":[null,null,null,null,"_blank"]},"system":{"ssid":"Shortys-Wlan","device_name":"AHOY-DTU","version":"0.5.41","build":"dec333f","ts_uptime":1184,"ts_now":1669117660,"ts_sunrise":0,"ts_sunset":0,"ts_sun_upd":0,"wifi_rssi":-79,"mac":"30:83:98:B5:90:DA","hostname":"AHOY-DTU","pwd_set":false,"sdk":"2.2.2-dev(38a443e)","cpu_freq":80,"heap_free":8184,"sketch_total":2662400,"sketch_used":468,"core_version":"3.0.2","flash_size":4096,"heap_frag":31,"max_free_blk":5008,"reboot_reason":"Software/System restart","esp_type":"ESP8266"},"inverter":{"inverter":[{"id":0,"name":"Dach","serial":"114181013059","channels":2,"version":"10010","ch_max_power":[0,0],"ch_name":["",""]}],"interval":"10","retries":"5","max_num_inverters":10},"mqtt":{"broker":"192.168.11.99","port":"1883","user":"","pwd":"","topic":"inverter"},"ntp":{"addr":"pool.ntp.org","port":"123"},"sun":{"lat":"","lon":"","disnightcom":false},"pinout":{"cs":15,"ce":2,"irq":0,"led0":2,"led1":255},"radio":{"power_level":3},"serial":{"interval":5,"show_live_data":false,"debug":false},"static_ip":{}}_

Hallo Zusammen,

ich habe die DTU jetzt einige Tage laufen und funktioniert vom Auslesen der Werte sehr gut. Allerdings habe ich das Problem, dass der ESP8266 ständig rebootet. Dazu habe ich ein Chart angehängt welches die "Uptime" der DTU per MQTT aufgezeichnet hat. Spätestens nach ca. 7000s startet der ESP neu und dann beginngt die uptime von vorn.

Kennt jemand das Problem?

Ich habe die "221119_ahoy_0.5.41_esp8266_dec333f.bin" aus dem aktuellen release per PIO geflasht.

Grüße Jens

[image]

DukeWolle, 2022-11-22:
Servus... Also an der Version liegt es nicht. Ich könnte mir vorstellen das die Spannung evtl nicht stabil ist. Wie hast du die Sendeleistung vom nRF eingestellt? Da kommt es gerne mal zu Schwankungen. Ich empfehle erst mal einen 100uf Elko direkt am Eingang des nRF ... 

Gruß Wolfi

van-shorty, 2022-11-22:
Sendeleistung steht auf High, da die DTU relativ weit weg ist. daran hatte ich aber auch noch nicht gedacht, einen Cap hab ich nicht dran, wär auch noch ne möglichkeit...ich berichte

stefan123t, 2022-11-22:
@van-shorty welches NRF24L01+ Modul hast Du denn ?
Vermutlich eines mit Brotbrett Antenne und keines mit einer echten externen Antenne. Die brauchen mehr Sendeleistung (HIGH oder MAX) um überhaupt den WR zu erreichen. Bei MAX wird der Spannungswandler auf dem ESP Board i.d.R. zu sehr unter Last gesetzt so daß die Spannung einbricht und es zu erratischem Verhalten kommt. Eine Option ist es zwar einen 47-100uF Elko an Pin 1 & 2 GND/+3,3V des NRF zu löten und/oder dickere Kabel/Litzen für die Versorgungsspannung zu verwenden. Aber besser sind die Module mit externer Antenne und PA+LNA Verstärker, die funktionieren i.d.R. bereits mit LOW oder sogar MIN Einstellungen und brauchen weniger Strom.
Hast Du die Option ein stärkeres Schaltnetzteil an den ESP8266 anzuhängen, bzw. einen separaten 3,3V Spannungsregler an parallel nur für den NRF24 zu betreiben ?

turbolift, 2022-11-22:
Ich hatte das selbe Problem (Level=LOW und Stabantenne). Ich habe die Version von ahoy sehr lange verwendet und viele Updates mitgemacht. Die Reboots sind über die Zeit in immer häufiger aufgetreten. Waren es anfänglich noch 2-3 Tage ohne Reboot, war es mit den letzten Versionen fast nie über 24 Stunden. Ich bin mittlerweile auf einem ESP32/OpenDTU und dort läuft es sehr stabil. Einen Elko habe ich aber in beiden Fällen nie dazwischen gehangen.

van-shorty, 2022-11-22:
> @van-shorty welches NRF24L01+ Modul hast Du denn ? Vermutlich eines mit Brotbrett Antenne und keines mit einer echten externen Antenne. Die brauchen mehr Sendeleistung (HIGH oder MAX) um überhaupt [...]

ich habe eins mit leiterplatten antenne, habe jetzt auf LOW gestellt, Kommunikation ist bis dato noch vorhanden. Schade das man im MQTT nicht den Heap ausgegeben bekommt, könnte ja sein das der voll läuft.

Debug "Logfiles" gibt es nach einem neustart wohl auch nicht oder??

stefan123t, 2022-11-22:
Heap wird mW ausgegeben und Logfiles sind im USB Serial Log, wo schaust Du denn ?

van-shorty, 2022-11-22:
also im mqtt sehe ich nur uptime und wifi_rssi, ich hab die DTU nicht dauerhaft an einem PC hängen.

gerade hatte ich einen reboot, wo mal ein anderer Grund angeben war: "Hardware Watchdog"

[image]

lumapu, 2022-11-22:
@stefan123t heap haben wir nicht mehr drin, da es nichts aufschlussreiches gebracht hat. Kann es gerne wieder per MQTT übertragen, denke dass Problem ist aber ab anderer Stelle

lumapu, 2022-11-22:
@stefan123t heap haben wir nicht mehr drin, da es nichts aufschlussreiches gebracht hat. Kann es gerne wieder per MQTT übertragen, denke dass Problem ist aber ab anderer Stelle

siehe auch issue #83

van-shorty, 2022-11-22:
also auch auf LOW rebootet er fröhlich vor sich hin, ich teste jetzt mal mit einem CAP. Kenne es auch aus Vorherigen Projekten, das der NRF gerne Probleme macht.

Zusätzlich finde ich es immer besser möglichst viele daten verfügbar zu machen, macht eine fehler suche oder beobachtung des ganzen, wenn es nicht am PC hängt deutlich angenehmer. :D

van-shorty, 2022-11-23:
Guten Morgen, habe gestern noch einen 100µF Kondensator an das NRF gelötetet, aber die Reboots halten weiter an. Letzter Reboot Grund: Software/System restart

hier wieder der chart:

[image]

Mir fällt dann jetzt nix mehr ein....

stefan123t, 2022-11-24:
@lumapu vielleicht wird es auch mit der getauschten MQTT Bibliothek besser, die bisherige PubSubClient Lib ist m.W. auch schon für ihre Probleme bei der Nutzung von Strings aufgefallen.

van-shorty, 2022-11-24:
d.h. im nächsten Release ist eine neue MQTT Bibliothek implementiert???

stefan123t, 2022-11-24:
@van-shorty nein noch nicht, lumapu hat einen extra Branch aufgemacht um die MQTT Bibliothek testweise auszutauschen.

lumapu, 2022-11-24:
@van-shorty @stefan123t Ja wenn die Tests mit der aktuellen development Version gut laufen, dann wird die neue Lib ins release kommen.
Positiver Nebeneffekt ist auch die TLS Unterstützung der neuen Lib.

schliepi, 2022-12-08:
Meine Erfahrung: Hardware wie eingangs genannt. Zusätzlich 100µF Kondensator.
Anschluss ESP an Apple 10W USB-Netzteil: 50 Tage lang Uptime schwankend, maximal 2 Tage.
Anschluss ESP an MacMini M1 USB-Ausgang: Uptime aktuell 26 Tage.

Ich würde schätzen, dass das eindeutig ein Problem der Stromversorgung war.

lumapu, 2022-12-08:
danke für den Bericht.
Ich würde vermuten, dass noch größere Kapazitäten am ESP auch helfen würden. evtl sogar auf der 5V Seite

closed 2022-12-30 by lumapu