lumapu/ahoy #448: HW/FW Version der Wechselrichter

Waldo56, 2022-11-22:
[image]

Was bedeutet die Info V10010 in Klammern nach dem Namen des WR? Ist das eine Versionsnummer? HW/FW? Beide Versionen wären natürlich interessant.

stefan123t, 2022-11-22:
@Waldo56 schau mal hier https://github.com/lumapu/ahoy/wiki/Protocol#inverterdevinform_simple--0x00 und hier https://github.com/lumapu/ahoy/wiki/Protocol#inverterdevinform_all--0x01 
Danke!

DukeWolle, 2022-11-22:
@Waldo56 
Ja das ist deine FWVersion. Kannst auch anschauen indem du folgende Seite bei dir aufrufst. 

http://deine_ip/api/record/info

Gruß Wolfi

Waldo56, 2022-11-22:
Ok. Danke Euch.

closed 2022-11-24 by stefan123t