lumapu/ahoy #449: kein Verbindung nach update auf 5.41

DennisXK, 2022-11-22:
Hi, nach dem update von 5.28 auf 5.41 bekomme ich keine Verbindung mehr auf die Weboberfläche. Habe nun die 5.41 via USB geflscht und mein WIFI und eine static IP eingestellt. Leider bekomme ich ebnfalls keine Verbindung hin! Hab das jetzt mehrfach wiederholt mit dem selben Ergebniss

ein auszug aus der esphomeflasher Console

Showing logs:
[20:08:17]rl[20:08:17]I:  .. done
[20:08:17]I: loadDefaults
[20:08:17]I: Settings valid: true
[20:08:19]connect to network 'gt3' ...
[20:08:21]...........
[20:08:22]I: 
[20:08:22]
[20:08:22]----------------------------------------
[20:08:22]I: Welcome to AHOY!
[20:08:22]I: 
[20:08:22]point your browser to http://192.168.178.102
[20:08:22]I: to configure your device
[20:08:22]I: ----------------------------------------
[20:08:22]
[20:08:22]I: RF24 Amp Pwr: RF24_PA_I: LOW
[20:08:22]I: Radio Config:
[20:08:22]SPI Frequency		= 1 Mhz
[20:08:22]Channel			= 255 (~ 2655 MHz)
[20:08:22]Model			= nRF24L01+
[20:08:22]RF Data Rate		= 1 MBPS
[20:08:22]RF Power Amplifier	= PA_MAX
[20:08:22]RF Low Noise Amplifier	= Enabled
[20:08:22]CRC Length		= 16 bits
[20:08:22]Address Length		= 5 bytes
[20:08:22]Static Payload Length	= 32 bytes
[20:08:22]Auto Retry Delay	= 4000 microseconds
[20:08:22]Auto Retry Attempts	= 15 maximum
[20:08:22]Packets lost on
[20:08:22]    current channel	= 15
[20:08:22]Retry attempts made for
[20:08:22]    last transmission	= 15
[20:08:22]Multicast		= Allowed
[20:08:22]Custom ACK Payload	= Enabled
[20:08:22]Dynamic Payloads	= Enabled
[20:08:22]Auto Acknowledgment	= 0b111111
[20:08:22]Primary Mode		= RX
[20:08:22]TX address		= 0xffffffffff
[20:08:22]pipe 0 ( open ) bound	= 0xffffffffff
[20:08:22]pipe 1 ( open ) bound	= 0xffffffffff
[20:08:22]pipe 2 ( open ) bound	= 0xff
[20:08:22]pipe 3 ( open ) bound	= 0xff
[20:08:22]pipe 4 ( open ) bound	= 0xff
[20:08:22]pipe 5 ( open ) bound	= 0xff
[20:08:22]W: WARNING! your NRF24 module can't be reached, check the wiring
[20:08:22]I: added inverter xxxxxxxxxxxx
[20:08:22]E: inverter type can't be detected!

barnie1966, 2022-11-22:
Das selbe Phänomen bei mir, kein Zugriff mehr auf die Weboberfläche nach dem Update!

DukeWolle, 2022-11-22:
@DennisXK @barnie1966 Die Frage an beide  wie habt ihr geflasht und was war vorher drauf? 

Gruß Wolfi

barnie1966, 2022-11-22:
Also ich hatte die 5.17 drauf und bin über die Weboberfläche auf Firmwareupdate gegangen. Anschließend kam reboot, aber ich komme eben nicht mehr auf die Oberfläche.

DukeWolle, 2022-11-22:
Das erfordert eine komplette neue Installation... bin mir nicht sicher, aber müsste eine Install.html dabei sein.

DukeWolle, 2022-11-22:
Ansonsten auf 
https://ahoydtu.de/getting_started/

Glaub da sollte auch schon die neue unterstützt sein.

DukeWolle, 2022-11-22:
Hab eben noch die Info von Stefan bekommen. 

----
Die install.html ist m.W. nur im Development Build drin.
Damit der google-chrome die **ESP Web Tools** mit dem lokalen Bin Files Flashen kann,
muß man den **Chrome Tasktray-Daemon killen**/beenden 
und `google-chrome --allow-file-access-from-files` mit der Option starten sonst klemmts.

DennisXK, 2022-11-22:
Das einrichten ist kein Problem. Wie oben schon geschrieben habe ich die 5.41 mit usb geflashed. Anschließend verbinde ich mich auf die Ahoy-Dtu „192.168.4.1“. Stelle mein WiFi ein und vergebe eine static ip, speicher und trenne die Stromversorgung (neustart). In meiner Fritzbox wird mir die Ahoy-DTU angezeigt, aber es lässt sich keine Verbindung herstellen.

SeBo-86, 2022-11-23:
Ich hatte genau das selbe Problem beim Wechsel von der 5.28 auf die 5.41.
Nach mehrmaligen erfolglosen Versuchen, habe ich im ersten Setup dann mal nur SSID + WLAN Passwort sowie Gateway ausgefüllt, gespeichert sowie reboot durchgeführt was zum Erfolg führte.
Die restlichen Parameter, Inverter MQTT usw... habe ich dann erst später ergänzt.
Zuvor erhielt der ESP8266 laut Serial Console immer als IP 255.0.0.0

DennisXK, 2022-11-23:
@SeBo-86 hattest du zuvor auch eine feste ip vergeben? und er hat dir trotzdem die 255.0.0.0 vergeben?

SeBo-86, 2022-11-23:
Nein, zuvor in der 5.28 immer nur via DHCP / ohne feste IP.
Ich hatte auch zwischenzeitlich Versucht, in der config_overide die WLAN Anmeldedaten zu hinterlegen was jedoch auch zu keinem Erfolg geführt hat.

DennisXK, 2022-11-23:
Ok, ja ich hab auch wieder die 5.28 draufgemacht. muss noch nachgebessert werden🤷‍♂️

stefan123t, 2022-11-24:
@SeBo-86 warum füllst Du dann die Informationen für die Vergabe einer festen IP Adresse überhaupt aus und dann auch noch nur die Gateway Adresse ? Entweder alle fünf Felder ausfüllen oder komplett leer lassen für DHCP wie bisher.

@DennisXK hast Du es mal mit http://ipadresse/setup versucht ? Vorsicht ab v0.5.41 muß man zwingend ein Passwort ganz oben unter AHOY-DTU vergeben, damit man sich noch mit der Web Oberfläche verbinden kann. Wenn man das vergißt wars das evtl und man muß wieder flashen bzw. auf Factory Defaults resetten.

stefan123t, 2022-11-24:
@barnie1966 eventuell auch mal ein Erase Flash machen, ich glaube in den ESP Web Tools unter https://ahoydtu.de/getting_started wird das auch angeboten. VOn v0.5.17 auf 0.5.41 ist ein ganz schöner Sprung und zwischendrin wurde die Konfiguration auf JSON umgestellt, das kann ganz schön schiefgehen wenn man den ESP vorher nicht sauber macht.

barnie1966, 2022-11-24:
@stefan123t vielen Dank für den Tipp. Ich werde das am Wochenende nochmal testen.

SeBo-86, 2022-11-24:
@stefan123t das war eigentlich ehr ein verzweifelter Versuch, nachdem ich bereits 2 Tage versucht habe die 5.41 zum Laufen zu bekommen. 
Ein Erase Flash habe ich vorher jeweils durchgeführt & den Versuch einer festen IP mit allen entsprechenden Parametern hatte ich auch bereits gemacht.
Warum es ausgerechnet nach der Konstellation funktioniert hat weiß ich leider nicht.
Ich hab mich aber auch noch nicht wieder getraut Sie zu überschreiben um es erneut zu Versuchen jetzt wo einmal alles wieder läuft  :-)
Vielen Dank für das tolle Projekt 👍

DennisXK, 2022-11-24:
@stefan123t die Verkabelung passt, ich hatten dem wemos abgezogen zum flaschen. Der Rest der dtu ist in der Garage neben dem Wechselrichter!

Ein Passwort hatte ich nicht vergeben, muss ich testen ob es daran liegt.

Warum gibt es eig 2 Versionen von der 5.41? wo liegt der unterschied, wo 1mb dabei steht?

DukeWolle, 2022-11-24:
@DennisXK 1m ist die 1mb Flash Version  ...

DukeWolle, 2022-11-24:
@DennisXK sprich die 1m Version ist eigentlich eine 8265 Version.. wird glaube ich demnächst auch umbenannt...

DennisXK, 2022-11-24:
Ok, und was ist der unterschied? ob man jetzt die 1mb Version aufspielt oder die kleine Version?

DukeWolle, 2022-11-24:
Der d1 mini esp8266 hat default 4mb Flash.... man nimmt also die ohne 1m ...

kki03, 2022-11-25:
Ich hatte nach dem Flashen der 0.5.41 ein ähnliches Problem. Beim ersten Start hat er sich im AP Modus gestartet und ich habe dann meine WLAN Zugangsdaten hinterlegt. Nach dem Neustart war Ahoy nicht mehr erreichbar und hat auch nicht versucht sich zu meinem WLAN zu verbinden. 

Ich hatte zuvor in meiner config_override.h die WLAN Zugangsdaten drinnen:
```
#undef FB_WIFI_SSID
#define FB_WIFI_SSID    "<MY-SSID>"
#undef FB_WIFI_PWD
#define FB_WIFI_PWD     "<MY-PWD>"
```
Erst als ich die folgende Zeile ergänzt habe hat es wieder funktioniert.
```
#define FB_WIFI_OVERRIDDEN
```

stefan123t, 2022-11-25:
@kki03 ja das ist richtig die Option wurde ergänzt und nur wenn diese gesetzt ist werden auch die fix hinterlegten SSID & Passwort verwendet. Sonst versucht er sich automatisch mit den Platzhaltern YOUR_SSID_NAME/PASS zu verbinden was auch nicht gut ist.

@DennisXK hat mW aber auch nicht selbst kompiliert sondern das Binary verwendet.

kki03, 2022-11-26:
@stefan123t ja das Szenario war bei mir durchaus etwas anders aber der eine oder andere kann da durchaus auch noch in das fehlende FB_WIFI_OVERRIDDEN rein laufen.

Auf jeden Fall hatte ich zuvor eine selbst gebaute 0.5.28 mit config_override.h laufen die ich dann über die Web-Oberfläche mit der 0.5.41 aktualisiert habe. Im AP-Modus konnte ich dann zwar meine WLAN Zugangsdaten hinterlegen die wurden jedoch beim Neustart immer ignoriert und die DTU hat mit der IP 192.168.4.1 gestartet (ohne AP Modus) und war logischerweise nicht erreichbar da falsches Subnet.

stefan123t, 2022-11-26:
siehe auch #459

lumapu, 2022-11-27:
> Auf jeden Fall hatte ich zuvor eine selbst gebaute 0.5.28 mit config_override.h laufen die ich dann über die Web-Oberfläche mit der 0.5.41 aktualisiert habe. Im AP-Modus konnte ich dann zwar meine [...]

Das ist interessant mit dem falschen Subnetz. Ich muss mal schauen, ob man es irgendwie reproduzieren kann.

closed 2022-12-30 by lumapu