lumapu/ahoy #450: Live Ansicht ohne Funktion 5.28

DennisXK, 2022-11-22:
Die live Ansicht in der Version 5.28 funktioniert auf meinem IPhone Safari und Firefox leider nicht. Am Pc (win10 firefox und MacOs Safari) funktioniert die Live Ansicht.

[URL=https://bilderupload.org/bild/848546235-a1e55fa8-570b-446e-b3b3-0][IMG]https://bilderupload.org/image/thumbnail/resized-848546235-a1e55fa8-570b-446e-b3b3-0.png[/IMG][/URL]

roku133, 2022-11-22:
Was passiert, wenn du auf das Symbol mit den drei Balken links oben klickst?

DennisXK, 2022-11-22:
Dann geht das Menü auf. Drücke ich wieder auf die 3 Striche, verschwindet das Menü und ich sehe wieder die weise Oberfläche mit der Info "Every seconds the values are updated"

roku133, 2022-11-22:
Was hat dein IPhone für eine Display-Auflösung?
Was wird bei deinem IPhone im Querformat angezeigt?

DennisXK, 2022-11-22:
Im Querformat ist das Problem weiter vorhanden. Mein Iphone hat ein 4,7Zoll Display mit einer Auflösung von

closed 2022-11-22 by DennisXK

knickohr, 2022-11-22:
Laß mich raten :  iOS 14 oder älter 😉 Ich kenne das Problem, hier liegt ein altes iPad mini 2 rum 😞

DennisXK, 2022-11-23:
ja und woran liegt es jetzt? kann ja nicht sein nur weil mein nicht das neuste ios installiert hat, dass es dann nicht angezeigt bekommt!

knickohr, 2022-11-23:
Doch, ist so. Leider 😞