lumapu/ahoy #451: Limit unter Live in Prozent und Watt

Leviathan09, 2022-11-23:
Hi,
wäre es möglich unter dem Reiter "Live" nicht nur den prozentualen Limit-Wert sondern auch das Limit in Watt anzuzeigen?

stefan123t, 2022-11-23:
Der Wechselrichter liefert das aktuelle PowerLimit **nur** in Prozent an die AhoyDTU zurück. D.h. man könnte es aus dem Prozentwert und der Leistung des Wechselrichters berechnen.

@Leviathan09 das ist prinzipiell möglich wenn man die maximale Leistung des WR kennt. @tbnobody hat das in der OpenDTU basierend auf der HW ID des Wechselrichters eine Lookup Tabelle mit den einzelnen Modellen und deren max Leistung umgesetzt.

https://github.com/tbnobody/OpenDTU/issues/363

stefan123t, 2022-11-24:
Siehe auch tbnobody/OpenDTU#35 für dessen Implementierung, speziell hier:
https://github.com/tbnobody/OpenDTU/issues/35#issuecomment-1264626167

lumapu, 2023-09-16:
duplicates #201

closed 2023-12-29 by lumapu