lumapu/ahoy #455: absolutes Powerlimit Verhalten - WR oder Ahoy?

AsFaBw, 2022-11-24:
Hallo,

(grosses Lob für das Projekt. Ahoy ist der Grund warum ich überhaupt ein Balkonkraftwerk gekauft habe.)

Ich habe bei meinem HM800 ein persistentes Limit von 200W absolut gesetzt. Ich habe 2 Module an dem WR.

Mit der tief stehenden Sonne war ein Modul heute überwiegend verschattet. Das zweite schön in der Sonne.
Ich habe bemerkt, dass das verschattete Module so zwischen 30 und 107 Watt P_DC heute sich bewegt hat. Das nicht verschattete war sehr konstant bei ~107W P_DC. In Summe bewegte ich mich also zwischen ~130 und ~204W am WR Ausgang (meist unter dem Limit von 200W).

Da die 107W P_DC and dem einen Modul sehr konstant waren, habe ich das Limit mal auf 400W gesetzt, Siehe da, die Sonne reichte für ~170W auf dem unverschatteten Modul und bin so auf ~240W Gesamtleistung gekommen.

Es scheint als würde das Gesamt WR Limit also zu jeweils 50% auf die beiden Kanäle aufgeteilt. Unabhängig davon, ob der eine Kanal unterperformt und somit der andere Kanal komepensieren dürfte. 
Oder anders ausgedrückt. Bei einem Gesamt-Limit von 200W für den WR, dürfte der eine Kanal 150W liefern, wenn der andere nur 50W wegen Verschattung bringt. Durch das Aufteilen der Limitierung auf die einzelnen Kanäle erreicht man aber nur 150W gesamt, wenn der eine Kanal nur 50W bringen kann.

Meine Annahme:
Ahoy stellt das Limit für den WR Ausgang ein, und die Steuerung des Limits ist etwas unglücklich im WR selbst implementiert. Richtig?

## Hardware
  * [ ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.28
Github Hash: 2e08ee0

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [x] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

stefan123t, 2022-11-24:
@AsFaBw Richtig. Du könntest ja das Limit auf 2x170W setzen, dann fährst Du dauernd mit angezogener Handbremse oder Du lässt es mit 100% permanent gleich unlimitiert :) Bei den noch größeren HM-1000..1500 wird es glaube ich noch komplizierter weil sich da zwei Module einen MPPT teilen, d.h. der hat eigentlich nur zwei aber vier Anschlüsse.

AsFaBw, 2022-11-24:
@stefan123t vielen Dank für die schnelle Bestätigung.

Mein langfristiges Ziel ist Richtung 0-Einspeisung zu kommen (Speisung des WR aus Akku und Regelung der Leistung abhängig vom Momentanverbrauch). 

Kurzfristig (bis die Anmeldung möglich ist) möchte ich nur den Ertrag soweit begrenzen, dass der elektronische Zähler keine Einspeisung registriert (deswegen das harte Limit auf 200W, das Tagsüber immer unter meinem Momentanverbrauch liegt). Nach der Anmeldung könnte ich, bis die Akkulösung da ist, ohne Limit laufen lassen. 

Richtig, HM1xxx haben nur zwei MPPT für die 4 Module.

Bei der Akkulösung habe ich jetzt vermutlich das Problem, dass ich nur 400W aus dem HM800 herausholen kann, wenn das Limit auf die Kanäle aufgeteilt wird und ich den Akku nicht an beide Eingänge schalte (ein Kurzschliessen der Eingänge ist im Handbuch des WR untersagt). Ursprünglich hatte ich vor, zwischen ~150W Grundlast und 600W legale Balkonkraftwerkgrenze die Einspeisung mit dem HM800 aus einem Akku zu realisieren.

stefan123t, 2022-11-24:
Nein mehr als 100% geht nicht. Der Inverter speichert das Limit intern nur in Prozent %.
Auch wenn er ein Limit ggf. auch mit Watt-Angabe akzeptiert wird beim `0x15 DeviceInfo` Command `0x05 SystemConfigPara` SubCommand nur das Limit in % an die AhoyDTU zurückgegeben. 
Und nein, mehr als 100% akzeptiert er da auch nicht, egal wie man es ihm setzt oder beibiegen möchte. Ggf. setzt er es einfach auf 100% wenn man ihm größere Werte per Active PowerLimit übergibt.

Was macht er denn wenn Du ihm 75% als PowerLimit mitgibst, dann sind das nur 300W pro MPPT ?
D.h. Du kannst ihn nur bis 400W pro MPPT regeln, weil ein einzelner MPPT Eingang auch nicht für >400..800W dimensioniert ist.

knickohr, 2022-11-25:
Solange Du nicht über 600W pro Zähler gehst, kräht kein Hahn danach, selbst wenn es die 600 komplett einspeist.

AsFaBw, 2022-11-25:
@stefan123t 
Vielen lieben Dank für die Rückmeldung. Das bringt mir Klarheit und wir können das Thema damit von mir aus schliessen.

>Was macht er denn wenn Du ihm 75% als PowerLimit mitgibst, dann sind das nur 300W pro MPPT ?

Habe ich noch nicht ausprobiert (mir fehlt ein wenig Sonne). Aber ich habe noch mal Screenshots mit 200W Limit und 500W Limit gemacht. Bei dem 200W Limit sieht man schön die Begrenzung des einen Kanals auf ~100W (107.1W P_DC).
[image]
[image]

>D.h. Du kannst ihn nur bis 400W pro MPPT regeln, weil ein einzelner MPPT Eingang auch nicht für >400..800W dimensioniert ist.

Bezüglich Dimensionierung der beiden Kanäle bin ich mir nicht sicher, ob die Grenze tatsächlich 400W ist.
Laut Datenblatt liegen die Max Ratings bei 12,5A, Max V 60V und der MPPT Arbeitsbereich bis 48V.
Als Zusatzangabe gibt es ein „commonly used module power(W) 320~500“. Das hört sich so ein wenig nach Empfehlung an, ist aber auch schon mal über die 400W.
Wenn man mal nur im MPPT Arbeitsbereich bleibt, hat man pro Kanal 12,5A * 48V = 600W
[image]

Und das war mein Hintergedanke. Nimm den HM800 um ihn mit einem 48V Akku im 600W Balkonkraftwerkbereich über einen Kanal zu betreiben.

Mir fehlt die Sonne, aber ich werde im Sommer sicher mal ausprobieren, wie sich der HM800 mit einem Modul verhält, wenn kein Limit gesetzt ist. Liefert er tatsächlich nur 400W oder liefert er mehr.

@knickohr
>Solange Du nicht über 600W pro Zähler gehst, kräht kein Hahn danach, selbst wenn es die 600 komplett einspeist.

Ein wenig off-topic von mir (Hintergrund warum ich das Limit benutzen möchte).
Ich habe Platz für ~4-5 Module, die ich selbst ohne Handwerker installieren kann. Um ohne Handwerker die Anmeldung hinzubekommen, möchte ich im Balkonkraftwerkbereich bleiben. Also ist eigentlich nur die 0-Einspeisung sinnvoll. Das, was ich tagsüber zu viel ernte, soll in einen Akku gehen und abends/nachts verbraucht werden.

Zu dem Thema gibt es ein schönes Video von Andreas Schmitz  ([Balkonsolar mit AKKU - So geht's richtig ! - YouTube](https://www.youtube.com/watch?v=yOcoux9IbzM)). BTW: Mit dem EVT560 WR hätte ich auch Bedenken, bezüglich Überlastung eines Kanals.
[image]

Mit der Erkenntnis darüber, wie der HM800 das Limit intern behandelt, könnte ich auch den billigeren HM400 nehmen. Anbei mein aktueller Plan, wie ich im Balkonkraftwerkbereich trotzdem die Energie von 5 Modulen nutzen kann…
[image]

closed 2022-11-30 by AsFaBw