lumapu/ahoy #456: Feature Request: Setzen der Limite über serielle Schnittstelle am ESP

cbscpe2, 2022-11-24:
Ich wünsche mir, dass man die Limit nicht nur via MQTT sondern auch über eine serielle Schnittstelle am ESP setzen könnte. Im Moment betreibe ich nur wegen dem Setzen der Limit einen MQTT Server.

Freeenergy-man, 2022-11-24:
Hallo, habe aktuell eine Solaranlage mit Speicher und möchte gerne das ganze mit Nulleinspeisung realisieren.
DTU Ahoy und Tasmota am Zähler habe ich schon zum laufen gebracht.
Jetzt fehlt nur noch die automatische Leistungsregelung.
Dazu hatte ich heute eine sehr gute Idee :)
Kann man Ahoy Dtu und Tasmota nicht zusammen als ein Projekt realisieren?
Dann hätte man eine Steuerung am Zähler die alles macht.
Ich glaube, das würde sehr vielen helfen.

stefan123t, 2022-11-24:
Siehe #431 Steuerung des PowerLimits nach Werten aus MQTT Topic, TTL Eingang oder Modbus RS485 Abfrage eines SmartMeters.

closed 2022-11-24 by stefan123t