lumapu/ahoy #457: Settings JSON file

bnnbg, 2022-11-24:
## Hardware
  ESP8266
Modelname: wmos mini
Retailer URL: ______

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [X] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [X] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.42
Github Hash: 43af077

### Build & Flash Method:
* [ ] Arduino
* [X] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [X] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo allerseits,
Erstmal vielen Dank für euren Einsatz bei der Entwicklung der DTU hier...
Evtl. stehe ich etwas auf dem Schlauch... Ich habe hier schon gesucht, aber keine passende Antwort gefunden.
Ich habe gestern über das Handy die dtu auf 0.5.42 upgedatet und danach war sie nicht mehr erreichbar.
Es war dann der AP Modus aktiv und nach dem sie wieder im WLAN war, wollte ich das vorher heruntergeladene json Sicherung wieder einspielen.
Nur finde ich keine Option zum zurückspielen.
Gibt's die Option nicht?
Für was ist dann die Funktion der Sicherung der settings da?
Sorry für die Frage, hoffe eine hilfreiche Antwort zu bekommen...
Grüße

stefan123t, 2022-11-24:
Ja das ist ein neues Feature und die zweite Option Upload JSON Settings gibt es noch nicht.
Ich lasse das mal als Reminder für @lumapu hier offen, dann kann er das ja bei Zeiten mal implementieren ...

bnnbg, 2022-11-24:
Ja super... Vielen Dank für die schnelle Reaktion... Top

closed 2023-05-25 by lumapu