lumapu/ahoy #458: [ESP8266/ESP32/RaspberryPi] Problem Description / Beschreibung

Karlo49, 2022-11-25:
### Platform

ESP8266

### Model name

D1 Mini Pro - ESP8266, CP2104

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.41

### Github Hash

dec333f

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

Hallo,
erstmal Danke für das Projekt.

Nachdem ich die ahoy_v0.5.41 über Update installiert habe habe ich mich das ersre Mal mit MQTT beschäftigt.Habe den Mosquitto-Broker auf einem alten Linux Notebook zum Test installiert. Dann hatte ich nach Eintrag der Server-IP inden MQTT-Settings sofort Daten auf dem Notebook.
Habe das Notebook und damit den Mosquitto-Broker am Abend ausgeschaltet.
Am nächsten Morgen bekam ich keine Verbindung vom Ahoy-DTU zum Inverter.
Kein reset, stromlos, nichts half. Erst nach einschalten des Notebooks war die Verbindung wieder da.
Habe dann Versucht durch Löschen der Server-IP die Vorherige Situation wiederherzustellen.
Aber die Server-IP kommt nach dem Speichern der Settings immer wieder.

lumapu, 2022-11-25:
danke für den Fehlerbericht. Wäre es möglich die Development Version zu testen? Zu finden unter https://ahoydtu.de

Die broker IP man man evtl durch eine einzelne '0' löschen.

Karlo49, 2022-11-25:
Danke für die Antwort,

Die 0.5.43 hab ich schon getestet. Die friert dauernd ein.

Die "0" teste ich

Karlo49, 2022-11-26:
Auch mit 0 als Server-IP am Morgen keine Verbindung zum Inverter.
Es funktioniert aber mit meiner Router_IP

stefan123t, 2022-11-26:
@Karlo49 danke für die Tests, es scheint mir so zu sein, daß die MQTT Bibliothek blockiert bzw. sehr viel Rechenzeit benötigt falls man eine IP / Namen konfiguriert hat, diese aber aus unterschiedlichen Gründen (WLAN Qualität, Server down, etc.) nicht erreichbar ist, dann ist der ESP m.W. sehr viel damit beschäftigt das herauszufinden und kommt nicht mehr dazu die etwas niedriger priorisierten Tasks (TX & RX zum WR, etc.) abzuarbeiten.

@lumapu entweder das wird mit der neuen MQTT Bibliothek besser oder wir müssen uns das Task Management / Priorisierung nochmal genauer ansehen.

knickohr, 2022-11-27:
Kann ich bestätigen 😲 Nachdem die Verbindung heute Nacht weg war hat sich die DTU aufgehängt. Das übliche Bild :  Seite lahmt, Inverter nicht erreichbar. Erst ein Reboot der DTU brachte das normale Verhalten zurück. Kurioserweise war der Broker aber (wieder) erreichbar. Die DTU lieferte sogar noch Daten ein, hat sich aber nicht mehr erholt.

lumapu, 2022-12-01:
@Karlo49 @knickohr bitte mit `0.5.45` Dev testen

Karlo49, 2022-12-01:
Update von 0.5.41 auf 0.5.45 ok
Umschalten im Menü und es hängt sich auf. Im Dunkeln, war vorher auch so.
Teste Morgen, wenn der Inverter wieder Daten liefert.

reserve85, 2022-12-01:
Gleiches Verhalten auch bei mir. DTU quasi tot.

Karlo49, 2022-12-02:
MQTT Fehler wie oben beschrieben  ist behoben!
Was bleibt sind die Abstürze bei Bedienung des Menües wenn Inverter keine Daten liefert, oder beim Ändern der Sttings.

Snoopy8700, 2022-12-02:
Bestätigung meinerseits: ESP6288 D1 Mini 
Inverter Nummer eingetragen und Panel Daten.
Inverter #0: 800HRSO.EU.HM (v0) is available and is not producing
WARN: MQTT is not connected

closed 2022-12-19 by Karlo49

laserdoktor, 2022-12-19:
Guten Tag,
nur eine kurze Frage als dankender stiller Teilnehmer: Ich kann keine neue Version zum Laufen bringen. Bis 221106_ahoy_0.5.32_esp8266_1m_700c5c7 ist alles gut, eine neuere Version bekommt gar keine Verbindung bzw. ist extrem zäh beim Setup sodas ich wieder zurück auf 0.5.32 flasche. Liegt es an mir oder ist in der Hardwate etwas nach 0.5.32 anders:

knickohr, 2022-12-19:
Welche Versionen wurden ausprobiert ?  Auch die stable 41 ?

Karlo49, 2022-12-19:
Nein, nicht die 0.5.41. Da ist der Fehler ja noch drin. eine neuere Version. Weis nicht mehr welche.
Habe jetzt die 0.5.56

knickohr, 2022-12-19:
Erstaufführung die 41, dann per OTA auf 56

laserdoktor, 2022-12-19:
habe die 0.5.41 flaschen können über OTA, es hat aber recht lange gedauert bis Daten vom WR kamen, Habe bis "jetzt", 2 Stunden, keine Auffälligkeiten gesehen. Die 56 werde ich morgen, wenn es hell ist, probieren und berichten. Ich betreibe einen ESP12 als Rechner, da ist das "Hardware-Flaschen" nicht ganz so entspannt.

Snoopy8700, 2022-12-19:
habs auch geflashed mit dem NodeMCU ... bei mir ging es rasch. Nun brauche ich die Sonne morgen, um es testen zu können. Danke Karlo49 !

laserdoktor, 2022-12-20:
Malzeit;
Mein Flash-Board har geglüht, aber leider umsonst:
von 0.5.32 auf 0.5.41 ging es ohne Probleme, etwas zäh beim 1. Daten holen, von 41 auf 56 geht gar nicht: flash über OTA wie mit Flash-Board keine Verbindung zum WLAN, die baue LED blinkt aber kein WLAN.
Wenn ich dann mit dem Flash-Board zurück auf 0.5.32 flashe ist sofort alles da, es sind keinerlei Setup Eingaben erforderlich.
Meie HW ist ein ESP12F und NRF24L01+, Spannungsversorgung 5V -> 3,3V, 100 µF, 220µF, 100nF
Vielen Dank bis hierher für Eure Arbeit

laserdoktor, 2022-12-20:
Nachtrag:
habe gerade gesehen, daß es ein Update auf 0.5.57 gibt und auch gleich über OTA geflsht: Es ging sofort, alles scheint ok zu sein.
Man sollte das "Communication Enable" bei Settings-> Inverter NICH übersehen. sonst kommt keine Verbindung zum WR zustande.

laserdoktor, 2022-12-21:
die 0.57 ging sofort, aber man muss "Communication Enable" bei Settings-> Inverter (über Serial Number) aktivieren.
Die 0,58 habe ich nicht getestet, bleibe erst einmal bei 0,57

Ein Frohes Fest und ein Gesundes Neues Jahr allen im Forum und nochmals herzlichen Dank für die tolle Arbeit!