lumapu/ahoy #459: Zeit synchronisiert nicht wenn statische IP gesetzt ist

vobo-1969, 2022-11-26:
### Platform

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

--

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.41

### Github Hash

dec333f

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

--

### Debug Serial Log output

```shell
--
```

### Error description

Hallo,

folgenden Fehler habe ich bei der aktuellen Version 0.5.41 festgestellt:

wenn eine statische IP eingestellt ist, erfolgt keine Zeitsynchronisation via NTP.
(die IP, das Subnetz, DNS und Gateway sind korrekt eingestellt!)
Es findet keine Kommunikation mit dem Wechselrichter statt bis die Zeit per Button mit dem Browser synchronisiert wurde.
Wenn das geschehen ist, funktioniert alles normal.

Wird die IP Adresse per DHCP angefordert, tritt der Fehler nicht auf; das Modul synchronisiert normal per NTP.

stefan123t, 2022-11-26:
@vobo-1969 ja das hatte jemand anderes auch schon reported dass wenn er den DNS Eintrag gemacht hat die Zeit nicht synchronisiert wurde und damit auch kein WR abgefragt wurde. Erst nach manuellem Sync funktionierte AhoyDTU wie es soll.

@lumapu da scheint noch was mit der NTP Logik nicht ganz zu passen. Sollen wir nicht nach jedem Reconnect mit dem WiFi auch ein NTP Sync machen oder wird das dann zu häufig bei schlechten WLAN Verbindungen ?

stefan123t, 2022-11-26:
Ich habe aktuell **keine statische IP** gesetzt, bekomme aber auch folgende Meldungen auf der Serial Console (nach über 48 Stunden uptime). Das UI ist seitdem auch nicht mehr erreichbar.
```c
W: Time not set or it is night time, therefore no communication to the inverter!
W: Time not set or it is night time, therefore no communication to the inverter!
[129601000][E][WiFiGeneric.cpp:1438] hostByName(): DNS Failed for pool.ntp.org
[129601002][E][WiFiUdp.cpp:183] endPacket(): could not send data: 118
I: [NTP]: n/a UTC
...
W: Time not set or it is night time, therefore no communication to the inverter!
W: Time not set or it is night time, therefore no communication to the inverter!
[166261000][E][WiFiGeneric.cpp:1438] hostByName(): DNS Failed for pool.ntp.org
[166261001][E][WiFiUdp.cpp:183] endPacket(): could not send data: 118
I: [NTP]: n/a UTC
```

ebs37, 2022-11-27:
Ich kann den Fehler nicht bestätigen. Habe auch eine feste IP vergeben, NTP-Server ist mein Router. Nach dem Neustart/Reboot dauert es genau eine Minute bis zur Zeitsynchronisation, danach werden alle Werte richtig angezeigt. 
V5.41, ESP8266

reserve85, 2022-11-28:
ich habe festgestellt, dass die Zeit nicht synchronisiert wenn man mit ständiger F5-Taste auf der Main-Seite wartet bis über NTP die Zeit synchronisiert wird. Kann das jemand bestätigen? Scheint als würde der Sync dann abgebrochen werden. FW 0.5.42

ebs37, 2022-11-28:
> ich habe festgestellt, dass die Zeit nicht synchronisiert wenn man mit ständiger F5-Taste auf der Main-Seite wartet bis über NTP die Zeit synchronisiert wird. Kann das jemand bestätigen? Scheint als würde der Sync dann abgebrochen werden. FW 0.5.42

Kann ich auch nicht bestätigen. Im Chrome nach 1 min 7 s und "vielen F5" steht die Verbindung, Zeit ist ok (ntp-Quelle ist der Router).

reserve85, 2022-11-28:
Kannst du mal versuchen die Inverter-Polltime auf 5s zu stellen?

lumapu, 2022-11-28:
ich habe gesehen, das automatisch an die IP noch was angehängt wurde. Habe das jetzt behoben und bei mir erfolgreich getestet

lumapu, 2022-12-01:
@vobo-1969 konntest du nochmal testen mit der dev Version?

vobo-1969, 2022-12-01:
@lumapu nein, leider noch nicht. Bin momentan auf einer Dienstreise. Werde es am WE aber testen!

vobo-1969, 2022-12-03:
@lumapu so ... habe gerade mal getestet, hier die Ergebnisse:

1. die Dev Version 0.5.47 d11d313 mit dem PyFlasher aufgespielt (mit Erase) -> der AP wird nicht aufgebaut, keine Verbindung möglich
2. die 0.5.41 wieder aufgespielt und Ahoy, wie gehabt, mit statischer IP in das Netz gebracht, dann die Version 0.5.47 per Webupdate aufgespielt -> Update hat geklappt -> Verhalten hat sich nicht geändert, auch nach 15 Minuten hat sich die Zeit nicht synchronisiert
3. anstatt des vorgegebenen Zeitservers meine Fritzbox eingetragen -> Zeit wird sofort richtig synchronisiert

Zusatzinfo:
mein Adressraum im Netz umfasst 4094 mögliche Host´s, daher ist meine Netzwerkmaske 255.255.240.0 (/20).
Der Adressraum reicht von 192.168.112.1 bis 192.168.127.254
(das sollte aber normalerweise nichts mit dem Fehler zu tun haben)

Viele Grüße und ein schönes WE
Volker

lumapu, 2022-12-03:
welchen DNS verwendest du? `1.0.0.1` oder `8.8.8.8`

Ich habe bisher nur mit der IP von der FritzBox getestet. Ich habe auch ein /20 Netz.
Ein Test soeben war bei mir mit der `1.0.0.1` und FW `0.5.47` sofort erfolgreich.

[image]

vobo-1969, 2022-12-03:
Ich habe den Router als DNS eingetragen.

Aber ich habe gerade was anderes festgestellt ....

Als Gateway habe ich natürlich auch den Router eingetragen (192.168.115.1)
Wenn ich Ahoy dann aber neu starte, steht da auf einmal 192.168.115.10 ?
Das habe ich mehrfach ausprobiert ....
Auf einmal hat es dann geklappt, nach dem Neustart stand da noch die 192.168.115.1 ... und die Zeit hat sich synchronisiert.
Also scheint es irgendwas mit dem Speichern der Einstellungen zu tun zu haben ?
[image]

PeterObs, 2022-12-03:
Ist bei mir auch so, nur wird eine 3 statt einer 0 hinzugefügt. Aus 192.168.3.1 wird dann 192.168.3.13.

lumapu, 2022-12-03:
ja ich dachte ich hab das behoben, ich schaue nochmal in den Code. Bei mir war es eine 4 damals

HerbV, 2022-12-29:
Hallo, bei mir sollte es 192.168.1.1 sind doch will er immer .11 statt .1 speichern. Überhaupt kann ich keine Änderung der Konfiguration durchführen. Um den MQTT Server einzutragen habe ich alles neu installiert, dabei gleich den Server eingetragen, damit gings, aber das Gateway bekomme ich niemals richtig hin. Da nke für Eure Bemühungen

lumapu, 2022-12-30:
@HerbV welche Version ist im Einsatz?

closed 2022-12-30 by lumapu