lumapu/ahoy #460: Total W via MQTT

Ximon84, 2022-11-28:
Moin,

kann man auch die Values untern total via MQTT abfragen?
An die Werte der Inverter komme ich heran, das läuft sehr gut.

DANKE

DukeWolle, 2022-11-28:
@Ximon84 mit was willst du denn abrufen ?

Ximon84, 2022-11-28:
@DukeWolle die Gesamt-Leistung W aktuell aller Wechselrichter oder auch die gesamt kWh..

DukeWolle, 2022-11-28:
@Ximon84 ich stell meine Frage gerne nochmal. "Mit was willst du abfragen?". Was du willst ist mir klar. Hast du homeassistant laufen ... oder ioBroker, Mosquitto etc. ???

Ximon84, 2022-11-29:
Rennt, DANKE!

[image]

stefan123t, 2022-11-30:
Prima dann mache ich das Issue mal als gelöst zu.

closed 2022-12-01 by lumapu