lumapu/ahoy #463: [ESP8266/ESP32/RaspberryPi] Problem Description / Beschreibung

kawi73, 2022-11-29:
### Platform

ESP8266

### Model name

D1 Mini

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.44

### Github Hash

0a5833e

### Build & Flash Method

Platform IO (build & flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 114173307265
    - Name: TSOL-M800
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 400, 400
    - Module Name: N/A
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123

### Error description

Die Version 0.5.44 reportet in der Web Serial Console die UTC Zeit. Datum und Uhrzeit in der Übsersichtseite werden richtig angezeigt, d.h. die Zeitsynchronisierung funktioniert.

kawi73, 2022-11-30:
Also in der Version 0.5.41 wird die Zeit richtig angezeigt in der Webconsole. In der Version 0.5.4 wir die UTC Zeit angezeigt. Deshalb habe ich es erwähnt.

kawi73, 2022-11-30:
Einen Screenshot habe ich auf dem Discordserver hich geladen

kawi73, 2022-11-30:
[image]

kawi73, 2022-11-30:
Das ist ein Screenshot mit der Version 0.5.41 gerade eben erstellt.

kawi73, 2022-11-30:
Beides der Chrome. Einmal unter Windows und einmal auf Android.

kawi73, 2022-11-30:
Aber wenn 0.5.44 wie erwartet reagiert, ist das ok. Dann kann man den Thread schließen.

Gerri1, 2022-11-30:
Im Firefox und Edge wird bei mir auch die UTC Zeit angezeigt (44), bei der 41 war das i.O.

stefan123t, 2022-11-30:
Ja das unterwartete in 0.5.41 mit Chrome unter Android kann ich mir aktuell nicht erklären, außer dass der Browser eine lokale Übersetzung der Zeiten macht. Manchmal werden ja auf dem Smartphone auch Telefon Links u.a. anders markiert damit man sie direkt anklicken / anrufen kann. Vielleicht gibt es das auch für die Uhrzeit / Lokalisierung ?

@kawi73 kannst Du das auch unter v0.5.44 mit Chrome auf Android reproduzieren ? 
@Gerri1 wir müssen hier m.E. auch noch die Plattform Android / Desktop unterscheiden ?

lumapu, 2022-12-01:
der Fehler war im Zusammenhang der Web API. Die habe ich umgestellt und die Web Console vergessen nachzuziehen

Gerri1, 2022-12-01:
TOP ... läuft wieder!!! 😊👍

closed 2022-12-30 by lumapu