lumapu/ahoy #465: Display & Security

Peertje62, 2022-11-30:
Hi,
( I am Dutch &My English is a bit rusty)
- An 2.7inch E-Paper Display Module 264x176 Resolution would be nice.
- Password protection is super but, the live page could do without an pw (if possible)

Kind regards
Peertje

stefan123t, 2022-11-30:
Hoi @Peertje62 mijn nederlands is ook prettig rusty 😁 
You may look at #276 there gf2hx has provided code to hook a nokia 5110 or an OLED display into AhoyDTU. Though it is a PR for an older version you should be able to follow the general idea. Note that the code base in AhoyDTU has been restrictured lately.
Dui, Stefan

Peertje62, 2022-11-30:
Thx,
That's ''Adakadabra'' for me,
I'll leave it as it is, working perfect!
Regards Peertje

Peertje62, 2022-12-02:
Hi, it me again
correct me if I'am wrong (pls)
Buy a nokia5110-display, connect VCC , GND, D1 - CE & D2 -DC and it will work?
Kind regards
Peertje

Peertje62, 2022-12-03:
Hi, Stefan,
I'am still learning every day!
Could you/Would you  include this in the next update ?
(Ahoy, esp8266, nokia5110)
Kind regards
Herman

lumapu, 2022-12-03:
I have the pull request and its code, but it has to be converted to a class. Was not included until now

Peertje62, 2022-12-03:
Hi,
that would be nice if you did, 
kind regards,
Peertje

closed 2022-12-03 by Peertje62