lumapu/ahoy #466: Fehler bei der sunrise/sunset Berechnung

beegee3, 2022-11-30:
Hab' mich über die Sonnenaufgang und -untergang Zeiten gewundert:
in calculateSunriseSunset ist die Formel für omega falsch (vgl. Wikipedia):
statt
  double omega = ACOS(SIN(-0.83) - SIN(lat) * SIN(delta) / COS(lat) * COS(delta));
sollte es
  double omega = ACOS((SIN(-0.83) - SIN(lat) * SIN(delta)) / (COS(lat) * COS(delta)));
sein.

knickohr, 2022-11-30:
Ja, die Werte sind bei meinem Standort um ca. 10min daneben. Hat mich aber bis jetzt nicht wirklich gestört. Natürlich wäre es schön, wenn man den Fehler korrigiert.

homeautomation2022, 2022-11-30:
@beegee3 Wie groß war der Unterschied im Ergebnis(den Zeiten)?

Kleine Korrektur habe ich auch noch, wenn wir einmal dabei sind:
Ganz am Anfang die Korrektursekunden müssten auch mal gelöscht werden (+ 0.008 enfernen), da die UTC Zeit bereits UT1 korrigiert wird und Schaltsekunden hat.

beegee3, 2022-12-01:
@homeautomation2022 der Unterschied war knapp 15 min. Da Sonnenaufgang bzw. -untergang als Transit minus bzw. plus omega gerechnet werden, waren die berechneten Werte entsprechend zu früh bzw. zu spät.
Habe mal die Korrektursekunden gelöscht, das passt auch besser, wenn man mit gegoogelten Zeiten vergleicht!

beegee3, 2022-12-01:
@lumapu sorry, der sunrise/sunset fix ist leider nicht ok,
da fehlt "Klammer zu" hinter SIN(delta) und "Klammer auf" vor COS(lat)

nur um es deutlich zu machen:

omega = ACOS( Zähler / Nenner ) mit
Zähler = SIN(-0.83) - SIN(lat) * SIN(delta)
Nenner = COS(lat) * COS(delta)

Zähler und Nenner müssen in Klammern gesetzt werden _(sonst geht's schief)_, d.h:

double omega = ACOS(`(SIN(-0.83) - SIN(lat) * SIN(delta))` / `(COS(lat) * COS(delta))`);

lumapu, 2022-12-01:
danke fürs gegenchecken, hab's nur flüchtig überflogen und zwei zusätzliche Klammern gesehen, morgen dann noch ein fix

knickohr, 2022-12-02:
Jetzt paßts, bis auf wenige Sekunden 👍 Anyway, ist nur eine Näherung, mach zu das Ding 😎

closed 2022-12-02 by beegee3

homeautomation2022, 2022-12-04:
Danke @knickohr fürs finden des Fehlers. Beim Abschreiben der Formel von Wikipedia ist mir der große Bruchstrich da wohl untergegangen. (Ich hatte Anfangs sehr große Probleme mit den Ergebnissen, da ich nicht wusste, das C Winkelfunktionen default mit Radial rechnen und habe da sehr viel rumprobiert woran es hängt...)

Falls du oder ein anderer Tag/Nacht kürzer oder länger haben möchtest, dann einfach mit der -0.83° spielen, das ist der Winkel der Sonnenhöhe, in dem Minus Fall noch hinterm Horizont. Jede Webseite definiert den Winkel des Sonnenauf-/Untergangs etwas anders und bei den PV-Modulen kommt es eh wieder auf die Ausrichtung und das Wetter drauf an, weshalb man da nie einen "Richtigen" Wert für alle finden kann. Die -0.83° passten bei mir schon relativ gut im Sommer und Herbst. Geht ja eh nur um paar Sekunden/wenige Minuten.

@lumapu Um noch etwas Aufzuräumen, der Kommentar sollte nach dem Entfernen der Korrektursekunden auch noch angepasst werden:
// Julian day since 1.1.2000 12:00 + correction 69.12s
in
// Julian day since 1.1.2000 12:00 UT

knickohr, 2022-12-04:
Nein, ich schmücke mich nicht mit fremden Federn. @beegee3 hat den Fehler gefunden und eine Lösung präsentiert.

homeautomation2022, 2022-12-05:
> Nein, ich schmücke mich nicht mit fremden Federn. @beegee3 hat den Fehler gefunden und eine Lösung präsentiert.
> 
> Es sind nur wenige Sekunden, das macht das Kraut nicht fett, laß es dabei 👍 Ich war mit den 10 Minuten Abweichung auch zufrieden 😉

Ah, OK.
Korrektursekunden sind bereits raus, nur das Kommentar passt nun nicht mehr ganz, das wollte ich nur nochmal erwähnen(für nebenbei beheben), aber der Code ansich sollte nun im dev03 in Ordnung sein.

commit 614af26c (2022-12-02): fix #466