lumapu/ahoy #468: Anpassung einiger MQTT Topics

knickohr, 2022-11-30:
Endless MQTT 🤪

Bitte den Topic (inverter)/status umbenennen in (Inverter)/mqtt und als Value nicht online/offline mitgeben, sondern besser connected/not connected. Wenn ich das richtig verstanden habe, gibt dieser Topic den Status der Verbindung zum Broker an.

Folgende Werte als retained flaggen
- YieldTotal
- YieldDay

und den folgenden Status korrekt am Ende des Tages ausgeben und ebenfalls als retained flaggen
- available_text

Wird leider beim letzten noch „lebenden“ WR nicht mehr korrekt gesetzt 😞

[image]

und korrigiert bitte den Text
- not yet available and (hier fehlt was 😲)
Mit persönlich würde auch ein „not available“ reichen

Das war’s vorerst mal, bis mir wieder was einfällt 🤪

lumapu, 2022-12-01:
danke für die Zusammenfassung, für kann man sehr gut einarbeiten

knickohr, 2022-12-01:
Danke für die schnelle Umsetzung des ersten Teils 😎👍
[image]

Warten wir gespannt was noch kommen wird 😅

knickohr, 2022-12-02:
Die Yield-Werte sind jetzt mit der 46 auch retained geflaggt 👍

stefan123t, 2022-12-04:
@knickohr fehlt hier noch was oder kann @lumapu das issue mit dem nächsten Release zu machen ?

lumapu, 2022-12-04:
@stefan123t ja es fehlt noch was:

> Wenn alle Inverter nicht mehr available sind, könnte man ein Status offline setzen. Hierzu könnte man den alten Topic (inverter)/status verwenden, würde wesentlich besser passen.

knickohr, 2022-12-04:
> 

Korrekt.

und der available_statur wird immer noch nicht nach Sonnenuntergang korrekt über MQTT ausgegeben, bleibt immer wieder noch beim vorletzten Wert hängen 🙁

stefan123t, 2022-12-06:
@knickohr, also einmal kurz nach Sonnenuntergang "0" Werte ausgeben bzw. den available_status einmalig auf offline / unavailable setzen, wenn das Häkchen für die Sundown Regelung aktiv ist. Richtig ?
Und für alle noch zusätzlich einmal um 0:00 Uhr die Yield Daily Werte auch auf den "0" Wert setzen.

knickohr, 2022-12-06:
Nur der available_text korrekt über MQTT auch nach Sonnenuntergang setzten, mehr braucht’s nicht mehr. Die Werte kann ich dann selbst nullen, nicht das es nochmal eine heftige Diskussion gibt 😉

Und vielleicht den satus auf offline setzen.

roku133, 2022-12-06:
Mir gefällt der Vorschlag von @stefan123t weiter gut, kann aber auch mit der reduzierten Variante von @knickohr arbeiten. Das entspricht dann auch der Anzeige auf der Weboberfläche. 
Vielleicht kann @stefan123t in diesem Schritt auch den MQTT-Wert für last_success reparieren. Er erhöht sich zwar beim Standardintervall von 30 s um 30000 (ms-Ticks?), der Startwert scheint aber eine Zufallszahl zu sein. Sie ist bei mir negativ.

lumapu, 2022-12-06:
wird alles behoben, an dem zeitgesteuerten setzen / zurücksetzen bin ich schon dran.
Es benötigt aber zusätzliche Optionen im Scheduler, den ich gerade entsprechend erweiterte.
Eine Testversion läuft bei mir lokal, die ist aber noch zu instabil um sie auf euch loslassen ;-)
Leider ist die heute ca. 10 min vor Sonnenuntergang abgestürzt ...

knickohr, 2022-12-07:
> Standardintervall

Macht er das wirklich ?  Dann wäre es ja korrekt wenn er immer um 30 Sekunden hoch geht, dann wäre es ein Teitstempel „vor xxx Millies“ 🤔

roku133, 2022-12-07:
Der Parameter last_success wird nur während der aktiven Stromerzeugung angezeigt. Wenn es ein Zeitstempel "vor xxx Millies) wäre, dürfte dann der Wert bei regelmäßiger Kommunikation höchstens -30000 (ms-Ticks) betragen. Die aktuelle Zahl ist bei mir jedoch last_success -327819144 (v0.5.41 - Laufzeit ohne Unterbrechungen seit mehr als 17 Tagen!).

stefan123t, 2022-12-08:
@roku133 das ist vermmutlich nicht als unsigned int/long definiert daher das negative Vorzeichen.

@lumapu kannst Du das in Deinem Branch / MQTT Patch berücksichtigen ? Wäre vermutlich ausreichend hier nur Sekunden auszugeben 😄

knickohr, 2022-12-08:
Naja, ich würde mal sagen, so wie die uptime, als Unix Timestamp.

roku133, 2022-12-08:
kleine Ergänzung:
uptime beinhaltet eine relative Zeitangabe (Sekunden seit Neustart) - geeigneter als Beispiel sind daher für mich sunrise/sunset mit absoluter Zeitangabe.

knickohr, 2022-12-08:
Jupp

lumapu, 2022-12-08:
bei dem last_success war komischer Weise ein *1000 eingebaut. Dadurch läuft der 32 Bit Integer über.

roku133, 2022-12-09:
@lumapu - perfekt, last_success zeigt mit v0.5.50 jetzt Zeit (unixtime) korrekt an - vielen Dank 👍

knickohr, 2022-12-11:
@lumapu @stefan123t 

Eine kleine Erweiterung des letzten Punktes, der noch nicht in Angriff genommen wurde.

Alt:
„ Wenn alle Inverter nicht mehr available sind, könnte man ein Status offline setzen. Hierzu könnte man den alten Topic (inverter)/status verwenden, würde wesentlich besser passen.“

Neu:
- alle Inverter online -> online
- alle Inverter offline -> offline
- einer oder mehr, aber nicht alle online oder offline -> partial
im status ausgeben.

littlebilly, 2022-12-13:
Danke für das tolle Project!
Gibt es eigentlich irgendwo eine Topic Liste?

knickohr, 2022-12-15:
52 funktioniert alles soweit, zumindest heute mal. Ich würde gerne noch ein paar Tage abwarten bevor wir das zu machen. Will testen ob es Auswirkungen hat wenn das Wetter gut ist und es länger hell ist, die Inverter länger (länger als Sonnenuntergang) arbeiten.

Eine kleine Anmerkung habe ich noch. Irgendwo kommt zwischendrin mal der available_text „not available and producing“. Das kann nicht sein 😉 Wenn nich da, dann kann nich produzieren. Ich glaube das tritt bei available (Status) 1 auf.

knickohr, 2022-12-20:
@lumapu 
Ich muß das leider nochmal rauskamen 🤪

Jetzt mit dem Communication Start/Stop fällt mir auf, das der MQTT Topic available und available_text viel zu spät kommen am Ende des Tages. Morgens funktioniert es prima, aber Abends setzt er diese Briten Stati erst beim Communication Stop. Besser wäre es, er würde sie auch setzen wenn er es erkennt. Also so wie in der Webgui (not yet available).

knickohr, 2022-12-25:
Ich schließe das weil ich die letzte Anpassung in #522 neu definiert habe.

commit 9da0fc40 (2022-12-08): fix #468 MQTT status at sunset
commit d9290d9f (2022-12-08): fix #480 AP Mode on fresh ESP
included #483 improvements
fix #468 last_success MQTT
fix #468 update available status at sunset
fix #447 reorderd enqueue commands to not have same payload length in a row
added ssd1306 and nokia display to build script
closed 2022-12-25 by knickohr