lumapu/ahoy #471: [ESP8266/ESP32/RaspberryPi] Problem Description / Beschreibung

hoffmakl1961, 2022-12-02:
### Platform

ESP8266

### Model name

ESP-WROOM-32 auf einem ESP32 DEVKITV1 Board

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.41

### Github Hash

dec333f

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: MY_WIFI_SSID 
    - Password: My_WIFI_PWD
### Inverter
#### Inverter 0
    - Address: 1141XXXXXXXX
    - Name: HM600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 300, 300
    - Module Name: M1, M2
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT - wird nicht benutzt
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (ESP32Wemos)
    - CS: D05   (GPIO05)
    - CE: D04   (GPIO04)
    - IRQ: D16 (GPIO16) - RX2

#### Radio (NRF24L01+)
    - Amplifier Power Level: HIGH
#### Serial Console
    - print inverter data: [ ]
    - Serial Debug: [ ]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Debug Serial Log output

```shell
nicht verhanden
```

### Error description

Ich habe WiFi - Verbindungsprobleme, wenn sich mein ESP32 mit dem WLAN-AP meiner Fritz!Box 7490 (FRITZ!OS: 07.29 - Version aktuell) verbindet.
Das Problem ist, dass beim ESP32 nur nach jedem 2. Programmstart eine WLAN Verbindungsversuch erfolgreich ist.

Die Ursache ist von ullixesp » Fri Apr 30, 2021 9:10  beschrieben: https://www.esp32.com/viewtopic.php?t=16622.

Die meisten modernen Fritz!Boxen erfordern - wenn auch nicht ausschließlich - einen Double-Hitter-Ansatz, 
um eine WiFi-Verbindung herzustellen. Das heißt, man macht einen ersten Versuch, der fast immer fehlschlägt, 
dann versucht man es erneut, was "fast" immer gelingt.

Die Probelmlösung war bisher immerein Software Reset:
Nach einer Zeitüberschreitung in der Verbindungsroutine (z.B. 8 erfolglose Verbindungsversuche) wird ein "ESP.restart();" ausgeführt.
Erst dann kommt die WiFi Verbindung zu stande. 

Das sieht bei mir so aus.
while(WiFi.status() != WL_CONNECTED){
delay(500);
Serial.printf(".");
++CounterWLAN;
if(CounterWLAN > 8) {
Serial.println();
ESP.restart(); }
}

Könntet ihr das mit einbauen, um diesem bug zu umgehen?
Vielen Dank.

lumapu, 2022-12-03:
ich höre das erste mal von dem Fehler und dem 'Double-Hitter' Ansatz. Gibt es außer dem Forum Literatur dazu? Wie verhält sich die aktuelle Dev `0.5.47` (hier wurde stark im Bereich der Wifi-Verbindung umgebaut)

hoffmakl1961, 2022-12-03:
Das Problem existiert generell mit dem ESP32, aber nicht mit dem ESP8266. 
Bisher konnte ich es immer mit einen Software Reset umgehen. (Wie oben beschrieben.) 

Bei Fragen zum 'Double-Hitter' Ansatz wäre es sicherlich sinnvoll direkt 
ullixesp » Fri Apr 30, 2021 9:10 beschrieben: https://www.esp32.com/viewtopic.php?t=16622. zu kontaktieren.
Vielen Dank.

lumapu, 2022-12-03:
unter https://ahoydtu.de/web_install

stefan123t, 2022-12-04:
@lumapu aktuell muss man immer die development builds herunterladen und lokal ausgepackt mit Parameter den Chrome starten damit man die flashen kann.

Wäre es möglich das als zweite Option direkt in der Seite https://ahoydtu.de/web_install bzw /web_install_dev anzubieten ?

lumapu, 2022-12-04:
@stefan123t ich kann es probieren, aber oft gibt es da dann Probleme mit den Security Headern 'Same-Origin' usw.
Jedes Mal die dev builds auch dort hochzuladen ist mir zu aufwendig und fehleranfällig.
Evtl. geht es über die Actions, dass man es dort automatisiert aktualisiert

lumapu, 2022-12-04:
nein bei Chrome oder Edge sollte ein blauer `connect` Button erscheinen. Dort drauf klicken und die Firmware aufspielen

hoffmakl1961, 2022-12-05:
Wahrscheinlich mache ich etwas vollkommen falsch.
Wenn ich connect drücke kann ich nur INSTALL ESPHOME auswählen, aber nicht die aktuelle Firmware

Wie auch immer, ich habe jetzt mit dem "ESP32Flash download TOOL V3.9.3" geflashed. Das gibt es keine AP. 
Analog Issue #472? 

Ich kann also nicht die WLAN Daten anpassen.

dj9717, 2022-12-05:
> Wahrscheinlich mache ich etwas vollkommen falsch. Wenn ich connect drücke kann ich nur INSTALL ESPHOME auswählen, aber nicht die aktuelle Firmware
> 
> Wie auch immer, ich habe jetzt mit dem "ESP32Flash download TOOL V3.9.3" die Firmware 221203_ahoy_0.5.47_esp32_d11d313.bin geflashed. Es gibt kein AP. Analog Issue #472?
> 
> Ich kann also nicht die WLAN Daten anpassen.
> 
> PS: Nutze eigentlich lieber ein Tool auf dem PC, als ein Webtool, wo ich meine WiFi Daten eintippen muss.

Kein AP nach Upload hatte ich auch bereits, geholfen hat: "platformio run -e esp32-wroom32-release -t erase"
Anschliessend nochmal Upload und der AP funktionierte wieder..
Anderes Problem mit Workaround da Fehler noch nicht analysiert. Tritt seit 0.5.41 bei mir auf.
WA: Mosquitto Container stoppen. Dann war die DTU wieder da, MQTT erstmal raus konfiguriert.
Mosquitto hatte bisher keine Probleme bereitet wurde auch nicht verändert.
Allerdings auch hier bei mir seit 0.5.41 und auch mit 0.5.47 Verbindungsverluste zur Fritzbox Modell 7490 FW 07.29.  zb wenn des Nächtens das Fritzbox WLAN aus war.

hoffmakl1961, 2022-12-06:
@dj9717 
Kein AP nach Upload hatte ich auch bereits, geholfen hat: "platformio run -e esp32-wroom32-release -t erase"
Danke für den Tipp.
I am still old fashioned und benutze die Arduino IDE und nicht platformio.

stefan123t, 2022-12-06:
@hoffmakl1961 
ich zitiere mal den OP in dem Thread auf den sich Dein mit der FritzBox vermutetes Problem bezieht:
> It was later pointed out that the current ESP_IDF is version 4.x, while Arduino still uses 3.x.

Wir verwenden 5.2 (!) d.h. das Problem sollte eigentlich nicht auftreten bzw. schon lange gefixt sein.
Dort ist auch das eigentliche Issue mit dem MESH mode und dem `4WAY_HANDSHAKE_TIMEOUT` der Fritz!Box beschrieben und gefixt bzw. geschlossen: espressif/arduino-esp32 issue#2501

@lumapu unterstützen wir überhaupt noch die Arduino IDE ?
Ich dachte die hätte hoffnungslos veraltete Arduino ESP32 3.x Bibliotheken und wäre m.W. deprecated.

PS: @hoffmakl1961 das Webtool ist lokal auf Deinem Rechner und läuft im Browser, d.h. es ist eben in JavaScript bzw. TypeScript programmiert und wenn Du den Zugriff auf Deine lokale serielle Schnittstelle frei gibst (so wie die Kamera / Mikrofon bei einer WebConf), dann kann Dein Browser eben auch damit direkt flashen. Läuft also alles lokal bei Dir ab.
Hat den unschlagbaren Vorteil daß wir Dir die aktuelle Software direkt zum installieren anbieten können und Du nicht veraltete ESP Flasher Pakete / Tools installieren musst, nur damit es dann doch nicht geht weil irgendein Offset oder binary File nicht dazu passt. Das steht nämlich alles im `manifest.json` :grinning: Einfach F5 und er holt sich die aktuellen Version des ESP Web Tool Flashers eben nochmal vom Server der Entwickler vom ESP Web Tools Projekt und unsere manifest.json sagt ihm welche Dateien wo hin geflashed werden sollen.

PPS: Wo gibst Du denn Deine SSID Daten ein, das machst Du doch auch mit dem selben Browser oder verwendest Du dazu CURL und unsere REST API im CLI Modus :nerd_face: ?

hoffmakl1961, 2022-12-06:
@ stefan123t
Mir war nicht klar, dass das Webtool lokal auf meinem Rechner läuft. 
Es sollte also kein Problem sein, es zu verwenden. Danke. 

Als ich den Issue aufgemacht habe, habe ich das latest Release 221119_ahoy_0.5.41_esp32_dec333f.bin verwendet und mit ESP-Flasher-x86.exe hochgeladen. 
Das Firmware Relaese ist heute 17 Tage alt. 

Damit sollte es doch funktionieren? 

PS: (Meine Lib's und IDE werden gar nicht verwendet.)

https://github.com/espressif/arduino-esp32/issues/2501
Das beschreibt genau das Problem. (gefixt ist es leider nicht, oder habe ich da was überlesen. 
Ein Software reboot wäre für das Problem eine einfache Lösung.

lumapu, 2022-12-06:
> @lumapu unterstützen wir überhaupt noch die Arduino IDE ?
Ich dachte die hätte hoffnungslos veraltete Arduino ESP32 3.x Bibliotheken und wäre m.W. deprecated.

nicht dass ich wüsste, ich bin nur noch in PIO unterwegs

lumapu, 2022-12-20:
@hoffmakl1961 bitte nochmal mit der dev `0.5.58` testen, evtl. hat sich durch die letzten Änderungen etwas verbessert.

hoffmakl1961, 2022-12-20:
Guten Abend, 
das Problem existiert bei mir seit Jahren mit dem ESP32, aber nicht mit dem ESP8266.

Bei meiner Fritz!Box Umgebung (Implementierung) baut ein ESP32 nur nach jedem 2. Programmstart erfolgreich eine WiFi Verbindung zur Fritz!Box auf. Das Probelm konnte ich bisher immer mit einem Software Reset - bei erfolglosen Verbindungaufbau - lösen. 
Siehe oben, Error description. 

Ich verwende jetzt einen ESP8266, da habe ich seit 14 Tagen keine WiFi Verbindungsprobleme.

Vielen Dank für eure Hilfe.

stefan123t, 2024-08-14:
Wir haben deutliche Hinweise, daß die Probleme bei der Fritz!Box mit Repeater im sog. "Mesh-Steering" an der Unterstützung der beiden IEEE Standards **Basic Service Set (BSS) transitions 802.11k** und **Wireless Network Management 802.11v** liegen könnte. Zusätzlich gibt es noch **Fast BSS Transitions (FT) 802.11r** die alle drei beim Roaming in einem WLAN Mesh zum Einsatz kommen können und deshalb vom ESP32 unterstützt werden müssen, damit es ggf. im Mesh klappt.

https://github.com/tbnobody/OpenDTU/issues/2202#issuecomment-2289412118