lumapu/ahoy #472: WIFI - Kein AP wenn WLAN nicht vorhanden

HFarryranz, 2022-12-02:
## Hardware
  * [ x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard
* [ ] 

 

## Software
* [ x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.46
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ x] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [x ] Windows
* [ ] Mac OS

### Debugging:
* [x] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

-  ESP8266 mit blank_4MB.bin geflasht
 - ESP8266 mit DEV 0.5.46 aus github geflasht.
 - Verbunden mit AP AHOYDTU
- 192.168.4.1 geöffnet und  SSID und Password eingetragen, Jedoch gibt es die SSID (hier) nicht!! Tippfehler.
- speichern und Neustart
- Der ESP öffnet keinen AP AHOYDTU mehr, die blaue LED ist dauerhaft aus.

Lösung: ESP8266 mit blank_4MB.bin flashen und von vorne beginnen.

Konsolenausgabe ist folgende:

```
lI: Initializing FS ..
I:  .. done
I: loadDefaults
I: Settings valid: true
I: RF24 Amp Pwr: RF24_PA_I: LOW
I: Radio Config:
SPI Frequency           = 1 Mhz
                               Channel                  = 3 (~ 2403 MHz)
Model                   = nRF24L01+
RF Data Rate            = 250 KBPS
RF Power Amplifier      = PA_LOW
RF Low Noise Amplifier  = Enabled
CRC Length              = 16 bits
Address Length          = 5 bytes
Static Payload Length   = 32 bytes
Auto Retry Delay        = 250 microseconds
Auto Retry Attempts     = 0 maximum
Packets lost on
                   current channel      = 0
Retry attempts made for
                           last transmission    = 0
Multicast               = Disabled
Custom ACK Payload      = Disabled
Dynamic Payloads        = Enabled
Auto Acknowledgment     = Disabled
Primary Mode            = RX
TX address              = 0xe7e7e7e7e7
pipe 0 (closed) bound   = 0xe7e7e7e7e7
pipe 1 ( open ) bound   = 0x9423318301
pipe 2 (closed) bound   = 0xc3
pipe 3 (closed) bound   = 0xc4
pipe 4 (closed) bound   = 0xc5
pipe 5 (closed) bound   = 0xc6
connect to network 'WLANWOLKA2A' ...
...........
........................................
........................................
........................................
I:

   ----------------------------------------
I: Welcome to AHOY!
I:
   point your browser to http://192.168.4.1
I: to configure your device
I: ----------------------------------------

```

Kein AP, keine weiteren Ausgaben, tot.

PS: Könnte man in die Konsolenausgabe zu Beginn die Versionsnummer mit ausgeben?

lumapu, 2022-12-02:
danke für die detaillierte Beschreibung, ich versuche da nachzustellen und zu beheben

Gerri1, 2022-12-02:
War das nicht mal bei den älteren Versionen so, wenn das WLAN Netzwerk nicht erreichbar war, wurde der AP wieder aktiv.
Ich habe die SSID mal absichtlich falsch eingegeben und ein erneuter Flash war fällig!

lumapu, 2022-12-08:
bitte nochmal mit der neuesten Dev Version testen, sollte mW. jetzt funktionieren

commit 7c6ab679 (2022-12-03): fix #472 refactored ahoyWifi class completely
-> now Ahoy opens an AP during boot. This will be closed once a station WiFi connection is established
improved NTP after boot, will be synced immediately after successful WiFi connection
closed 2022-12-30 by lumapu