lumapu/ahoy #475: Wemos D1 with different schematic

Stratis63b, 2022-12-03:
I recently bought a wemos D1 that has different schematic ie
SCK/D5
Misc/D6
Mosi/D7
Ss/d8
D7
D6
D5. The question is do I wire on D5 or on SCK/D5 etc?

Stratis63b, 2022-12-03:
[image]

Solarteur, 2022-12-05:
Comparing the connectors from your Wemos D1R2 (?) board to those from the Wemos D1 mini, you can see that
1. the D"x" pins of both boards are connected to the same pins of the ESP8266,
2. the connections D5 and SCK/D5 of the Wemos D1R2 are connected to GPIO14 of the ESP8266 - so they are equivalent.
[sch_d1_mini_v3.0.0.pdf](https://github.com/lumapu/ahoy/files/10153540/sch_d1_mini_v3.0.0.pdf)
[wemos-d1-r2-schematic.pdf](https://github.com/lumapu/ahoy/files/10153544/wemos-d1-r2-schematic.pdf)

Sinned42, 2022-12-13:
[image]
[image]
[image]
[image]
So funktioniert es bei mir. Sorry für die diletantische Darstellung.

closed 2023-03-27 by lumapu