lumapu/ahoy #476: [ESP32]  'Invalid Header' nach Flashing

FabianCzaja, 2022-12-03:
### Platform

ESP32

### Model name

AZ-Delivery ESP-WROOM-32

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection diagram

-

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.41

### Github Hash

4c8a6da

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

Ich komme nicht zum setup

### Error description

1 - ESP32 mit externer Antenne verkabelt
2 - ESP32 an den Rechner angeschlossen
3 - Wie beschrieben den NodeMCU PyFlasher benutzt und mit der ESP32 bin geflashed
4 - Flash war erfolgreich
5 - ESP32 blinkt nach Neustart rapide rot
6 - ESP32 eröffnet keinen Access Point

NodeMCU PyFlasher Output:
[image]

PuTTY Output:
[image]

lumapu, 2022-12-03:
bitte auch mal https://ahoydtu.de/web_install/ ausprobieren. Evtl klappt es dann

FabianCzaja, 2022-12-04:
@lumapu hat geklappt, danke.

closed 2022-12-04 by lumapu