lumapu/ahoy #477: MQTT Topic on User_Manual is wrong

SeibeGitHub, 2022-12-04:
The topic "PowerLimit" in the user_manual.md should be changed to "active PowerLimt". 

Regards

Gerri1, 2022-12-09:
Mir ist was in der dev03 User Manual AhoyDTU (on ESP8266) aufgefallen.

In der Abteilung Control via MQTT unter Power Limit absolute persistent [Watts] steht:
<TOPIC>/ctrl/limit_persistent_relative/<INVERTER_ID>    müsste das nicht
<TOPIC>/ctrl/limit_persistent_absolute/<INVERTER_ID>  /  (auch Example) heißen!

Das Gleiche mit Power Limit absolute non persistent [Watts]:
<TOPIC>/ctrl/limit_nonpersistent_relative/<INVERTER_ID>    müsste das nicht
<TOPIC>/ctrl/limit_nonpersistent_absolute/<INVERTER_ID>  /  (auch Example)   heißen!

SeibeGitHub, 2022-12-09:
inverter/ctrl/limit_nonpersistent_relative/<INVERTER_ID> 
inverter/ctrl/limit_nonpersistent_absolute/<INVERTER_ID> 

ist so korrekt (das Topic inverter leitet sich aus den MQTT settings aus AHOY ab)

closed 2023-02-07 by lumapu