lumapu/ahoy #479: Inverter Type cannot be detected

uwejanssen19, 2022-12-06:
Hello,

I am using a 8266 WEMOS MINI and NRF24 with external antenna.
Ahoy software is 0.5.4.1
Inverter is Hoymiles HM-600 purchased 2021

The inverter can't be detected .
I checked various wirings, and it did not work.

But I found something strange (to me ;-) ) :

My Inverter Serial number is **1141** 720 16837 (see attached)
In the ID coding rules the correct number then would be **1041** 72016837 .

Does it matter ?

Best Regards,
Uwe Janssen

See log files:

[esp-web-tools-logs.txt](https://github.com/lumapu/ahoy/files/10166285/esp-web-tools-logs.txt)
[WEBLOG.txt](https://github.com/lumapu/ahoy/files/10166286/WEBLOG.txt)
[image]

uwejanssen19, 2022-12-06:
Yes, I am aware of that. 
Although I do not know if it is enough. Do you know how much power is required ?

[image]

Solarteur, 2022-12-06:
I don't know the exact minimum, but with an output of approx. 4W, my HR-600 can be reliably reached.

uwejanssen19, 2022-12-06:
Then it should have worked with a minimum of daylight. 
I usually wait until it produces at least 15 Watts before starting tests.
I also applied a capacitor of 100 uF at the NRF24.
Distance from NRF24 to my HM-600 is maximum 15m (through a window) 
No idea why it does not work.
Do you the range/power setting relation, just roughly?

Thank you for your help anyway

stefan123t, 2022-12-06:
Hi @uwejanssen19,

I looked at your logs and apparently the NRF24 can be found via SPI. We can not test via software whether the IRQ is connected too. There are several reports that the out-of-box pinout may not work for a Wemos D1 mini. Though we do not know the exact reason as it seems to be depending on the model / make of the Wemos D1 mini. Maybe you can supply a picture of yours.
Common workaround is to either switch GPIO0 D3 & GPIO2 D4 connected to IRQ and CE. Some have also reported that instead using GPIO5 D1 and GPIO4 D2 for these two connections gives good results too.
As you can see the green SPI lines are fixed to D5..D8 as well as the GND and +3.3V.
[image]

The PA Level ist set to `RF Power Amplifier	= PA_HIGH` for your NRF24.
The reason we recommend a good Power Supply, new cables and an Elko close to the NRF module is that the regulator for 3.3.V supply on the Wemos D1 mini boards is a little bit low spec for the current consumption of the NRF during sending.
Did you try to use a different USB Power Supply, shorter / different cable or keep it plugged into the PC ?

uwejanssen19, 2022-12-06:
Hello Stefan,
thank you very much for your info.
Power supply:
I have a wall Plug with 5V/2A and for working with the PC connection I am using a good USB 3.0 Hub with external power.
 
Connections:
 As you and others already recommended I swapped IRQ and CE. 
 In my last setup I used D2 for IRQ and D1 for CE .
Of course I changed the assignment in ahoy.
You might connect to my device if it is of any use:

[http://xdbu92gos1irkzbd.myfritz.net:8070/](url)

I'll leave that connection available .

[image]
[image]
[image]

stefan123t, 2022-12-07:
Hi Uwe,
thanks for the URL to your AhoyDTU, I have flipped some switches, i.e. 
* I added some arbitrary Max Module Power of 300 Wp given you have an HM-600.
[image]

* I also added the [x] Serial Debug option.

Unfortunately I cannot see the Boot messages where it should tell us, whether it could initialize the NRF module.
Can you upload the **USB Serial boot logs** tomorrow again ?

Could you also post a picture of your NRF model and fill in your Pinout diagram?

nRF24L01+ Pin | ESP8266 GPIO
-- | --
Pin 1 GND [*] | GND
Pin 2 +3.3V | +3.3V
Pin 3 CE | GPIO5 CE D1
Pin 4 CSN | GPIO15 CS D8
Pin 5 SCK | GPIO14 SCLK D5
Pin 6 MOSI | GPIO13 MOSI D7
Pin 7 MISO | GPIO12 MISO D6
Pin 8 IRQ | GPIO4 IRQ D2

The Wemos D1 mini you have shown should be capable of running the show, especially given that you want to substitute the Voltage regulator by an external one.
Oh but please make sure to solder the electrolytic cap directly to the NRF24 modules GND and +3.3V pins 1&2 as this is what actually draws the current while sending.
There have been suggestions that using thicker cables instead of jumper wires may improve the situation too.
Though I have mine running with an ESP8266 NodeMCU v3 board with normal 10cm Jumpers and no problems whatsoever.
But then this might have a different voltage regulator on board.

uwejanssen19, 2022-12-07:
Good Morning,

First: It still does not work 👎 
  
(Although) made changes as proposed:

- shorter cables
- cap 100uF to VCC and GND close to RF24 (applied yesterday already)
- extra power supply for RF24 (3.4 V, max. 0.3 A which never has been exceeded)

Wiring is the same you sent. I double checked it during exchanging jumper cables.

Boot log from serial:
[image]
NRF24 Picture:
[image]
Inverter Power:
[image]
Log File
[esp-web-tools-logs (1).txt](https://github.com/lumapu/ahoy/files/10174496/esp-web-tools-logs.1.txt)

uwejanssen19, 2022-12-07:
Hi Stefan,
I read [something](https://wolles-elektronikkiste.de/nrf24l01-2-4-ghz-funkmodule) about NRF24 and got at least some data from inverter:
[image]
I wrapped the board in plastic then in aluminium tape in order to shield it.
It is not yet stable at all but at least the wiring and config appear correct.

Thank you a lot for your effort,
Uwe

closed 2022-12-08 by uwejanssen19

stefan123t, 2022-12-08:
From your link:

Falls ihr ein „nRF24L01 + PA + LNA“ Modul verwendet, dann wird es auf der Transmitterseite mit der **Stromversorgung** über den 3.3 Volt Pin **eng**. Der Arduino UNO und der Nano liefern nur 50 Milliampere. Für den Testbetrieb über kurze Distanz könnt ihr die **Sendeleistung herunterregeln**, ansonsten solltet ihr eine potentere Stromquelle wählen.