lumapu/ahoy #485: keine Verbindung im Inverter

Basti77, 2022-12-09:
Hallo Zusammen.

Der Zusammenauen und das Flashen der DTU war nicht schwer.
Der Adapter wird scheinbar auch erkannt vom ESP aber irgendwas scheint das nicht zu passen. Ich bekomme immer nur die Meldung:
Inverter #0: Garage (v0) is not available and is not producing

Ich habe diverse Build Versionen ausprobiert 
Darunter die aktuelle 0.5.14. und die dev

Inverter HM-1200 für die SN bin ich eben extra noch mal aufs Dach geklettert.

Zwischen der DTU und dem Inverter sind 4m und eine Wand.
Ich weiß nicht wie sich eine nicht + Version verhält. Kann es sein das mir eine falsche Version geliefert wurde, oder wie kann ich das erkennen? 

## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: (https://de.aliexpress.com/item/32280738812.html?spm=a2g0o.order_list.order_list_main.35.39fe5c5fdi2ftH&gatewayAdapt=glo2deu)

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode ? sollte es sein
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ x] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5_.17__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ x] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ x] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Gerri1, 2022-12-09:
Siehe mal https://github.com/lumapu/ahoy/issues/473
Hat das nRF24L01 eine aufschraubbare -oder die Leiterbahnantenne?
Bei der 2. Variante müsstest du auf alle Fälle eine gute Stromversorgung (auch Kondensator) zum Modul haben und die Sendeleistung eher höher stellen! Bei meinem nRF24L01+ Modul mit ext. Antenne reicht die MIN Einstellung völlig aus!
Auch wirst du um diese Uhrzeit keine Verbindung mehr zu den WR bekommen!

Basti77, 2022-12-09:
danke für die Antwort. 
Offensichtlich ist das ganze doch nicht Plug n play.
ich lesen mich dann man tiefer ein...

stefan123t, 2022-12-10:
@Basti77 laut dem Link auf die Händler Seite handelt es sich um ein kleines Modul mit Leiterbahnantenne.
Es gibt eine Rückmeldung von einem Schweizer Käufer, dass das Modul auch mit AhoyDTU geht :grinning: 
Auf dem Chip (Foto) kann man die Bezeichnung NRF24L01+ lesen, obwohl die Beschreibung also die 250kBps nicht explizit erwähnt würde ich also mal davon ausgehen, daß das Modul in Ordnung ist.
Die kleinen Module brauchen i.d.R. ein stärkere Sendeleistung. Versuche mal ob Du mit PA HIGH Empfang bekommst. MAX ist nicht zu empfehlen, da das meist die Spannungsversorgung nicht verkraftet. Es ist auch zu empfehlen einen Elektrolyt Kondensator mit 47/100uF an die Pins 1 & 2 (GND & +3.3V) des NRF Moduls zu löten damit die Spannung nicht einbrechen kann.
Du hast das o.g. Connection Diagram nicht ausgefüllt. Bitte oben anpassen und mit der von Dir verwendeten Schaltung neu speichern. Danke! Ein Bild vom Aufbau Deiner Schaltung sagt meistens auch mehr aus.

Basti77, 2022-12-10:
Der Elko wurde QnD angelötet.
Die verdrahtungsliste wurde oben aktualisiert. 
Ich hatte initial mit der Grundkonfiguration begonnen und probiert IRQ auf benachbarte Pins zu mappen.
Das einzige was ich dabei bemerke, das die wenn die Verdrahtung nicht zur Config passt, dass das Modul dann nicht gefunden wird. 
ansonsten passiert leider nichts. :(
Der  HM-1200 Inerter arbeitet auch. 
[image]
[image]

Gerri1, 2022-12-10:
Wie hast du das verbunden?
Auf dem Bild unten ist meine Anschlussbelegung, die Standardbelegung CE --> D4 und IRQ --> D3 hatte auf keinen ESP funktioniert!

[image]

Basti77, 2022-12-10:
CE habe ich durch Deinen Hinweis auf D2 geändert.
Nun warte ich auf morgen.

stefan123t, 2022-12-11:
Hallo @Basti77 das heisst Du hast jetzt eine Konfiguration gefunden bei der Du immerhin ab und zu Daten vom WR bekommst oder das NRF Modul gefunden wird ?
Vielleicht kannst Du noch mit unterschieldichen Netzteilen für das Wemos D1 mini Board experimentieren ? Hier gab es schon Berichte dass evtl selbst Markenprodukte evtl schlechte Glättung der Spannung o.a. haben und selbst damit kein ordentlicher Betrieb möglich war. Auch die Ausrichtung horizontal und vertikal der Antenne kann grosse Auswirkung auf den Empfang haben. Aus was ist denn die Wand die zwischen Dir und dem WR ist ? Hier sind es Ziegel und Dach da geht es auf 8m mit HIGH und am PC/Laptop.

Basti77, 2022-12-11:
hi Stefan. 
Nein laut log sendet das Modul fleißig aber es kommt nie etwas zurück, egal bei welcher PIN Konfig. 
Aktuell sind nur noch 2m zwischen Modul und WR und eine Wand.
leider werde ich vor Freitag keine weiteren Erkenntnisse gewinnen (wegen Arbeit und danach keine Sonne) 
dann werde ich andere Netzteile / Powerbanks testen. 
danach fehlt eigentlich nur ein Test mit einem andern WR für ein vollständige Aussage.

stefan123t, 2022-12-12:
Andere Nutzer berichten dass es bei Ihnen nur mit D1 IRQ geht. Vielleicht nach D2 auch noch einen Versuch wert ?
Kannst Du mal ein USB Serial Log vom Booten des ESP8266 anhängen, damit sollte geklärt werden ob überhaupt eine Kommunikation mit dem NRF24 modul möglich ist und wenn ja mit welcher RF Data Rate ?
Oder habe ich das USB Serial Log das im Issue Default Template angefragt wird übersehen ?

bombastic99, 2022-12-13:
Hallo,

habe genau selbiges Problem :Inverter #0: Hoymiles 1000 (v0) is available and is not producing

Modul ist direkt neben Wehselrichter.

Basti77, 2022-12-13:
@bombastic99 
nicht ganz. du bist ein stück weiter. 
meins " is **not** available"

lumapu, 2022-12-14:
@Joe1002C was heißt diverse Pin Konfigurationen genau? Hardware und Software (Setup mit reboot) oder nur eines von beiden?

Joe1002C, 2022-12-14:
@lumapu habe zu jeder Hardwareänderung auch die entsprechende Änderung in setup vorgenommen und danach neu gestartet. So sieht der log aus:
I: enqueuedCmd: 1
I: enqueuedCmd: 5
I: add inverter: PV Garage, SN: 1xxxxxx2264, Power Limit: 100 in %
I: connect to network 'WLAN-Z2xxx ...
...........
.
I: 

----------------------------------------
I: Welcome to AHOY!
I: 
point your browser to http://192.168.2.201
I: to configure your device
I: ----------------------------------------

I: RF24 Amp Pwr: RF24_PA_I: HIGH
I: Radio Config:
SPI Frequency		= 1 Mhz
Channel			= 3 (~ 2403 MHz)
Model			= nRF24L01+
RF Data Rate		= 250 KBPS
RF Power Amplifier	= PA_HIGH
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 250 microseconds
Auto Retry Attempts	= 0 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 15
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Enabled
Auto Acknowledgment	= Disabled
Primary Mode		= RX
TX address		= 0xdeadbeef01
pipe 0 (closed) bound	= 0xdeadbeef01
pipe 1 ( open ) bound	= 0x1234567801
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6
I: [NTP]: 2022-12-14 22:09:59
I: Requesting Inverter SN 1161xxxxxx

lumapu, 2022-12-14:
könntest du im Setup noch den Haken bei System Config -> Serial Debug setzen und dann nochmal das Log posten?
Bitte beachte, das der Wechselrichter nur bei Tageslicht antwortet, wenn die PV Module Strom liefern (ca. 1W)

Joe1002C, 2022-12-14:
Ich habe einfach zu wenig Ahnung von der Materie. Das obige log ist mit den ESP Home Webtools erstellt.
Den Haken im Setup habe ich gesetzt aber ich weiss nicht, wie ich an das log komme.
Vielen Dank für deine Mühe!

lumapu, 2022-12-15:
Im Menü der Weboberfläche gibt es einen Punkt `Serial / Control`. Dort steht dann der Inhalt in den großen Fenster, das aktualisiert sich alle paar (~30s) Sekunden.

Joe1002C, 2022-12-15:
Hm, ich finde diesen Punkt nicht. Kann es sein, dass es den in meiner Version 0.5.17 nicht gibt? Oder ist die Weboberfläche immer dieselbe?

reserve85, 2022-12-15:
das gibt es da nicht, du bist mit der Version 0.5.17 ca. drei Monate zurück.

bombastic99, 2022-12-15:
Hallo,
hab nun anderes Funkmodul (longrange)  ran...selbes Problem.

Uptime: 0 Days, 00:25:53

ESP-Time: 15.12.2022, 14:37:50

WiFi RSSI: -72 dBm

Statistics:

RX success: 0
RX fail: 0
RX no answer: 7
Frames received: 0
TX cnt: 37
Inverter #0: Hoym1000 (v0) is not yet available
WARN: MQTT is not connected
Every 30 seconds the values are updated

###############

version: 0.5.41
wifi_rssi: -71
mac: xxxanaonym
sdk: 2.2.2-dev(38a443e)
cpu_freq: 80
heap_free: 12064
sketch_used: 468
core_version: 3.0.2
flash_size: 4096
heap_frag: 1
max_free_blk: 11968
reboot_reason: External System
esp_type: ESP8266

bombastic99, 2022-12-15:
...nicht das bei uns die falsche Seriennummer draufgeklebt wurde ;-)

Basti77, 2022-12-15:
`n/a15:30:20 I: [NTP]: 2022-12-15 14:30:20 UTC
15:30:33 I: enqueued cmd failed/timeout
15:30:33 I: (#0) I: no Payload received! (retransmits: 0)
15:30:33 I: resetPayload: id: 0
15:30:33 I: (#0) Requesting Inv SN 106160900697
15:30:33 I: (#0) enqueuedCmd: 11
15:30:33 I: (#0) enqueuedCmd: 1
15:30:33 I: (#0) enqueuedCmd: 5
15:30:33 I: (#0) sendTimePacket
15:30:33 I: TX 27B Ch40 | 15 60 90 06 97 81 48 09 38 80 0b 00 63 9b 2f 89 00 00 00 00 00 00 00 00 a4 c7 3a 
15:30:34 W: while retrieving data: last frame missing: Request Retransmit
15:30:34 I: (#0) sendTimePacket
15:30:34 I: TX 27B Ch61 | 15 60 90 06 97 81 48 09 38 80 0b 00 63 9b 2f 89 00 00 00 00 00 00 00 00 a4 c7 3a 
15:30:36 W: while retrieving data: last frame missing: Request Retransmit
15:30:36 I: (#0) sendTimePacket
15:30:36 I: TX 27B Ch75 | 15 60 90 06 97 81 48 09 38 80 0b 00 63 9b 2f 89 00 00 00 00 00 00 00 00 a4 c7 3a 
15:30:38 W: while retrieving data: last frame missing: Request Retransmit
15:30:38 I: (#0) sendTimePacket
15:30:38 I: TX 27B Ch3 | 15 60 90 06 97 81 48 09 38 80 0b 00 63 9b 2f 89 00 00 00 00 00 00 00 00 a4 c7 3a 
15:30:40 W: while retrieving data: last frame missing: Request Retransmit
15:30:40 I: (#0) sendTimePacket
15:30:40 I: TX 27B Ch23 | 15 60 90 06 97 81 48 09 38 80 0b 00 63 9b 2f 89 00 00 00 00 00 00 00 00 a4 c7 3a 
15:30:43 W: while retrieving data: last frame missing: Request Retransmit
15:30:43 I: (#0) sendTimePacket
15:30:43 I: TX 27B Ch40 | 15 60 90 06 97 81 48 09 38 80 0b 00 63 9b 2f 89 00 00 00 00 00 00 00 00 a4 c7 3a 
15:31:03 I: enqueued cmd failed/timeout
15:31:03 I: (#0) I: no Payload received! (retransmits: 5)
15:31:03 I: resetPayload: id: 0
15:31:03 I: (#0) Requesting Inv SN 106160900697
15:31:03 I: (#0) sendTimePacket
15:31:03 I: TX 27B Ch61 | 15 60 90 06 97 81 48 09 38 80 01 00 63 9b 2f a7 00 00 00 00 00 00 00 00 0f 18 6a 
15:31:04 W: while retrieving data: last frame missing: Request Retransmit
15:31:04 I: (#0) sendTimePacket
15:31:04 I: TX 27B Ch75 | 15 60 90 06 97 81 48 09 38 80 01 00 63 9b 2f a7 00 00 00 00 00 00 00 00 0f 18 6a 
15:31:06 W: while retrieving data: last frame missing: Request Retransmit
15:31:06 I: (#0) sendTimePacket
15:31:06 I: TX 27B Ch3 | 15 60 90 06 97 81 48 09 38 80 01 00 63 9b 2f a7 00 00 00 00 00 00 00 00 0f 18 6a 
15:31:08 W: while retrieving data: last frame missing: Request Retransmit
15:31:08 I: (#0) sendTimePacket
15:31:08 I: TX 27B Ch23 | 15 60 90 06 97 81 48 09 38 80 01 00 63 9b 2f a7 00 00 00 00 00 00 00 00 0f 18 6a 
15:31:11 W: while retrieving data: last frame missing: Request Retransmit
15:31:11 I: (#0) sendTimePacket
15:31:11 I: TX 27B Ch40 | 15 60 90 06 97 81 48 09 38 80 01 00 63 9b 2f a7 00 00 00 00 00 00 00 00 0f 18 6a 
15:31:13 W: while retrieving data: last frame missing: Request Retransmit
15:31:13 I: (#0) sendTimePacket
15:31:13 I: TX 27B Ch61 | 15 60 90 06 97 81 48 09 38 80 01 00 63 9b 2f a7 00 00 00 00 00 00 00 00 0f 18 6a 
15:31:33 I: enqueued cmd failed/timeout
15:31:33 I: (#0) I: no Payload received! (retransmits: 5)
15:31:33 I: resetPayload: id: 0
15:31:33 I: (#0) Requesting Inv SN 106160900697
15:31:33 I: (#0) sendTimePacket
15:31:33 I: TX 27B Ch75 | 15 60 90 06 97 81 48 09 38 80 05 00 63 9b 2f c5 00 00 00 00 00 00 00 00 a9 ac 1e 
15:31:34 W: while retrieving data: last frame missing: Request Retransmit
15:31:34 I: (#0) sendTimePacket
15:31:34 I: TX 27B Ch3 | 15 60 90 06 97 81 48 09 38 80 05 00 63 9b 2f c5 00 00 00 00 00 00 00 00 a9 ac 1e 
15:31:37 W: while retrieving data: last frame missing: Request Retransmit
15:31:37 I: (#0) sendTimePacket
15:31:37 I: TX 27B Ch23 | 15 60 90 06 97 81 48 09 38 80 05 00 63 9b 2f c5 00 00 00 00 00 00 00 00 a9 ac 1e 
15:31:39 W: while retrieving data: last frame missing: Request Retransmit
15:31:39 I: (#0) sendTimePacket
15:31:39 I: TX 27B Ch40 | 15 60 90 06 97 81 48 09 38 80 05 00 63 9b 2f c5 00 00 00 00 00 00 00 00 a9 ac 1e 
15:31:41 W: while retrieving data: last frame missing: Request Retransmit
15:31:41 I: (#0) sendTimePacket
15:31:41 I: TX 27B Ch61 | 15 60 90 06 97 81 48 09 38 80 05 00 63 9b 2f c5 00 00 00 00 00 00 00 00 a9 ac 1e 
15:31:43 W: while retrieving data: last frame missing: Request Retransmit
15:31:43 I: (#0) sendTimePacket
15:31:43 I: TX 27B Ch75 | 15 60 90 06 97 81 48 09 38 80 05 00 63 9b 2f c5 00 00 00 00 00 00 00 00 a9 ac 1e 
15:32:03 I: enqueued cmd failed/timeout
15:32:03 I: (#0) I: no Payload received! (retransmits: 5)
15:32:03 I: resetPayload: id: 0
[... 58 lines cut ...]
14:34:21 I: resetPayload: id: 0
14:34:21 I: (#0) Requesting Inv SN 106160900697
14:34:21 I: (#0) sendTimePacket
14:34:21 I: TX 27B Ch61 | 15 60 90 06 97 81 48 09 38 80 01 00 63 9b 30 6d 00 00 00 00 00 00 00 00 0f 07 a0 
14:34:22 W: while retrieving data: last frame missing: Request Retransmit
14:34:22 I: (#0) sendTimePacket
14:34:22 I: TX 27B Ch75 | 15 60 90 06 97 81 48 09 38 80 01 00 63 9b 30 6d 00 00 00 00 00 00 00 00 0f 07 a0 
14:34:24 W: while retrieving data: last frame missing: Request Retransmit
14:34:24 I: (#0) sendTimePacket
14:34:24 I: TX 27B Ch3 | 15 60 90 06 97 81 48 09 38 80 01 00 63 9b 30 6d 00 00 00 00 00 00 00 00 0f 07 a0 
14:34:27 W: while retrieving data: last frame missing: Request Retransmit
14:34:27 I: (#0) sendTimePacket
14:34:27 I: TX 27B Ch23 | 15 60 90 06 97 81 48 09 38 80 01 00 63 9b 30 6d 00 00 00 00 00 00 00 00 0f 07 a0 
14:34:29 W: while retrieving data: last frame missing: Request Retransmit
14:34:29 I: (#0) sendTimePacket
14:34:29 I: TX 27B Ch40 | 15 60 90 06 97 81 48 09 38 80 01 00 63 9b 30 6d 00 00 00 00 00 00 00 00 0f 07 a0 
14:34:31 W: while retrieving data: last frame missing: Request Retransmit
14:34:31 I: (#0) sendTimePacket
14:34:31 I: TX 27B Ch61 | 15 60 90 06 97 81 48 09 38 80 01 00 63 9b 30 6d 00 00 00 00 00 00 00 00 0f 07 a0 
`

ich habe es heute etwas früher nach Hause geschafft. 

wenn man Commands auf Restart drückt kommt es zu einem Speicher Überlauf  im log den text konnte ich nicht kopieren weil es nur noch am scrollen war. 
[image]

Joe1002C, 2022-12-15:
Bei mir hat es jetzt geklappt. Ich habe heute ein anderes NRF24L01+ bekommen und angeschlossen. Hat dann auf anhieb funktioniert. Dann noch ein Update auf die aktuelle Version gemacht und siehe da, hat auch geklappt. Das NRF24L01+ was ich zuerst dran hatte war eines mit externer Antenne, das aktuelle ist ohne externe Antenne. Habe  ca. 10n Abstand zum WR und 2 Wände dazwischen. Scheint aber stabil zu laufen.
Vielen Dank für eure Mühe!

Basti77, 2022-12-16:
@Joe1002C 
Hast Du eine Link zu dem Modul welches bei Dir funktioniert?

Joe1002C, 2022-12-16:
@Basti77 
Das Modul ist vom Makershop über Ebay.

stefan123t, 2022-12-19:
@Basti77 Du hast m.E. einen von AhoyDTU aktuell (noch) nicht unterstützten Wechselrichter der Gen2: MI-XXX
Da wirst Du vermutlich eher mit dem Fork von Ziyat T. zu einem Ergebnis kommen.
> 15:30:33 I: (#0) Requesting Inv SN 106160900697

@lumapu ich glaube wir können an dieser Stelle das Issue zu machen, da wir aktuell keine MI- Wechselrichter in Ahoy unterstützen. 
Eventuell sollten wir in unserem Issue Template die Frage nach den ersten fünf Stellen der Serial Number stellen, damit wir solche Schwierigkeiten in Zukunft frühzeitig erkennen können ?

stefan123t, 2022-12-19:
Siehe Link zu https://github.com/Ziyatoe/DTUsimMI-Hoymiles in #496.

closed 2022-12-19 by stefan123t