lumapu/ahoy #486: [ESP8266] Wrong temperature values

ps915, 2022-12-10:
### Platform

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.17

### Github Hash

g5402e9b

### Build & Flash Method

Platform IO (build & flash)

### Desktop

Windows

### Setup

[image]

### Error description

After it got cold in germany, my AHOY seems to send some strange temperature values. It happens from time to time, and works normally again randomly (no reboot required)

See attached screenshot:

[image]

[image]

Its seems, that the value of the **Active Power Limit** will overwrite the temperature reading of the inverter. When i delete the **Active Power Limit** and hit save, the same value will appear again.

Gerri1, 2022-12-10:
Nimm mal die 0.5.28 zum Testen!

knickohr, 2022-12-10:
Ich würde wohl eher sagen, irgendwas kann meine Minustemperaturen handeln. Ob’s jetzt die Anzeige auf der Seite oder generell ein Variablenproblem ist, muß noch geklärt werden.

Bei mir ist es noch nicht so kalt. Aber warten wir mal bis Montag.

Nutzt Du MQTT ?  Was kommen da für Werte ?

Gerri1, 2022-12-10:
Bis jetzt läuft meine Temperaturerfassung ohne Probleme (0.5.28)! 

[image]

ps915, 2022-12-10:
Ich logge auch alles per MQTT -> FHEM -> MySQL -> Grafana. Die falschen Werte werden von MQTT bis Grafana so durchgereicht. Ich habe es nur bemerkt, da mein Verlaufsdiagramm so komisch aussah, da 45 Datensätze mit einer Temperatur von 6553°C gespeichert wurden. 

Ich tippe auch auch Minusgrade. Als es kälter wurde, fing es auch an mit den falschen Werten. Vorher lief es viele Monate fehlerfrei. 

Mein AHOY PCB ist allerdings keinen kalten Außentemperaturen ausgesetzt, nur die beiden Wechselrichter.

Gerri1, 2022-12-10:
Flash einfach mal die 0.5.28, bei waren schon minus Grade!

knickohr, 2022-12-10:
Warum nicht gleich die 41 ?

ps915, 2022-12-10:
habe die 41 geflasht und werde berichten, ob der Fehler nochmal auftritt

mroenne2022, 2022-12-11:
Hier sind heute morgen Minus Grade , meine DTU (D1 Wemos Mini) mit 0.5.17 zeigt ebenfalls 6553 Grad an. Eine andere DTU zum Test angeschlossen (ESP32 WROOM mit 30PIN) mit der FW 0.5.41 zeigt heute garnicht mehr die Temp. an !? Anscheinend gibt es tatsächlich Probleme mit den Minus Temp.
[image]

mroenne2022, 2022-12-11:
Inzwischen knapp die Plus Grade erreicht im WR , schon ist die Temp. Anzeige wieder mit dabei !
[image]

stefan123t, 2022-12-11:
@lumapu Temperatur Temp scheint als unsigned int mit fix Komma interpretiert zu werden ist aber offensichtlich ein signed int mit fix Komma. Bitte ggf korrigieren. Danke !

knickohr, 2022-12-11:
Ich kann leider heute nicht mitreden, meine DTU ist heute Nacht ausgestiegen und ich habe es erst jetzt bemerkt 😲

knickohr, 2022-12-11:
Trotzdem isses ein Bug 🐞

Gerri1, 2022-12-11:
Bis heute hat die Temperaturerfassung ohne Probleme funktioniert. Es waren zwar nur -1,8C, aber es wurde richtig ausgelesen!

[image]

homeautomation2022, 2022-12-11:
Wie viel hier geschrieben wird zu einem Bug, der schon seit geraumer Zeit behoben ist, nur das hier anscheinend einige nicht regelmäßig die neueste FW updaten...
https://github.com/lumapu/ahoy/blob/59b8baab418bcd1dbe04efd2f8292732f1ac3711/src/hm/hmInverter.h#L243

Werte die exakt 0 sind zeigt das Ahoy Webif nicht an, nur so nebenbei zur Info.

stefan123t, 2022-12-12:
Fixed & working in v0.5.41