lumapu/ahoy #488: Ahoy-DTU in Home Assistant einbinden

feindbilder, 2022-12-11:
Hallo,

zunächst einmal vielen Dank für Eure großartige Arbeit!
Ich bin neu in dem Thema ESP und Home Assistant. Ich habe eine Balkonalage mit HM-600 und bin auf diese alternative DTU gestoßen und habe daher eine gebaut und Home Assistant aufgesetzt. Der Datentransfer via MQTT läuft, aber leider bin ich bislang nihct in der Lage, ein Gerät oder ein Entity zu finden.
Googlen half mir bislang auch nicht. Ich habe das Gefühl, dass ich ESPHome Integration in Home Assistant benötige, leider hat mir der Versuch meine lauffähige Installation gelöscht (Anfängerfhler:-) ).
Könnt Ihr mir sagen, wie ich weiter vorgehen muss?

Vielen Dank im Voraus
Rainer

## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.41
Github Hash: dec333f

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO
* [x] online Installer von ESPHome 

### Desktop OS:
* [x] Linux
* [x] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

DukeWolle, 2022-12-11:
Komm einfach in den discord chat. Da wird dir geholfen. Da findest über die Suche auch vieles, da es da schon oft zum Thema stand .

stefan123t, 2022-12-13:
@feindbilder hast Du im MQTT topic bzw den Inverter- / Modulnamen evtl Leerzeichen ? Das sorgt evtl für Probleme beim Auto Discorvery in Home Assistant.

feindbilder, 2022-12-13:
Hallo zusammen,
vielen Dank für Eure Antworten!
Ich habe es mittlerweile zum Laufen gebracht, indem ich die Hardware mit Ahoydtu neu geflashed habe und alle Konfigurationen neu vorgenommen habe. Obwohl ich mir sicher bin, alles genau so gemacht zu haben, wie vorher auch, tauchte nun ein Gerät unterhalb von MQTT auf.
Es waren und sind keine Leerzeichen im Topic, ich habe auch beim ersten Versuch eine laufende MQTT Einbindung in Mosquitto gehabt (ich konnte die Nachrichten sehen) und ich hatte auch in den Settiings auf "Discovery Config" geklickt.

closed 2022-12-13 by feindbilder

lumapu, 2022-12-13:
> @lumapu
Vorschlag von maddash war alle Leerzeichen im Topic durch _ underscore zu ersetzen bzw mein Vorschlag wäre Leerzeichen in den Input Feldern gar nicht erst zu erlauben:

> ZB
<input type="text" pattern="[a-zA-Z0-9_]{3,25}">

cool was HTML inzwischen alles schon selbst kann