lumapu/ahoy #489: keine Daten mehr über mqtt

DennisXK, 2022-12-11:
Zeit ich auf die 5.28 gewechselt bin bekomme ich keine daten mehr in meinen IObroker. Habe auch schon die 5.40 versucht, aber mit der Version bekomme ich erst gar keine Verbindung über den Browser hin....(das ist aber ein anderes Thema)

Für was ist der Button "Discovery Config (homeassistant)" da? Muss man den drücken, dass er die Daten sendet? homeassistant möchte ich eig. nicht wieder in meinem IObroker aufgeführt bekommen...

## Hardware
  * [x ] ESP8266

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* [ x] circuit board

### Power Stabilization:
* [x ]  100uF Electrolytic Capacitor 

## Software
* [x ] AhoyDTU

### Version / Git SHA: 
Version: 0.5.28

### Build & Flash Method:
* [x] ESP Tools

### Desktop OS:
* [ x] Windows

Gerri1, 2022-12-12:
Ich gehe davon aus, dass die Einstellungen in der Ahoy-DTU und ioBroker richtig sind und auch der MQTT- Adapter einmal neu gestartet wurde!
Wenn das alles dennoch passen sollte, würde ich die FW über USB neu Flashen, mit Erase !

stefan123t, 2022-12-12:
@DennisXK  der Button "Discovery Config (homeassistant)" ist genau dafür da: Auto-Discovery der Topics in Home Assistant. Da Du das nicht hast und nicht möchtest solltest Du da auch nicht draufdrücken, sonst gibt es die Struktur eben wieder :smile:

DennisXK, 2022-12-13:
Ja, die Einstellungen passen und der Adapter wurde auch neu gestartet. In der dtu Übersicht steht auch das er mit mqtt verbunden ist.

lumapu, 2022-12-17:
bei mir klappt es seit Anfang an mit ioBroker, hier meine mqtt Instanz. Evtl. mal versuchen die Instanz neu zu starten.

[image]

man kann ganz gut mit MQTTX die Daten prüfen: https://mqttx.app/

mein ioBroker läuft auf `10.20.1.1`, hier meine MQTTX Einstellungen
[image]

dann connecten und noch das Topic des Inverters subscriben:
[image]

DennisXK, 2022-12-17:
Bei mir hat mqqt auch tiptop von Anfang an funktioniert, bis ich auf 5.28 upgedatet habe🤷‍♂️.

lumapu, 2022-12-17:
kannst gerne auch die aktuelle Development Version testen:
https://ahoydtu.de/web_install/

gerade sind wir bei 0.5.55

DennisXK, 2022-12-18:
@lumapu habe mqqtx mal ausprobiert. Kommen leider keine Daten rein. ich werde jetzt mal den D1 mini erasen und neu flashen. evtl gehts dann wieder.

DennisXK, 2022-12-18:
So habe jetzt die  D1 neu geflasht mit der 5.41. jetzt kommen die Dten im iobroker wieder an. Aber warum legt er eine neune ordner mit mqqt/0/PV-Anlage an? Nur um die Uptime abzulgen? Unter dem eig. mqqt Pfad wird doch schon ein uptime angelegt! leider aber mit dem Wert 0.

[image]

lumapu, 2022-12-18:
bitte die letzte Änderung der MQTT Daten in ioBroker einblenden, nicht dass man hier alte Daten von vorherigen Versuchen sieht.

DennisXK, 2022-12-18:
Wenn ich den Ordner mqqt unter dem Pfad mqqt/0/ lösche wird er wieder erstellen beim erhalten von neuen Daten🤷‍♂️

DennisXK, 2022-12-21:
Der IoBroker bringt ebenfalls Meldungen das etwas nicht stimmt!

[image]

lumapu, 2022-12-21:
wenn ich es richtig sehe ist ein Leerzeichen vor 'PV-Anlage', das wird durch ein '_' ersetzt

DennisXK, 2022-12-22:
Ok, leider ist keine Leerzeichen in den Topic Einstellungen vor dem "PV-Anlage" eingetragen. Finde es extrem merkwürdig, dass er den Ordner HM-600_1 und den mqqt Ordner anlegt.

lumapu, 2022-12-22:
der Ordner `mqtt.0` ist normal. Danach sollte dein in der Einstellungen gesetztes Topic kommen. In der nächsten Ebene kommen die Namen deiner Wechselrichter. 
Hinweis: Jeder Punkt im Topic oder Namen führt zu einer weiteren Ebene

Gerri1, 2022-12-22:
> Aber warum legt er eine neune ordner mit mqqt/0/PV-Anlage an? Nur um die Uptime abzulgen? Unter dem eig. mqqt Pfad wird doch schon ein uptime angelegt! leider aber mit dem Wert 0.
Diesen Fehler hatte ich auch schön des öfteren!

Ich hatte den ESP Stromlos gemacht und im ioBroker den kompletten neu erstellten Baum ab MQTT gelöscht. Danach den richtigen Baum komplett zu machen und den Fokus nicht auf MQTT setzen und MQTT neu starten.
Danach den ESP wieder einschalten uns es sollte der Baum wieder richtig angelegt werden!

DennisXK, 2023-01-02:
@Gerri1 Habe ich gemacht. hat auch zum Erfolg geführt. Aber Irgendwie legt er mir immer noch einen Ordner an. 

Mit der ältern Version hatte ich keine Topic "PV-Anlage" aber diesen Ordner HM-600_1 wird mir immer nach dem ich ihn gelöscht habe am nächsten Tag wieder erzeugt.
[image]

Gerri1, 2023-01-02:
@DennisXK 
Hast du nach dem Ausschalten der Ahoy-DTU und dem Löschen des Ordners HM... den MQTT Adapter im ioBroker auch mal neu gestartet? Evtl. dein ioBroker auch mal neu Durchstarten!

DennisXK, 2023-01-02:
@Gerri1 Ja klar erst den Adapter pausiert, Verzeichnisse gelöscht, Ahoy stromlos gemacht, Adapter neu gestartet, Ahoy mit Strom versorgt!

Gerri1, 2023-01-02:
Ist der HM... Ordner auch unter dem Ordner PV-Anlage vorhanden?
Am besten den auch nochmal "so" löschen!

DennisXK, 2023-01-02:
Ja unter dem Ordner PV-Anlage ist alles richtig jetzt mit deinem tipp! 
Den Ordner HM-600_1 unter dem Ordner mqtt/0 hatte ich zur mit gelöscht gehabt. kommt aber wie gesagt immer wieder.

Gerri1, 2023-01-02:
Hast du wirklich beide Ordner mal gelöscht?!

[image]

closed 2023-03-08 by lumapu