lumapu/ahoy #492: Can not connect to HM600 Inverter

matzelkoenig, 2022-12-13:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: different Wemos D1 Mini devices tryed out
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [x] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [x] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.51 and before
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [x] Platform IO

### Desktop OS:
* [ ] Linux
* [x] Windows
* [ ] Mac OS

### Debugging:
* [x] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Ich habe versucht Ahoy zum laufen zu bekommen, aber kann keine Verbidung zu meinem Inverter herstellen. Habe verschiedene IO Ports für IRQ und CE versucht. Alles ohne Erfolg. Hilfe....

I:  .. done
I: Settings valid: true
I: RF24 Amp Pwr: RF24_PA_I: HIGH
I: Radio Config:
SPI Frequency		= 1 Mhz
Channel			= 3 (~ 2403 MHz)
Model			= nRF24L01+
RF Data Rate		= 250 KBPS
RF Power Amplifier	= PA_HIGH
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 250 microseconds
Auto Retry Attempts	= 0 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 15
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Enabled
Auto Acknowledgment	= Disabled
Primary Mode		= RX
TX address		= 0xdeadbeef01
pipe 0 (closed) bound	= 0xdeadbeef01
pipe 1 ( open ) bound	= 0x2624168601
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6
connect to network 'Aqxxxxxxxx' ...
I: added inverter 11418316xxxx

[WiFi] Connected
[WiFi] AP disabled

--------------------------------
Welcome to AHOY!

point your browser to http://192.168.178.140 (Station)
to configure your device
--------------------------------

I: [NTP]: getNtpTime failed
I: Inverter: 0
I: 
I: [NTP]: 2022-12-13 11:37:24 UTC
I: MQTT connected
I: enqueued cmd failed/timeout
I: (#0) I: no Payload received! (retransmits: 0)
I: resetPayload: id: 0
I: (#0) Requesting Inv SN 11418316xxxx (SN muss nicht jeder sehen ;)
I: (#0) enqueuedCmd: 1
I: (#0) enqueuedCmd: 11
I: (#0) enqueuedCmd: 5
I: (#0) sendTimePacket
I: TX 27B Ch40 | 15 83 16 67 30 86 16 24 26 80 01 00 63 98 64 0E 00 00 00 00 00 00 00 00 2D F6 8E 
W: while retrieving data: last frame missing: Request Retransmit
I: (#0) sendTimePacket
I: TX 27B Ch61 | 15 83 16 67 30 86 16 24 26 80 01 00 63 98 64 0E 00 00 00 00 00 00 00 00 2D F6 8E 
W: while retrieving data: last frame missing: Request Retransmit
I: (#0) sendTimePacket
I: TX 27B Ch75 | 15 83 16 67 30 86 16 24 26 80 01 00 63 98 64 0E 00 00 00 00 00 00 00 00 2D F6 8E 
W: while retrieving data: last frame missing: Request Retransmit
I: (#0) sendTimePacket
I: TX 27B Ch3 | 15 83 16 67 30 86 16 24 26 80 01 00 63 98 64 0E 00 00 00 00 00 00 00 00 2D F6 8E 
W: while retrieving data: last frame missing: Request Retransmit
I: (#0) sendTimePacket
I: TX 27B Ch23 | 15 83 16 67 30 86 16 24 26 80 01 00 63 98 64 0E 00 00 00 00 00 00 00 00 2D F6 8E 
W: while retrieving data: last frame missing: Request Retransmit
I: (#0) sendTimePacket
I: TX 27B Ch40 | 15 83 16 67 30 86 16 24 26 80 01 00 63 98 64 0E 00 00 00 00 00 00 00 00 2D F6 8E 
I: enqueued cmd failed/timeout
I: (#0) I: no Payload received! (retransmits: 5)
I: resetPayload: id: 0
I: (#0) Requesting Inv SN 114183166730
I: (#0) sendTimePacket
I: TX 27B Ch61 | 15 83 16 67 30 86 16 24 26 80 0B 00 63 98 64 2B 00 00 00 00 00 00 00 00 B6 5A 96 
W: while retrieving data: last frame missing: Request Retransmit
I: (#0) sendTimePacket
I: TX 27B Ch75 | 15 83 16 67 30 86 16 24 26 80 0B 00 63 98 64 2B 00 00 00 00 00 00 00 00 B6 5A 96 
W: while retrieving data: last frame missing: Request Retransmit
I: (#0) sendTimePacket
I: TX 27B Ch3 | 15 83 16 67 30 86 16 24 26 80 0B 00 63 98 64 2B 00 00 00 00 00 00 00 00 B6 5A 96 
W: while retrieving data: last frame missing: Request Retransmit
I: (#0) sendTimePacket
I: TX 27B Ch23 | 15 83 16 67 30 86 16 24 26 80 0B 00 63 98 64 2B 00 00 00 00 00 00 00 00 B6 5A 96 
W: while retrieving data: last frame missing: Request Retransmit
I: (#0) sendTimePacket
I: TX 27B Ch40 | 15 83 16 67 30 86 16 24 26 80 0B 00 63 98 64 2B 00 00 00 00 00 00 00 00 B6 5A 96 
W: while retrieving data: last frame missing: Request Retransmit
I: (#0) sendTimePacket
I: TX 27B Ch61 | 15 83 16 67 30 86 16 24 26 80 0B 00 63 98 64 2B 00 00 00 00 00 00 00 00 B6 5A 96

mroenne2022, 2022-12-13:
Hallo @matzelkoenig , ist dein NRF Modul erkannt worden, also zeigt er diesbezgl. keine Fehlermeldung in dem Bereich an (siehe Screenshot) und holt er sich die aktuele Uhrzeit ?  PS: die SN ist an anderer Stelle kpl. sichtbar ;-) 
[image]

matzelkoenig, 2022-12-13:
_SN sichtbar_ - Oha, das dachte ich mir schon :D

NRF Modul wird erkannt und sendet wohl auch: 
[image]

mroenne2022, 2022-12-13:
Die SN wirst du ja sicherlich mehrfach kontrolliert haben. Laufen müsste der WR ja noch , ist ja noch etwas "Licht". Blinkt die grüne LED am WR (falls du das sehen kannst) ? Ich fürchte ansonsten fällt mir nichts ein.

matzelkoenig, 2022-12-13:
In der Tat habe ich alles mehrfach kontrolliert und auch neu verdrahtet. Der WR läuft (blinkt grün) und die WLAN Steckdose meldet aktuell ca. 20W. Erstmal Danke für die schnelle Antwort! Ich werde am WE weiter versuchen das zum Laufen zu bekommen.

mroenne2022, 2022-12-13:
Viel Erfolg wünsch ich ! Ein paarmal hatte ich den Fall das er das NRF Modul nicht erkannt hatte und/oder ich musste im Setup (+Verkabelung) bei den 3 Ports IRQ...usw Anpassungen vornehmen. Aber dann gabs auch immer eine Fehlermeldung dazu.

mroenne2022, 2022-12-13:
Kann natürlich noch an der Entfernung vom Ahoy zum WR liegen oder an der Sendeleistung des NRF , kann man auch mal versuchen zu ändern (High/Low!) , das hab ich auch schon mehrfach gelesen das es daran lag.

matzelkoenig, 2022-12-13:
Ich bin mit dem Teil mal direkt neben den WR nach draußen gegangen. Leider auch keine Verbindung. Dann habe ich es in der Nähe, aber in der Wohnung platziert. Nach einiger Zeit hat er dann tatsächlich mal ein Datenpaket empfangen! Aber das ist nicht stabil:
[image]

mroenne2022, 2022-12-13:
OK , vielleicht mal auf LOW stellen die Sendeleistung ? Meine steht auf LOW , bin ca 5-6m entfernt.

matzelkoenig, 2022-12-13:
Die Leistung die in der Übersicht angezeigt wird (Screenshot), ist die vom WLAN, oder? Mit dem WiFi habe ich keine Probleme. Netzteil habe ich verschiedene probiert. Kein Unterschied :( Probiere nochmal LOW. Mal schauen.

mroenne2022, 2022-12-13:
Oh ja - sorry , das ist die WiFi Leistung. Aber mit High/Low kann man mal testen.

matzelkoenig, 2022-12-14:
Inzwischen läuft es etwas stabiler. Ca. 1/3 der Requests gehen durch. Habe den 100uF Elko gegen einen 10uF getauscht und nochmal ein anderes USB Netzteil drangehängt. Mit LOW ging allerdings nichts. Aktuell steht es auf HIGH. Subjektiv hat wohl auch die Reduktion der Retries von fünf auf zwei Besserung gebracht. Alles in Allem scheint es allgemein nicht so weit her zu sein mit der Stabilität der NRF Module (Fertigungstoleranzen, Spannungsversorgung?). Wenn man googelt findet man viele Seiten auf denen dazu Probleme beschrieben werden.
Aktuell kann ich halbwegs damit leben. Werde demnächst ein neues NRF besorgen und es damit nochmal probieren.

mroenne2022, 2022-12-14:
@matzelkoenig : Ich hab mir am Anfang mal eine Ahoy DTU von jemand anderen (mit externener Antenne + esp8266) , mit dem hab ich bis heute ewig connect Probleme. Mittlerweile hab ich mir mehrere AhoyDTU mit esp8266 und auch mit ESP32 WROOM gebaut , und auch 2 openDTU mit ESP32 gebaut(einfach weil ich Lust hatte und zum testen tlw. mit versch. FW :-) ) . Alle selbst gebauten laufen prima und alle hab ich mit den NRF's ohne ext Antenne gebaut (die auf dem Bild von Ama.....n). Allerdings hab ich nur wenige Meter zwischen DTU und WR. 2 davon mit Elko , die anderen ohne, laufen alle gut.
[image]

matzelkoenig, 2022-12-17:
@mroenne2022 Ich habe mir jetzt diese NRF's von AZDelivery besorgt. Angeschlossen. Läuft! Ohne Elko! Ab und an geht mal ein Datenpaket verloren. Lasse es aber auf LOW laufen. Wahrscheinlich wird es noch stabiler, wenn ich es hochdrehe. 
[image]

mroenne2022, 2022-12-17:
@matzelkoenig : Super,  dann hast du das Problem beseitigt. Wenn ab und zu ein Paket flöten geht ist es ja nicht schlimm. Vollkommen ausreichend, ich hab sowieso auf 60sec gedreht, reicht völlig.

lumapu, 2022-12-18:
sehr schön zu hören. Zusammenfassend kann man sagen, dass die Hardware schuld war am Problem? Können wir hier noch irgendwas für die Zukunft in die Doku übernehmen, damit andere nicht das gleiche Problem haben?
Bitte den Issue schließen wenn alles abgearbeitet wurde.

mroenne2022, 2022-12-18:
@matzelkoenig : Ich denke du kannst bestätigen das es sehr warscheinlich an der HW (NRF) lag !? Bei mir laufen die DTU's mit den NRF's ohne ext. Antenne einfach besser,  ich hab aber auch nur wenige Meter und eine Aussenwand zum WR.
@lumapu : Und übrigens....Vielen Dank für diese klasse Arbeit, dieses tolle Projekt. Ihr seid echt super 👍👌

matzelkoenig, 2022-12-18:
Dem möchte ich mich anschließen: Klasse Projekt, sehr gut umgesetzt und sehr einfach zu nutzen - wenn die HW mitspielt 😂.
Für die Doku würde ich vorschlagen, darauf hinzuweisen, dass die NRF Hardware generell eine Fehlerquelle sein kann und man im Fall des Falles mehrere NRFs durchzuprobieren.

closed 2022-12-18 by matzelkoenig

stefan123t, 2022-12-25:
@matzelkoenig bitte noch ein Bild von dem NRF Modul anhängen das nicht funktioniert hat. Am besten so gut dass man die Beschriftung des NRF auch gut erkennen kann. Danke!