lumapu/ahoy #493: [ESP8266] MQTT Bug ...

DukeWolle, 2022-12-13:
### Platform

ESP8266

### Model name

D1 Mini  ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

board

### Connection diagram

Default

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.51

### Github Hash

1bba068

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

Inverter 0

Serial Number (12 digits)*
11418xxxxx
Name*
HM-600_PV1
Max Module Power (Wp)
385
385
Module Name
Modul 1
Modul 2

### Error description

Der available Status wird per MQTT bei Sunset nicht korrekt übermittelt. Ich habe den Nachtmodus bewusst abgeschaltet, da er die Tage zuvor (mit Nachtmodus) immer abgewürgt hat. Tests unsererseits sind in #support-ahoy auf dem discord nachzulesen. 

Fazit: 
Sunset:  16:26:02
Um ca. 16:40 Uhr wurde das letzte Packet übermittelt. Der Status in MQTT erhielt kein Update mehr  !!!

RX success: 3534
RX fail: 0
RX no answer: 3811
Frames received: 17807
TX cnt: 23349
Inverter #0: HM-600_PV1 (v10010) is not yet available
-> last successful transmission: 13.12.2022, 16:39:52
INFO: MQTT is connected, 112840 packets sent

Der Status in MQTT blieb auf available = 1, available_text = available and not producing hängen. 

Unserer Meinung nach sollte hier der Status bei Änderung des Zustands auf jedenfall nochmal getriggert werden. Evtl. könnte man sogar die AC/DC Werte noch nullen.

Gruß DukeWolle

knickohr, 2022-12-13:
Ouhhhh, das mit dem Nullen gibt wieder heftige Diskussion 😅

DukeWolle, 2022-12-13:
Naja, wenn nichts mehr produziert wird können dies Werte doch auf 0 gezogen werden. Werte wie YieldDay und YieldTotal sollen ja erhalten bleiben.

knickohr, 2022-12-13:
Dann les dir mal #416 durch 🙉🙈🙊

DukeWolle, 2022-12-13:
Hab ich schon. Aber für mich ist das eigentlich recht einfach. Wenn ein Gerät ausgeschalten ist, dann liefert es auch nichts mehr.

SandnerSoft, 2022-12-14:
als BUG kann noch hinzugefügt werden, das ab Version 0.5.47 keine TOTAL Werte mehr gesendet werden. Gerade bei mehreren WR sehr ärgerlich.

knickohr, 2022-12-14:
Das ist nicht korrekt. Es werde. Keine NEUEN Werte mehr gesendet. Da diese Werte aber retained sind, hält sie der Broker vor bis neue Werte kommen.

wenn Du natürlich deinen Broker nach Sonnenuntergang neu startest, sind keine Werte vorerst mal da.

Das sind die retained Values:

[image]

lumapu, 2022-12-14:
@knickohr danke für die Erklärung, sonst hätte ich debuggen müssen

SandnerSoft, 2022-12-14:
trotzdem wird total/YieldTotal und total/YieldDay nicht übermittelt!!!
das kommt dann irgendwann mal am Tag, da steht auf dem Display schon ein ganz anderer Wert

lumapu, 2022-12-14:
es gibt eine Logik, die prüft für die Totalwerte, ob alle inverter 'available' sind. Erst wenn diese Bedingung wahr ist, werden die Werte geschickt.

knickohr, 2022-12-14:
> 

Welches Display ?  Das Nokia oder das OLED ?  Wenn dem so ist, dann wird immer nur alle aktiven Innverter angezeigt. Hast Du mehrere Inverter laufen ?

lumapu, 2022-12-14:
laut ursprünglicher Beschreibung gibt es nur einen Inverter.
@SandnerSoft Die Totalwerte werden nur bei mehreren angeschlossenen Invertern gesendet

SandnerSoft, 2022-12-14:
ich habe 3 HM-1500. Alle sind erreichbar und am OLED und in der /live werden alle Daten richtig angezeigt. Ab Version > 0.5.47 kommen aber die Total Werte nicht per MQTT!
Aktuell werden die Daten von vor 4 Tagen angezeigt per Retain.

Ich rechne dann wieder alle WR selbst zusammen, so war es ja am Anfang auch.

knickohr, 2022-12-14:
Moment, um das nochmal deutlich klar zu stellen.

Wir reden unter Tags, wenn die Inverter alle arbeiten, nicht Nachts ?

Dann kann ich das Problem nich nachvollziehen 😲 Bei mir sind die total-Daten alle immer aktuell. Nachts wird halt der letzte Wert angezeigt.

Ich habe 2 Inverter dran hängen, vielleicht tritt der Bug ab/bei 3 Invertern auf ?  Kannst Du mal die Werte direkt am Broker abfragen und nicht über Dritt-Tools ?

SandnerSoft, 2022-12-14:
Ja alles am Tage.

Ich teste morgen nochmal mit MQTT-Explorer und berichte dann

SandnerSoft, 2022-12-15:
[image]

So hier sieht man, das keine Daten im Channel total rein kommen. Achtet auf die Timestamps.
Aktueller yieldDay = 335Wh und Power = 535W

knickohr, 2022-12-15:
Ich kann’s leider nicht nachvollziehen. Zumindest nicht mit dem MQTT-Broker Mosquitto. Welchen Broker verwendest Du ?  Bei mir kommen die Daten aktuell.

Aber zurück zum eigenen Bug in diesem Issue @lumapu .  Mit der 52 kommen jetzt die Stati sauber 😎👍

SandnerSoft, 2022-12-17:
Nachtrag: der Fehler tritt nur bei der Displayversion auf!
Hier werden die Werte im Display korrekt angezeigt, aber über MQTT nicht mehr.
Nehme ich die Version ohne Display, läuft alles

lumapu, 2022-12-17:
danke für den Hinweis, das ist interessant.

commit ea6a9c15 (2022-12-17): changed sunrise / sunset calculation, angle is now `-3.5` instead of original `-0.83`
improved scheduler (removed -1 from `reload`) #483
improved reboot flag in `app.h`
fixed #493 no MQTT payload once display is defined
closed 2022-12-30 by lumapu