lumapu/ahoy #496: No connection to Hoymiles MI700 inverter

Agressor110, 2022-12-15:
Hi, i am not able to receive any data from my Hoymiles MI700 inverter.
The receiver modul - inverter distance is only 50cm.
Connection to the Receiver is perfekt, and connected to my privat Wifi network.
I use the standard Ahoy Firmware version: 0.5.41.

Any idea whats wrong?
Best regards
Agressor110

## Hardware
  * [ x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ x] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: version: 0.5.41
wifi_rssi: -90
mac: 50:02:91:B3:10:EA
sdk: 2.2.2-dev(38a443e)
cpu_freq: 80
heap_free: 11208
sketch_used: 468
core_version: 3.0.2
flash_size: 4096
heap_frag: 2
max_free_blk: 11048
reboot_reason: External System
esp_type: ESP8266
[Factory Reset](http://ahoy-dtu/factory)_._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [x ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ x] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

lumapu, 2022-12-15:
MI- Series are not supported for now. There are plans to include them in future, but not done yet.

Agressor110, 2022-12-16:
Hi, thank you for your feedback.
But I already find some infos at the project description that several MI inverter supported as well. For example the Mi 600 - and insider info is also that the Mi600 and the Mi700 use the same hardware.
Is this information correct?
Best regards 
Agressor110

DanielR92, 2022-12-19:
For MI go to the other solution please: 
https://github.com/Ziyatoe/DTUsimMI-Hoymiles

    MI1500, 3 and 4 PV's.
    MI600
    MI300
    TSUN800 (as MI600), 2 PV's
    should work with MI1200 also

Please close when that helps you. :)

Agressor110, 2022-12-19:
Thank for your feedback.My problem is inside the other project ( your propose) I did not find a flashable firmware?!Do you have a direct link to the MI inverter flash software ( esp8266) ?Thank you in advance.Best regards Agressor110

https://github.com/Ziyatoe/DTUsimMI-Hoymiles

stefan123t, 2022-12-19:
@Agressor110 Ziyat T.'s Repo does only provide the source code and has been forked from the old Arduino based code of Ahoy DTU somewhere in Summer 2022. So you will have to build the binaries yourself or ask Ziyat or Rejoe2 on the Discord if they can provide you with a binary.

This is the AhoyDTU project issue tracker and it makes no sense for you and us to track a build problem you have with DTUsimMI-Hoymiles here. Please open another issue over there or simply ask in the Discord for help. Thanks!

closed 2022-12-19 by stefan123t