lumapu/ahoy #497: successful tested EBYTE E01-ML01DP5 2,4 GHz Wireless Modul

raccoon0815, 2022-12-15:
cool project & super webinstaller
i successful tested ahoyDTU with ESP32 and EBYTE E01-ML01DP5 2,4 GHz Wireless Modul  with this solderless case ==> https://www.thingiverse.com/thing:5632374

## Hardware
  * [ ] ESP8266
  * [x ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [*] EBYTE E01-ML01DP5
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [*] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [*] AhoyDTU
* [*] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

stefan123t, 2022-12-19:
@raccoon0815 is this a question or what purpose has your issue ?

Given the AhoyDTU is a piece of software, what does it have to do with the hardware case,
which is designed for a specific ESP32 board plus a specific NRF24L01+ Module (with PA+LNA) ?

Or do you want us to document that the hardware case also fits the EBYTE E01-ML01DP5 Module ?
Please clarify what you want us to note ?

raccoon0815, 2022-12-20:
@stefan123t why question?
it is an info that EBYTE E01-ML01DP5 2.4GHz works perfectly. It works even better than the standard modules.

stefan123t, 2022-12-25:
Okay that was expected as they are shielded modules and therefor reception should be better.
So your hint on the printed case is that it also fits the Chengdu EByte module just as well as the other NRF24L01+ modules with PA+LNA. I was not aware that they share the same form factor and therefor would fit into the same case. Thanks for documenting this fact.

stefan123t, 2024-10-27:
@lumapu do we need to add that still to the documentation or is this already folklore and can be safely closed ?