lumapu/ahoy #499: [ESP8266/ESP32/RaspberryPi] Problem Description / Beschreibung

Lutz95326, 2022-12-16:
### Platform

ESP8266

### Model name

AHOY-DTU

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [X] I will attach/upload an Image of my wiring

### Version

AHOY :: 0.5.17

### Github Hash

Receive success: 928 Receive fail: 1070 Frames received: 4390 Send Cnt: 9344  Inverter #0: Hoymiles HM-600 (v10010) is not available and is not producing -> last successful transmission: 2022-12-15 16:11:40 Inverter #1: n/a Inverter #2: n/a MQTT: connected

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

1) Go to http://ahoy-dtu/setup
2) configure above settings
3) Reboot
4) I did this
5) I expected that
6) and something completely differen happened
The **temperatu**r of the Hoymile-Device is wrong bei negativ Degreess
[image]
This worong temperature is also send by MQTT see this short memo:
[MQTT-Log.txt](https://github.com/lumapu/ahoy/files/10244240/MQTT-Log.txt)

lumapu, 2022-12-16:
bitte auf das letzte release updaten, dann ist der Fehler nicht mehr da. Die Version `0.5.17` ist schon sehr alt.

Lutz95326, 2022-12-16:
Mache ich.
Vielen Dank für den Tipp!
Sind dann auch die Füllzeichen in MQTT für die Temperatur weg?

lumapu, 2022-12-17:
kann folgendes nicht verstehen, bitte nochmal genauer erläutern
> Sind dann auch die Füllzeichen in MQTT für die Temperatur weg?

Lutz95326, 2022-12-20:
Hallo Stefan,

die Temperaturanzeige im WEB-Interface ist seit dem Update fehlerfrei.
Ich habe die AHOY-DTU an meinen FHEM (Hausautomation) - Server per MQTT gekoppelt.
Seitdem läuft mir regelmäßig eine Log-Datei (FHEM.SAVE) voll:

Ausschnitt aus der oben angehängten Datei "MQTT-Log.TXT":
...
600/ch0/Temp\u0022,\u0022unit_of_meas\u0022:\u0022ÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂÃÂ**°C**\u0022,\u0022uniq_id\u0022:\u0022114182956521_ch0_Temp\u0022,\u0022dev\u0022:{\u0022name\u0022:\u0022Hoymiles HM-600\u0022,\u0022ids\u0022:\u0022114182956521\u0022,\u0022cu\u0022:\u0022http://192.168.0.129\u0022,\u0022mf\u0022:\u0022Hoymiles\u0022},\u0022exp_aft\u0022:95,\u0022dev_cla\u0022:\u0022temperature\u0022,\u0022stat_cla\u0022:\u0022measurement\u0022}","homeassistant/sensor/Hoymiles_HM-600/ch0_U_AC/config":"{\u0022name\u0022:\u0022Hoymiles HM-600 U_AC\
...
Wäre schön, wenn das auch noch behoben würde - Danke!

Schöne Grüße
Horst

Lutz95326, 2022-12-23:
Hallo Lukas,

können Sie das Problem nachvollziehen oder brauchen Sie weitere Informationen?

Schöne Grüße
Horst

lumapu, 2022-12-23:
@Lutz95326 Leider hab ich das bei mir noch nie beobachten können. Sieht bisschen danach aus, als würde ein uninitalisierter Speicherbereich übertragen werden.
Könntest du testweise auch einen anderen MQTT Broker ausprobieren, zB. mosquitto über Docker? Evtl. könntest du auch mal MQTTX ausprobieren, wie ich es hier beschrieben habe:
https://github.com/lumapu/ahoy/issues/489#issuecomment-1356234750

closed 2023-01-17 by lumapu