lumapu/ahoy #500: Vorschlag für MQTT-Ergänzung zu last_success

roku133, 2022-12-16:
Der Parameter last_success enthält den Zeitpunkt der letzten Aktualisierung der Wechselrichterdaten als unixtime.

Zusätzlich lässt sich diese Information auch als Text darstellen, dazu schlage ich zwei weitere Parameter vor:
last_success_utc
last_success_local // am Beispiel CEST/CET

In platformio.ini wird, falls die Lokalzeit angezeigt werden soll, eine Lib hinzugefügt:
jchristensen/Timezone@^1.2.4

In pubMqtt.h ist dann folgendes zu ergänzen:

#include <TimeLib.h>
#include <Timezone.h>   // zur Ausgabe der Lokalzeit

und im Abschnitt // inverter status:

                        time_t t = iv->getLastTs(rec);
                        snprintf(topic, 32 + MAX_NAME_LENGTH, "%s/last_success_utc", iv->config->name);
                        snprintf(val, 40, "%d.%d.%d %02d:%02d:%02d UTC",
                            day(t), month(t), year(t), hour(t), minute(t), second(t));
                        publish(topic, val, true);

                        TimeChangeRule *tcr; // pointer to the time change rule, use to get TZ abbrev
                        TimeChangeRule CEST = {"CEST", Last, Sun, Mar, 2, 120};   // Central European Summer Time
                        TimeChangeRule CET = {"CET ", Last, Sun, Oct, 3, 60};   // Central European Standard Time
                        Timezone myTZ(CEST, CET);
                        time_t local = myTZ.toLocal(t, &tcr);

                        snprintf(topic, 32 + MAX_NAME_LENGTH, "%s/last_success_local", iv->config->name);
                        snprintf(val, 40, "%d.%d.%d %02d:%02d:%02d %s",
                            day(local), month(local), year(local),
                            hour(local), minute(local), second(local), tcr->abbrev);
                        publish(topic, val, true);

Die neuen MQTT-Parameter stellen sich dann so dar:
last_success_utc 16.12.2022 12:17:26 UTC
last_success_local 16.12.2022 13:17:26 CET

Die vorliegende Implementierung der Lokalzeit ist auf die Zeitzone Mitteleuropa beschränkt, kann bedarfsweise erweitert werden.

Entsprechendes ist auch für sunrise/sunset denkbar.

Es lässt sich sicher diskutieren, ob diese Textdarstellungen in der DTU angeboten werden oder eher in den verarbeitenden SmartHome-Systemen. Das gilt dann auch für das bereits vorhandene Parameterpaar "available" und "available_text".

knickohr, 2022-12-16:
Nein, bitte nicht. Man kann das auch prima im Visualisierungstool machen. Im Node-Red z.B. nehme ich einfach einen Date/Time Formatter und schon paßt es für jede Zeitzone auf jedem Punkt dieses Planeten 😉

roku133, 2022-12-16:
@knickohr - nicht alle Systeme/Tools unterstützen diese Datum-/Zeit-Umwandlungen bzw. es fehlen entsprechende Erweiterungen. Es ist ein Angebot, vielleicht als Option in config.h.
Soll gemäß deiner Argumentation der Parameter "available_text" konsequenterweise entfallen? Der zugehörige Text kann aus "available" einfach abgeleitet werden.

knickohr, 2022-12-16:
Das ist korrekt, das ist auch doppelt gemoppelt 😉

lumapu, 2022-12-20:
soll ich `available_text` entfernen?

knickohr, 2022-12-21:
Ist eigentlich nicht nötig, da man es 1:1 aus dem available herleiten kann.

ja !

roku133, 2022-12-25:
available_text ist eine textuelle Ergänzung, die verwendet werden kann oder auch nicht. Das gilt entsprechend für meine vorgeschlagenen, neuen Topics. Es stört bei allen nicht, wenn sie vorhanden sind.

Vielleicht macht es Sinn, derartige (optionale) Ergänzungen in der Konfig-Datei aus- oder abwählen zu können.
Die große Lösung ist, auf der Weboberfläche alle MQTT-Topics auswählbar zu machen.

roku133, 2022-12-28:
Mit der Entscheidung von @lumapu, in v0.5.62 available_text zu entfernen, um damit redundante Darstellungen zu vermeiden, ziehe ich auch meinen Vorschlag zurück.
Den Ansatz, das MQTT-Interface schlank zu gestalten, finde ich bei einer DTU grundsätzlich sinnvoll.

Anwender, die die Umformatierung der Datums-/Zeitangabe verwenden wollen, können ja den Code-Schnipsel weiter zum Source-Code hinzufügen 😉

closed 2022-12-28 by roku133