lumapu/ahoy #501: No connection to Hoymiles HMS-2000 inverter

km1975, 2022-12-16:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: HMS-2000-4T
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [x] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: v0.5.41
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [x] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [x] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)
[image]
[image]
[image]

Hi Team, 
pls. help to connect a HMS-2000 2m in distance to  the AhoyDTU. 
I tested capture3 as well as CS-D8, CE-D2 and IRQ-D1 without success.
Thx for your great support. PLs. help.
Best Regards,
km

km1975, 2022-12-16:
Thx stefan123t for your fast answer, is there a roadmap or approx planning existing?

lumapu, 2022-12-17:
@km1975 Yes I'm currently developing a new version of Ahoy which should support next gen Inverters. I will last at least another month to have a binary here.

km1975, 2022-12-17:
Hi Lukas,

sounds great, if you want me to test something with my HMS-200 pls. let me
know. I'm really interested to do so.

Mit freundlichem Gruß,

K. Martin
Franz-Krämer Str. 18
82229 Seefeld
mobil: +49 (0)173 796 2859

bombastic99, 2022-12-19:
Hallo, 

habe den HMS 1000, habe ebenfalls Interesse. Falls ich was helfen testen kann bitte melden thx

Taube47, 2023-03-03:
Hallo, 

gibt es einen neuen Stand? Kann man sich Hoffnungen machen? Die elektrischen Daten des HMS-2000-4T würden bei mir sehr gut passen.

biasowicka, 2023-03-03:
Czy już działa na HMT-2250 ?

lumapu, 2023-03-06:
@Taube47 ja wir haben bereits erste Daten auslesen können, allerdings noch keine fertige Firmware. Dauert nicht mehr lange 😊

Taube47, 2023-03-06:
Hallo,

ich habe totale Hochachtung vor eurer Leistung.

Dann steht meine Entscheidung fest und ich bestelle drei HMS-2000 für drei Standorte. Zusätzlich komme ich günstig an eine "DTU Pro S" und einen Zähler "Chint DTUS666". Ich möchte damit an einem Standort eine Nulleinspeisung realisieren. Kann man später ja wieder verkaufen.

Wie kann ich helfen?  Geht die Entwicklung so weit, daß auch der Chint DTSU666 Zähler eingebunden wird?

lumapu, 2023-03-07:
das Thema Zähler haben wir noch garnicht betrachtet, daher nein.

closed 2023-08-05 by lumapu