lumapu/ahoy #502: Storing the data?

nicolaskern, 2022-12-17:
Hi,

This is an amazing project, thanks for the great work done!
I don't find any information about the long term storage of the data, either in this device or on the DTU Pro.

Thanks to the JSON ahoy can return, I can process it externally, but in case this is already embedded, that's great!

Thanks,
Nicolas

lumapu, 2022-12-17:
There is now long term data because of missing space. The AhoyDTU is more an interface / bridge than a smart home center.
We recommend to use MQTT to send collected data to a broker (eg mosquito)

stefan123t, 2022-12-18:
Thanks for the kudos. What are you looking for explicitly or in addition to JSON Format or MQTT Broker support. Your question is rather vague. Cheers Stefan

nicolaskern, 2022-12-19:
Hi,
Thanks for your answers. In fact I just wanted to be sure I wasn't missing a functionnality, but indeed due to the small space available on an ESP32 for example, that wouldn't be relevant to store data.
MQTT or JSON are more sufficient to do what I want to do!
Thanks,
Nicolas

closed 2022-12-19 by nicolaskern