lumapu/ahoy #503: ESP32 kein AP nach flash

shueni, 2022-12-17:
### Platform

ESP32

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

board

### Connection diagram

like shown in https://github.com/lumapu/ahoy/blob/main/Getting_Started.md#esp32-wiring-example

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.41

### Github Hash

dec333f

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

not to see

### Debug Serial Log output

```shell
---
```

### Error description

flashed the esp32 with NodeMCU and Firmware 0.5.41. no problems. 
After reconected the USB red led is blinking (about 2 per second) and no AP
AP was there at the beginning, after i tried to connect it disappeard and not coming back

Any ideas?

thanx

lumapu, 2022-12-17:
check #472

stefan123t, 2022-12-18:
This is fixed in the current development branch. Please try the dev version or use the next release when it becomes available.

shueni, 2022-12-19:
HI Thanx, but i think something with the flashing gone wrong. I originally flashed from a Win11 with NodeMCU. Here i had to manually install some drivers, but seemed ok at first, yesterdey i flashed again from a win10 machine direct form your site (web.Version flasher). Now it shows the ap.... 

Thanx again.

closed 2022-12-19 by lumapu