lumapu/ahoy #505: Sonnenaufgang/Sonnenuntergang falsch 0.5.55

Gerri1, 2022-12-17:
Mir ist aufgefallen das der Sonnenaufgang bei meiner Location 20 min zu früh und der Sonnenuntergang 20 min zu spät ist!

lumapu, 2022-12-17:
das ist bewusst so, da manche Ost-West Anlagen bereits mit Streulicht anfangen zu kommunizieren.
Wenn es stört wie es jetzt ist, würde ich hier Interesse einsammeln für einen weiteren Einstellparameter.

Gerri1, 2022-12-17:
Da wäre dann eine seperate einstellbare Zeit unter Sunrise & Sunset für das Offset besser!
Im ioBroker warte ich sowieso erst darauf ( +/- 5 min ), bis das die WR online sind, bevor ich da irgendwelche Aktionen starte.

knickohr, 2022-12-17:
Vielleicht sollen wir das nicht mehr Sunrise/Sunset nennen sondern communication start/stop ?

lumapu, 2022-12-17:
gute Idee @knickohr, wie stehst du dazu @Gerri1 ?

Gerri1, 2022-12-18:
Das wäre auch mein Vorschlag gewesen. 😁👍
Ein weiterer Vorschlag bzw. Frage wäre, den Kommunikation Start und Stopp auch per MQTT u.a. an Ahoy zu senden, wenn das möglich ist? ( für die To-du Liste )

homeautomation2022, 2022-12-18:
Bei mir heißen die Variablen und dementsprechend auch die Anzeige:
PV Sunrise
PV Sunset
Also ich habe einfach nur PV bzw. PV_... bei der Variable davorgehangen, um für mich einen neuen Sonnenaufgang/untergang aus Sicht der Photvoltaik zu definieren, was ihr von der simplen Idee haltet überlasse ich euch, wollte es nur mal als Idee erwähnt haben.
Ich würde zumindest nichts mit Dämmerung einbringen, das verwirrt denke ich eher, auch wenn es richtiger wäre. Genau wie eigentlich noch dran müsste *unter idealen Wetterbedingungen, außerdem muss die Zeit ja soagr noch etwas später nach Sonnenuntergang sein, damit der available Status über MQTT sich bei Nachtbeginn noch ändert, wie @knickohr herausgefunden hat. Ist also eher Pre Sunrise und Post Sunset.

knickohr, 2022-12-18:
Dämmerung ist eigentlich schon richtig 😉 Nur welche Dämmerung ?  Bürgerliche, Nautische oder Astronomische ?   Oder doch die Phototenprasslerdämmerung ?  😂

Da die eigentliche Intension war, die Stati von den WRs sauber mit zu bekommen, am Anfang oder am Ende eines Solartages und dementsprechend die Kommunikation an/abzuschalten, hat dieses Feature überhaupt nichts mit Sonnenauf- oder Untergang zu tun. Ist halt nur zufällig an die Sonne gebunden 🥸

lumapu, 2022-12-18:
über MQTT sollten wir tatsächlich `sunrise` und `sunset` umbennnen. Dafür würde ich den Vorschlag von @homeautomation2022 annehmen und es in `pv_sunrise` und `pv_sunset` umbennen.

stefan123t, 2022-12-18:
`pre_sunrise` und `post_sunset` bzw ` `pv_sunset` und `pv_sunrise` finde ich gut. Und die Winkel für die drei Dämmerungen würde ich per Pulldown anbieten: `common` **6‘/~24min**, `nautic` **12‘/~48min**, `astronomic` **18‘/~72min**. Das Grad Zeichen find ich grad net.
Welche Werte (1,2,3 oder lieber Text common,nautic,astronomic etc.) man dafür per MQTT setzen kann oder muss sollten sich die überlegen die das Feature implementieren.

beegee3, 2022-12-18:
Den Vorschlag von @stefan123t find ich gut! Zusätzlich bitte auch **0 min** falls man keine Korrektur wünscht.
Und in der Web Ansicht bitte den Sunrise / Sunset Text auch anpassen und die Einstellungsmöglichkeit geben.

homeautomation2022, 2022-12-18:
> `pre_sunrise` und `post_sunset` bzw `pv_sunset` und `pv_sunrise` finde ich gut. Und die Winkel für die drei Dämmerungen würde ich per Pulldown anbieten: `common` **6‘/~24min**, `nautic` **12‘/~48min**, `astronomic` **18‘/~72min**. Das Grad Zeichen find ich grad net.
> 
> Welche Werte (1,2,3 oder 6‘,12‘,18’ bzw ~24min,~48min,~72min oder lieber nur Text common,nautic,astronomic etc.) man dafür per MQTT setzen kann oder muss sollten sich die überlegen die das Feature implementieren. Man könnte ja sogar zwischen Grad oder Minuten umrechnen 1‘ entspricht ja ~4min bzw vielfache von ~4min oder 1‘ akzeptieren.

Könnte man da auch nicht gleich frei einstellbar machen (float +-10°)? Dann kann es jeder für seinen Zweck nutzen: wer es zum disablen der Nachtkommunikation nutzen möchte, wird paar Grad weiter negativ gehen, wer nur eine Uhrzeit zur Orientierung wissen möchte ab wann effektiv mit der Einspeisung zu rechnen ist, der wird sich um die 0° bewegen, je nach örtlicher Gegebenheiten.

Grad und Minuten verhalten sich zwischen Sommer und Winter unterschiedlich, wenn auch nicht viel. Man kann auch einen festen Winkel lassen (die originalen -0.83°) und von da aus mit Offset arbeiten. So wie ein Wert für früh und abends, Möglichkeiten gibt es da viele. Mit Minuten würden die meisten sicher besser als mit Grad zu Recht kommen.

beegee3, 2022-12-18:
😨 bitte nicht mit dem Winkel und Offset spielen, sondern den originalen -0.83° wieder einstellen. Der Winkel in der Formel hat mit der Standorthöhe zu tun (s. zug. Wikipedia Artikel) und daher nur indirekt mit den Angaben von @stefan123t. Der jetzige (-3.5°) entspricht z.B. einer Standorthöhe von ~5950 m. Da ist man schon im Himalaya!
Bitte **NACH** `calculateSunriseSunset` einen (vom Nutzer wählbaren) Offset (z.B. die oben genannten 0min, 24min, 48min, 72min) auf die berechneten Sonnenzeiten anwenden.

lumapu, 2022-12-19:
neue Version ist verfügbar, jetzt kann man das Offset in Minuten angeben (über die Settings) `0.5.57`

Gerri1, 2022-12-20:
TOP !!! 👍👍👍

beegee3, 2022-12-20:
Gute Umsetzung. Mir ist nur aufgefallen, dass auch bei abgeschalteter 'disable night communication' 'Communication start/stop' Zeiten angegeben werden. Dann sollte besser so etwas wie 'Communication always enabled' stehen.
Statt 'Communication' wäre vielleicht auch ein 'Radio Communication', 'Inverter Communication' oder 'IV Communication' besser, da die WLan Kommunikation ja nicht betroffen ist.

homeautomation2022, 2022-12-20:
Die Anzeige ist eher dafür gedacht, dass man sich unabhängig von der disable Funktion grob den theoretischen Einspeisebeginn und Ende anzeigen lassen kann. Du hats ja nicht im Kopf, wann heute die Sonne aufgeht und wann unter, dafür kann man es nutzen. Nutzt man die disable Funktino, dann ist es natürlich gleichzeitig auch Debugausgabe, von wann bis wann.

beegee3, 2022-12-20:
@homeautomation2022 Ich denke da auch eher an den Anwender, der diese Diskussion und die Hintergründe nicht kennt und sich wundert, dass da start/stop Zeiten angegeben sind, obwohl er keinen Haken bei 'disable' gesetzt hat.

Gerri1, 2022-12-20:
Die Start/Stop Zeiten sollten aber trotzdem mit angezeigt werden, evtl. nur den Infotext erweitern.
Auch würde anstatt des Infozeichens ein rotes X oder ein grüner Haken mir genügen, als Beispiel.

Gerri1, 2022-12-21:
😁👍

closed 2022-12-31 by Gerri1