lumapu/ahoy #508: Portfreigabe für Fritzbox einrichten

Loetnase, 2022-12-18:
Portfreigabe für meine Fritzbox einrichten

## Hardware
  * [X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: D1mini
Retailer URL: ______

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X] external antenna

### Power Stabilization:
* [X]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [X] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.41
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [X] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [X] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo zusammen,
ich finde toll was ihr hier so alles um das Thema Wechselrichter macht.

Ich komme eher aus der Hardwareecke und Software ist nicht mein Steckenpferd aber mir auch nicht ganz unbekannt :-)
aber die Weiterverarbeitung der Daten zur Visualisierung ist mir (vorerst) zu aufwendig, 
mir reicht die Darstellung im Live Fenster vollkommen aus. 

Zuhause an der Fritzbox funktioniert das auch ganz prächtig, unterwegs über das Internet aber leider gar nicht.

Ich habe schon vor einiger Zeit mal mit der Arduino IDE einen kleinen Webserver mit Temperaturanzeige auf einem ESP8286 gebastelt, da funktioniert das aber ganz gut. Dazu wurde am ESP8286 (mit WiFiServer server(280);) ein Port z.B. 280 eingerichtet welchen ich an der Fritzbox dann freigegeben habe. Mit "MyFritz" kann ich mich dann da drauf verbinden und alles schön anschauen bzw. bedienen.

Ist es möglich in der AhoyDTU eine fixe oder über ein Fenter einstellbare Serveradresse zur Portfreigabe einzurichten?
Das wäre toll, vieleicht sogar ein nettes Weihnachtsgeschenk von euch.  
 
Danke
Ich wünsche euch allen eine schöne Weihnachtszeit.

DukeWolle, 2022-12-18:
Für was eine Portfreigabe? Die Portfreigabe muss doch in die Fritz. AHoy verpasst du in den Wifi Settings eine Static IP. Port 80 von der Fritz weiterleiten an AHOY. 

Was natürlich brauchst ist einen DynDNS .. oder habe ich dich falsch verstanden? 

DukeWolle

tavoc, 2022-12-18:
Dyndns wird nicht gebraucht, da myfritz das schon hat 

Falls du es mit port Freigabe machen willst, dann irgendeinen port auf die IP von ahoy und port 80. 
Ich würde dir aber eher empfehlen djr das Thema VPN anzusehen. 
Mit einer 7590 und wireguard geht es super einfach und du brauchst nichts ins Internet öffnen. Sowas ist immer eine Angriffsfläche, z.b. könnte jeder dann das powerlimit auf 0 setzen oder weiter in das Netzwerk vordringen (falls ahoy Schwachstellen hat)

Loetnase, 2022-12-18:
Hallo DukeWolle,
du hast es richtig verstanden, 
aber der Port 80 ist ein Standardport und sollte nicht für so was verwendet werden, den kennt doch jeder. Siehe Antwort von tavoc. Deshalb möchte ich da einen anderen Port verwenden welchen ich aber irgendwo einrichten muss. 

Hallo tavoc,
das mit VPN werde ich mir wohl mal anschauen.

tavoc, 2022-12-18:
Du kannst in der fritzbox jeden beliebigen externen port auf den internen port 80 umbiegen. 
Es muss keine 1:1 Beziehung sein. 

Z.b. bei mir mein SSH Gateway
[image]

Trotzdem ist das mit dem VPN die bessere Wahl

DukeWolle, 2022-12-18:
@Loetnase 
Klar ist das ein Standardport, aber von dem hängt nicht die Sicherheit ab. Jeder Portscanner findet die Ports. Ausserdem war das ja auch nicht deine Frage. Wie @tavoc ja geschrieben hat kannst den externen Port ja selber bestimmen. Dennoch ist es eine Frage der Sicherheit. 

Von dem her würde ich auch auf ein anderes Pferd setzen.

Loetnase, 2022-12-18:
@tavoc 
das mit den Ports verbiegen kannte ich noch nicht, das ist mal ein interessanter Ansatz aber hilft nur für das erste Gerät, mehrmals geht das nicht. Das werde ich mal ausprobieren. Danke 
@DukeWolle 
das mit der Sicherheit ist mir bewusst, jeder offene Port ist ein offenes Loch für Bösewichte. 
VPN kann ich auf meinem Handy problemlos einrichten, aber auf meinem PC im Geschäft geht das gar nicht. Über den Browser usw. mit Port kann ich aber auf die Ahoy DTU draufschauen und das ist weil neu für mich doch sehr interessant.

tavoc, 2022-12-18:
Warum sollte das mit dem port nur einmal gehen? Es geht genau 65535 mal. 

DeineexterneIP:12345 zu Gerät 1 port 80
DeineexterneIP:12346 zu Gerät 2 port 80
Usw

Loetnase, 2022-12-19:
@tavoc 
Ich dachte einem Gerät ist ein interner Port zugewiesen welcher in der Portfreigabe der Fritzbox zugewiesen wird, der kann von einer oder mehreren externe  (umgeleitet) Portnummern angesprochen werden.
Wenn mehrere Geräte auf den Port 80 hören, woher weiß die Fritzbox oder die Geräte welches antworten soll.
Ich spreche ja nur eine Portnummer mit http://...........myfritz....:80 an.
Ich denke mit mehreren Portnummern ein Gerät ansprechen geht,
mit einer Portnummer mehrere Geräte ansprechen wird wohl nicht gehen.
Wenn du mir da noch einen Tipp dazu hast, das wäre toll.

tavoc, 2022-12-19:
Du sprichst es ja grade nicht mit port 80 an, sondern

Myfritz.de:12345. und das wandelt dann die fritzbox um zu der Freigabe. Man hat also eine Mapping Tabelle auf der Fritzbox und daher kann die Box es auch zuweisen

stefan123t, 2022-12-19:
@Loetnase Du bist Dir schon bewußt dass Dir die Kollegen hier gerade beibringen wie Du die Pistole laden musst, damit Du Dir selbst in den Fuß schießen kannst damit ? Bitte verwende im Namen und Sinne aller Internetnutzer ein VPN dafür, danke! Selbst ein Web Interface der AhoyDTU mit einem guten Passwort ist nicht sicher! Wir haben keine Brute Force Erkennung, etc.

@DukeWolle, @tavoc es ist dies m.E. ein Feature Request für eine offene Seite der AhoyDTU die nur das Dashboard anbietet auf dem Port und keine anderen Services wie REST API, etc. die ggf. auch schreibend die Konfiguration verändern können.
Da aber das Dashboard selbst die REST API braucht um die Daten zu visualisieren führt da m.E. kein Weg drum rum.
Keine Ahnung ob lumapu so eine "abgespeckte Variante" des Dashboards auf einem eigenen Port zur Verfügung stellen will, damit man diesen dann an der FritzBox oder einem anderen Router freigeben kann. :pray: hoffentlich nicht :grin: 
Es wäre zumindest ein großer zusätzlicher Aufwand für einen zweifelhaften Nutzen.

Da fände ich die Thingspeak Integration oder ähnliche Cloud Service Nutzungen interessanter und empfehlenswerter.

tavoc, 2022-12-20:
der Zugriff sollte meiner Meinung nach nicht direkt von ahoy erfolgen und auch nicht die Absicherung.
VPN per Wireguard ist in 5 Min eingerichtet und man öffnet sich kein Loch sonstwohin.

Aber ich hatte weiter oben bereits darauf hingewiesen

lumapu, 2022-12-20:
@tavoc das ist auch meine Meinung. VPN oder kein Fernzugriff. Wireguard läuft auch perfekt auf Android über Mobilfunk, man verwendet sogar die gleichen IP Adressen wie zu Hause und muss nicht umdenken (Bookmarks).

Loetnase, 2022-12-20:
Hallo und Danke an alle für eure Antworten

@stefan123t das mit der Pistole ist eine schöne vergeichende Aussage.
@tavoc, @DukeWolle  aus euren Antworten weiß ich jetzt mehr. Wie die Pistole bedient werden kann und ich sie doch besser entladen in den Schrank lege (noch offene Ports werde ich schließen).
Wireguard habe ich schon mal auf dem Handy drauf und versuche es heute Abend mal einzurichten.
Mit VPN werde ich wohl auch noch mehr Möglichkeiten haben auf meine internen Geräte zuzugreifen.
Auch da werde ich wohl für mich etwas Neues hinzulernen, lernen ist ja eigentlich immer gut und zur Not habe ich immer noch die Pistole im Schrank.

stefan123t, 2022-12-25:
@Loetnase vielleicht willst Du hier Deinen Erfolg beim Einrichten von Wireguard nich beschreiben / dokumentieren dann haben auch andere noch etwas davon die vor der selben Fragestellung stehen wie Du ? Danke!

Loetnase, 2022-12-26:
@stefan123t 
Das war eigentlich zu einfach. 
Ich hatte mir Wireguard auf dem Handy instaliert, da waren dann einige Eingabefelder mit denen ich nicht viel anfangen konnte, aber es gab auch die Möglichkeit einen QR-Code einzuscannen. Also ran an die Fritzbox und schauen wo der zu finden ist, ich fand auch ziemlich schnell die Stelle wo der stehen sollte aber da war kein QR-Code nur diverse Werte für die Eingabefelder. Ich habe leider eine ältere Miet Fritzbox von Vodafone mit gedeckeltem Fritzos 7.29 mit höheren Versionen kann man das dann einfach einscannen.
Dann habe ich noch entdeckt, Fritz hat da schon einen eigenen VPN eingebaut, die FritzApp habe ich ja auch schon drauf und nutze die für andere Dinge.
Also Fritz VPN eingeschalten und schon war alles fertig und hat funktioniert wie gewünscht.
Wireguard habe ich dann wieder runtergenommen vom Handy.
Eine Frage dazu: hat Wireguard Vorteile zum Fritz-VPN oder sind die gleichwertig.
Ich lasse mich da gerne von "Wissenden" eines Besseren belehren.
Danke für eure Hilfe

tavoc, 2022-12-26:
Wireguard geht erst ab fritzos 7.50

Es ist einfacher einzurichten und schafft schnellere Übertragungsraten. Das andere ist ikev1 basiert. Funktioniert aber auch.

Loetnase, 2022-12-26:
@tavoc 
Geht nur das QR-Code einzuscannen mit Wireguard erst ab fritzos 7.50 oder generell Wireguard erst ab fritzos 7.50.
Meine Fritzbox mit 7.30 hat mir schon auf der VPN Seite diverse VPN Daten wie Serveradresse, IPSecID, IPSec-Schlüssel usw. angezeigt die wären wahrscheinlich bei Wireguard in die unterschiedlichen Eingabefelder irgendwo reingekommen. Das hat schon passend ausgesehen und hätte funktionieren können. 
Mir hat aber die Fritz-VPN-Lösung einfacher ausgesehen und das hat auch sofort gefunzt.

lumapu, 2022-12-26:
sehr schön, dass es geklappt hat. Dadurch ist der Issue gelöst oder?

Loetnase, 2022-12-26:
@lumapu 
von meiner Seite aus ist der Issue gelöst.
Mir wurde geholfen, ich habe dabei auch wieder etwas gelernt auch etwas entfernt von meiner Frage.
Nach dem Port habe ich gefragt, den VPN erhalten, es gibt oft mehrere gute Ideen.
Vielen Dank, noch schöne Restweihnachten und einen guten Rutsch ins neue Jahr wünsche ich euch allen.

tavoc, 2022-12-26:
Wireguard ist eine andere Technologie, und wie gesagt erst ab 7.50 bei AvM verfügbar. Das ist eine der grossen Neuerungen. 

Die alten funktionieren aber auch gut, lassen sich ggf. aber schwieriger einstellen. 

Moderne Android Telefone ab Version 11 und Windows 10 aufwärts können von Haus z.b. nur IKEv2 und AVM hat nur IKEv1 
Da braucht man dann die richtigen Tools. 

Wichtig ist dass es für Dich funktioniert

closed 2022-12-26 by lumapu