lumapu/ahoy #509: Kein Reconnect bei WLAN Nachtabschaltung

Ollipop030, 2022-12-19:
### Platform

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.41

### Github Hash

dec333f

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

Bei mir wird Abends an der Fritzbox WLAN abgeschaltet (00:30), und morgens wieder an (07:00). Jedoch verbindet sich die DTU nicht automatisch wieder mit dem WLAN. Ich muss erst einmal die Spannungsversorgung trennen, und ein paar Sekunden später ist die DTU auch wieder da. Gibt es da einen Timeout, wenn die SSID für eine bestimmte Zeit nicht erreichbar ist? Denn: wenn ich WLAN an der Fritzbox mal manuell abschalte, und nach ein paar Minuten wieder ein, dann verbindet sich die DTU auch wieder. Scheint so, als ob das erst nach Stunden passiert. 0.5.53 hatte ich auch versucht, gleiches Phänomen.

lumapu, 2022-12-19:
muss ich testen, spontan wüsste ich nicht warum es einen Unterschied zwischen wenigen Minuten und Stunden geben soll.
Ist der Ahoy dann per AP erreichbar?

Wäre es möglich ein serielles Log zu bekommen?

Ollipop030, 2022-12-19:
Einen AP gibt es dann nicht, das war auch meine erste Idee. Zum Log: Wird schwierig, es gibt keinerlei Möglichkeit auf die DTU zu kommen, oder werden die Logs irgendwo hingespeichert?

lumapu, 2022-12-19:
nein da wird nichts gespeichert, das müsste man aktiv über USB mitloggen.
Ich schaue mal in den Source, evtl. kann man dadurch das Verhalten schon nachvollziehen

Karlo49, 2022-12-19:
Bei mir das gleiche als meinen FritzBox für 2 min stromlos war!

knickohr, 2022-12-19:
Das Problem haben viele ESP, leider. Ist also nicht eine Sache allein vom Ahoy.

roku133, 2022-12-19:
Mit v0.5.41 und den Versionen davor gab es bei mir mit der nächtlichen WLAN-Abschaltung meiner Fritzbox nie Probleme, der Reconnect hat monatelang funktioniert. Ab v0.5.52 (improved ahoyWifi class?) war die DTU morgens dann nicht mehr erreichbar.
Mit v0.5.56 hat es heute Morgen erstmals wieder funktioniert, ich werde das Verhalten mit dieser Version weiter beobachten.

Ollipop030, 2022-12-20:
Die v0.5.41 hatte ich jetzt knapp drei Wochen drauf, da hat der reconnect morgens nie funktioniert. Habe seit gestern die 0.5.56 drauf, und heute morgen ist die DTU wieder nicht zu sehen. Auch die Fritzbox sieht das Gerät in der Übersicht nicht.

roku133, 2022-12-20:
Heute Morgen hat der Reconnect mit v0.5.56 nicht funktioniert, die Verbindung gestern war offensichtlich eine positive Ausnahme.
AP wird auch nicht angezeigt.

beegee3, 2022-12-20:
Habe festgestellt, dass ein schlechter Wifi Rssi das connect/reconnect behindern kann (auch die Web Anzeige reagiert dann träge). Hatte ESP am Laptop angeschlossen mit einem Rssi von ~ -80 dBm. Im Terminal wurden permanent erfolglose Verbindungsversuche angezeigt. Standort geändert, Rssi ging auf ~ -45 dBm und die Verbindung kam sofort zustande.
Weiß nicht genau, wie die Fritz-Box nach einer WLAN Nachtabschaltung funktioniert. Zumindest nach manuellem ab- und anschalten wird die Leistung kurz auf 100% angehoben, bevor wieder ein 'Stromspar Modus' aktiviert wird. Meine ich zumindest, ohne sicher zu sein; lasse mich da gerne korrigieren. Wenn dem so ist, dann reicht das i.d.R aus, dass Ahoy wieder online geht.

beegee3, 2022-12-20:
Zusatz: Wenn die Fritz-Box WLAN Nachtabschaltung eingestellt ist, ist in dieser Zeit ein permanenter Verbindungsversuch von Ahoy natürlich sinnfrei. Da wäre es schön, auch eine Nachtabschaltung in Ahoy zu haben ("deepSleep"). Bevor hier jetzt jemand aufschreit und ein Shitstorm losbricht: ich glaube, dass ist gar nicht aufwändig, bzw. unsinnig und ließe sich auch gut mit den neuen IV Connect Zeiten kombinieren (v0.5.56). Ich schreib dazu mal in einem neuen 'issue'.

Ollipop030, 2022-12-20:
Das wäre eine Möglichkeit. Wobei es ja auch reichen würde, alle 10 Minuten mal zu sehen, ob die SSID wieder da ist. Ich weiß nicht wie andere Geräte das machen, aber 2x RPi und verschiedene Shellys sind immer kurz nach Beenden der Nachtschaltung wieder verbunden.

knickohr, 2022-12-20:
Ich kann nur sagen, ich habe Dutzende ESP am werkeln. Viele verlieren leider die Verbindung für immer wenn das WLAN mal weg war. Bisher hat da nur ein manueller Reboot geholfen. Ob es vielleicht an der benutzen Bibliothek im Code ,liegt ?  Könnte mir denken das viele Projekte die gleiche lib verwenden.

beegee3, 2022-12-20:
@knickohr die Bibliothek ist Standard Wifi des Arduino frameworks. Hab' mir mal bei schlechtem Rssi das serial-log am USB Port angeschaut. Da kommt immer wieder 'E (4534) wifi:Set status to INIT' , wobei die Zahl in Klammern variiert. Typisches Problem, wenn man Google glauben darf. Nur vernünftige und einfache sind wohl rar. Das passiert auch nach einem Update oder Click auf den Reboot Button (web System Seite). Bekomme nun aber nach leichter Codeanpassung doch schnell die Verbindung: @lumapu `mCnt = 0; mConnected = false; mReconnect = false;` aus `ahoywifi::ahoywifi()` nach `ahoywifi::setup` verlegt und in `ahoywifi::setupStation` als letzte Anweisung `mReconnect = true;` eingefügt. Vielleicht hilft das ja auch bei der Nachtschaltung.

lumapu, 2022-12-20:
@beegee3 Ich habe heute sehr viel an der `ahoywifi` geschraubt - kannst bitte nochmal mit klarem Kopf drüber schauen. Was ich heute dabei gelernt habe:

-  `WiFi.reconnect();` scheint bei ESPs nicht zu funktionieren, berichten mehrere im Netz
- Wenn man den ESP so konfiguriert, dass er gleichzeitig im AP und STA mode ist, dann wird der AP sehr lahm. Das liegt wohl daran, dass der ESP versucht die Verbindung mit STA aufzubauen und das auf allen Kanälen und ständig. Dadurch wird im AP mode die Website schleppend langsam bis gar nicht mehr verwendbar.
Als Lösung scannt der ESP jetzt nach dem Station Wifi. Erst wenn es gefunden wird, wird auch versucht eine Verbindung aufzubauen

knickohr, 2022-12-21:
Das dachte ich mir schon das diese Lib wohl schon wie Unkraut unter der Masse ist 😞

beegee3, 2022-12-21:
@lumapu 😞 funktioniert auch nicht. Die Idee mit dem Scan ist ja nicht schlecht, im log am USB Port sieht man auch, dass das der Router gefunden wird, aber die Verbindung nach einem Reboot oder Update klappt nicht (manchmal kommt wieder 'wifi:Set status to INIT', s.o.), da die bisherige Wifi Konfiguration wohl auch dann noch vorhanden ist aber ins Leere läuft.
Abhilfe schafft hier wieder ein `mReconnect = true;` nach der Zeile `WiFi.begin(...)` in `ahoywifi::setupStation`. Das sorgt dafür, dass in `ahoywifi::loop` zunächst mit `WiFi.disconnect();` die alte Wifi Konfiguration gelöscht wird und im nächsten Schritt neu verbunden wird (was auch mit einem `WiFi.reconnect();` erreicht würde).
Somit hat man mit der neuen `ahoywifi` leider nichts gewonnen 😢. Ich finde sie noch etwas unübersichtlich, die neuen Variablen und vor allem die Funktionsweise von `mLoopCnt` ist schwer nachzuvollziehen. Hier kann man natürlich noch dran arbeiten, mehr Struktur reinbringen und vereinfachen. Letztlich landet man dann wahrscheinlich wieder bei der vorherigen `ahoywifi`, ein wenig mit dem Scan aufgepeppt (den es m.E. aber hier nicht braucht). Ich würde daher wieder auf die alte Version zurückgehen mit dem oben vorgeschlagenen
>  `mCnt = 0; mConnected = false; mReconnect = false;` aus `ahoywifi::ahoywifi()` nach `ahoywifi::setup` verlegt und in `ahoywifi::setupStation` als letzte Anweisung` mReconnect = true;` eingefügt.

Was ich aber nicht verstehe, ist
> Wenn man den ESP so konfiguriert, dass er gleichzeitig im AP und STA mode ist, dann wird der AP sehr lahm.

Mit der Verbindung zum Router wird der AP doch abgeschaltet (zumindest finde ich den AP dann nicht mehr) und ich dachte, das ist auch so gewollt. Habe mich daher immer schon gefragt, warum man zunächst `WIFI_AP_STA` einstellt und nicht einfach `WIFI_AP`, nach der Router Verbindung aber `WIFI_STA`.
Vielleicht findet man hier einen Ansatzpunkt: beim Verbindungsverlust wird der Modus nicht zurückgestellt. Evtl. muss man hier den AP reaktivieren, sozusagen als 'fallback', also nicht nur in `setupAp`.
Habe leider frühestens heute Abend Zeit für mehr Tests.

beegee3, 2022-12-21:
@lumapu so, bin nochmal mit der alten `ahoywifi` Version gestartet und hab' einiges ausprobiert. Nun funktioniert (bei mir) der Reconnect für alle Ereignisse: Update, Reboot, WLAN aus ... (warten) ... an.
Dabei habe ich so viel hier und da 'geschraubt', dass es das Einfachste ist, die komplette `ahoywifi.cpp` zu posten (Entschuldigung), die `.h` ergibt sich entsprechend:
```
//-----------------------------------------------------------------------------
// 2022 Ahoy, https://www.mikrocontroller.net/topic/525778
// Creative Commons - http://creativecommons.org/licenses/by-nc-sa/3.0/de/
//-----------------------------------------------------------------------------

#if defined(ESP32) && defined(F)
  #undef F
  #define F(sl) (sl)
#endif
#include "ahoywifi.h"

// NTP CONFIG
#define NTP_PACKET_SIZE     48

//-----------------------------------------------------------------------------
ahoywifi::ahoywifi() {
    /*
    mCnt       = 0;
    mConnected = false;
    mReconnect = false;
    */
}

//-----------------------------------------------------------------------------
void ahoywifi::setup(settings_t *config, uint32_t *utcTimestamp) {
    mConfig = config;
    mUtcTimestamp = utcTimestamp;

    mCnt       = 0;
    mConnected = false;
    mReconnect = false;

    #if defined(ESP8266)
    wifiConnectHandler = WiFi.onStationModeConnected(std::bind(&ahoywifi::onConnect, this, std::placeholders::_1));
    wifiGotIPHandler = WiFi.onStationModeGotIP(std::bind(&ahoywifi::onConnect, this, std::placeholders::_1));
    wifiDisconnectHandler = WiFi.onStationModeDisconnected(std::bind(&ahoywifi::onDisconnect, this, std::placeholders::_1));
    #else
    WiFi.onEvent(std::bind(&ahoywifi::onWiFiEvent, this, std::placeholders::_1));
    #endif

    setupWifi();
}

//-----------------------------------------------------------------------------
void ahoywifi::setupWifi(void) {
    #if !defined(FB_WIFI_OVERRIDDEN)
        if(strncmp(mConfig->sys.stationSsid, FB_WIFI_SSID, 14) == 0)
            setupAp();
    #endif
    #if !defined(AP_ONLY)
        if(mConfig->valid) {
            #if !defined(FB_WIFI_OVERRIDDEN)
                if(strncmp(mConfig->sys.stationSsid, FB_WIFI_SSID, 14) != 0)
                    setupStation();
            #else
                setupStation();
            #endif
        }
    #endif
}

//-----------------------------------------------------------------------------
void ahoywifi::loop() {
    #if !defined(AP_ONLY)
    if(mReconnect) {
        if (WiFi.softAPgetStationNum() > 0) // do not reconnect if any AP connection exists
            return;
[... 200 lines cut ...]
            case SYSTEM_EVENT_STA_GOT_IP:
                welcome(WiFi.localIP().toString() + F(" (Station)"));
                break;

            case SYSTEM_EVENT_STA_DISCONNECTED:
                connectionEvent(false);
                break;

            default:
                break;
        }
    }
#endif

//-----------------------------------------------------------------------------
void ahoywifi::welcome(String msg) {
    DBGPRINTLN(F("\n\n--------------------------------"));
    DBGPRINTLN(F("Welcome to AHOY!"));
    DBGPRINT(F("\npoint your browser to http://"));
    DBGPRINTLN(msg);
    DBGPRINTLN(F("to configure your device"));
    DBGPRINTLN(F("--------------------------------\n"));
}

```

lumapu, 2022-12-21:
viel Dank, das muss ich mir morgen anschauen. Ich bin echt froh, dass du dich da auch einbringst. Ich war gestern echt am Verzweifeln mit dem WLAN. Bin gespannt, ob meine getesteten Szenarien auch funktionieren.

knickohr, 2022-12-22:
Nein Lukas @lumapu , machen sie offensichtlich nicht 😞

Die DTU bootet in ca. regelmäßigen Abständen von 7 Stunden. Hast Du an dem 8266 auch was bezüglich WiFi geändert ?

Der harte Wachhund schlägt zu.

lumapu, 2022-12-22:
@knickohr nein habe dort nichts verändert, meine beiden ESPs liefen durch ohne Probleme, hatte sie per OTA aktualisiert

knickohr, 2022-12-22:
Das ist seltsam, ich hatte in den letzten 24 Stundenn3 Rebbots, eben der Hardware Watchdog 🤔

beegee3, 2022-12-22:
@lumapu 
>if (WiFi.softAPgetStationNum() > 0) // do not reconnect if any AP connection exists
            return;
 
ja, aber das war ohne Funktion, da der AP mode nicht aktiviert wurde.
Habe den Code oben editiert: jetzt wird immer im AP mode gebootet und erst nach 10 Sekunden wird nach der SSID gescannt (hab' deine Idee geklaut, erst zu schauen, ob das Station Netzwerk überhaupt verfügbar ist 😄, mach' es allerdings nicht asynchron). Falls vorhanden, wird die Verbindung aufgenommen.
Geändert sind `setupWifi`, `loop`, `setupAp` (hab `mDns.setErrorReplyCode(DNSReplyCode::NoError);` auch von dir übernommen), `setupStation` und neu `scanStationNetwork`.
Soweit funktioniert das jetzt. Ich versuche nun, das reconnect in einen ticker zu verlagern und dabei eine Netzwerk Nachtabschaltung zu berücksichtigen.

@knickohr Ist das WLAN stabil? Hatte bei schlechtem RSSI auch ab und zu reboots

lumapu, 2022-12-22:
@beegee3 ich habe eine neue Version veröffentlicht, die bei mir jetzt in Wien Situationen läuft, basierend auf deiner Version von gestern Abend.
Deine neuesten Änderungen sind noch nicht umgesetzt, ich weiß noch gar nicht ob sie überhaupt noch relevant sind.

Bin auf dein Feedback gespannt.

lumapu, 2022-12-22:
wollen wir wirklich den AP aufmachen, wenn die STA Verbindung flöten geht, ich denke das sollten wir nicht machen.

beegee3, 2022-12-22:
@lumapu 👍 die neue Version ist ok (keine Testprobleme gehabt).  Ein Fallback in den AP muss nicht sein. Wäre nur eine Zugriffsmöglichkeit wenn STA reboot nicht funktioniert (Nachtabschaltung oder Netzwerkproblem). So muss man dann kurz den Strom wegnehmen und damit einen Reboot erzwingen. Sollte aber ja selten notwendig sein. Willst du es so lassen, oder was hälst du von der Idee, das reconnect in einen ticker zu verlagern? (ohne dir die Arbeit zu nehmen, würd ich es probieren, interessiert mich halt gerade)

knickohr, 2022-12-22:
@beegee3 Habe auf die 59 upgedatet, warten wir es mal ab. Bisher gabs bei mir ja nie Probleme mit dem WiFi. Allerdings habe ich 4 Repeater in einem Mesh.

beegee3, 2022-12-23:
@lumapu habe die `ahoywifi::loop` in einen Sekunden Ticker verlagert und aufgeräumt (s.o., Code nochmals editiert).
Es wird immer für mind. 10 sec im AP Modus gebootet. Danach bei Bedarf auf STA Modus umgeschaltet. AP_STA Kombi gibt es nicht!
In ahoywifi.h sind die Variablen in den letzten Zeilen nur noch
```
        bool mConnected, mReconnect, mIsApMode, mGotIp;
        uint8_t mCnt;
        uint32_t *mUtcTimestamp;
```
Dazu noch der Ticker:
in `app::setup`: `everySec(std::bind(&app::tickWifiLoop, this)); `
in `app::loop` `mWifi.loop();` gelöscht
und neu
```
void app::tickWifiLoop(void) {
    mWifi.loop();
}
```

Hätte gerne variable Tickerzeiten mit `once(...)` gemacht (dafür auch die Variable `mGotIp`), aber dabei bin ich auf ein Designproblem des Schedulers gestoßen:
wenn innerhalb eines `once(...)` Tickers ein neuer mit gleichem timeout erstellt wird, führt `checkEvery` diesen auch sofort aus und man gerät in eine Todesschleife (gleiches gilt für `checkAt`).
Es ist ein Wunder, dass das noch nicht aufgefallen ist. `tickNtpUpdate` kommt nur raus, weil das timeout nach erhaltener NTP Zeit vergrößert wird.
**Hier ist eine Anpassung notwendig!!!**

lumapu, 2022-12-23:
@beegee3 würde es helfen, wenn man eine ID vom Scheduler bekommt mit der man ein vorhandenes `once(...)` aktualisieren / ändern kann?
Zum Verständnis: Du hättest gerne ein `once(...)` für die WiFi loop eingeführt. In dieser Loop wird dann dynamisch, also zu einem beliebigen Zeitpunkt ein Event ausgelöst. Mit diesem Event hättest du gerne die Funktion selbst wieder durch ein weiteres `once(...)` aufgerufen. Wenn jetzt aber zufällig beide `timeouts` identisch sind, dann würde die WiFi loop ab dann doppelt aufgerufen werden, 2x direkt nacheinander.

lumapu, 2022-12-23:
> AP_STA Kombi gibt es nicht!

was bedeutet das deiner Meinung nach?

Folgendes Szenario geht mit dem heutigen Code von oben nicht: 

1. hänge an dein STA Wifi im Setup ein paar Buchstaben an zB. `er`. Dadurch wird es nicht mehr gefunden, was dem Fall einer Umbenennung des lokalen STA WiFi entspricht.
2. Starte den ESP neu. Jetzt hast du exakt 10s Zeit dich mit dem AP zu verbinden - danach nie wieder (außer Reboot)

beegee3, 2022-12-23:
> Zum Verständnis: Du hättest gerne ein once(...) für die WiFi loop eingeführt. ...

Ja, die Idee ist, zunächst im setup einen `once(...)` Timer für `tickWifiLoop` mit 10s einzustellen (AP mode beim boot).
`tickWifiLoop` so modifiziert:
```
void app::tickWifiLoop(void) {
    mWifi.loop();
    uint32_t nxtTrig = mWifi.getNextTrigger();
    once(std::bind(&app::tickWifiLoop, this), nxtTrig);
}
```
und in `ahoywifi`
```
uint32_t ahoywifi::getNextTrigger(void) {
    return (mReconnect || mIsApMode || !mGotIp) ? 1 : 10;
}
```
d.h. nur im STA Modus und bei vorhandener IP wird der Ticker wieder auf 10s (oder was auch immer).
Solange man im AP Modus ist, oder die Verbindung zum Router noch nicht abgeschlossen ist (Connected und GotIp),
läuft der Ticker auf 1s. Die 10s (oder was auch immer) sind ja nur um zu prüfen, ob ein Reconnect notwendig ist.

Wenn man es noch eleganter will, könnte man im STA Modus und bei vorhandener IP keinen neuen Ticker setzen und erst bei einem Disconnected event den Ticker für ein Reconnect neu starten. Das könnte über das Interface initiiert werden (analog zu `setRebootFlag`). Baut man noch ins Setting Nachtruhezeiten ein (2 Stunden+Minuten Abgaben für von ..., bis ...), könnte man auch darauf reagieren und Ticker für Wifi ab- und anschalten nutzen.

---
> Folgendes Szenario ...

Genau das meinte ich mit 
> AP_STA Kombi gibt es nicht

das ist meine Interpretation von

> wollen wir wirklich den AP aufmachen, wenn die STA Verbindung flöten geht, ich denke das sollten wir nicht machen.

Aber natürlich, bei fehlerhafter STA Wifi Angabe, oder sonstiges, bei dem kein GotIp zustande kommt, bleibt nur der Reboot und 10s um in den AP zu kommen. Wenn dies abgefangen werden soll, könnte man den Reconnect Counter `mCnt` so erweitern, dass nach einer bestimmten Zeit (60s oder mehr) doch ein Fallback in den AP Modus gemacht wird.

In diesem Sinne **frohe Weihnacht**!

lumapu, 2022-12-24:
ja von den 60s kommen wir her, wir wollten flexibler sein und nicht erst 60s warten wenn alle Einstellungen passen.

Vielen Dank, frohe Weihnachten 🌲

stefan123t, 2022-12-26:
@beegee3 danke für die vielen guten Vorschläge.
Die Option mit STA und AP Mode parallel kam daher, daß früher ein fester Timeout beim AP aktiv war. Dann haben wir herausgefunden dass der AP mode einen Connection Count hat, der uns zeigt ob noch jemand mit dem AP verbunden ist. So lange halten wir diesen ggf. offen. Den STA mode können wir initial ja nicht aufmachen, da die Dummy Daten SSID & Password natürlich nicht erreichbar sind. Bei schlechtem Empfang des WiFi Signals haben wir ebenfalls den Zustand dass der ESP mehrere Versuche macht den Route im STA mode zu kontaktieren. Zwischendrin wird/wurde immer der AP wieder aufgemacht. D.h. eine Minute AP, dann eine Minute STA probieren. Schlussendlich haben wir noch herausgefunden dass der ESP bzw. die WiFi library auch "problemlos" zwei IP Verbindungen aufmachen kann, nämlich das AP if0 und das STA if1. Wenn man den ESP mit den entsprechenden Debug Flags kompiliert, dann gibt er auch div. Informationen zur IP if State Machine aus.

Aber wie Du vermutlich richtig bemerkt hast ist die STA_AP Mode Kombination nicht gerade trivial, wenn / da man dann die State change Events beider Interfaces auswerten will & muß. Praktisch wäre es aber weiterhin denkbar den AP mode nicht nur beim Booten sondern auch jedesmal wenn wir die STA Verbindung verlieren für eine Minute aufzumachen. Eventuell gibt es auch ein Flag mit dem man den AP mode immer aktiv hält. Ich glaube das hatten bereits einige Nutzer nachgefragt, die das z.B. in einem Ferienhaus ohne Internet Verbindung so verwenden wollten. Daher auch die Präprozessor / Compile-time Option AP_ONLY.

beegee3, 2022-12-27:
@stefan123t Danke für deine Erklärung, ist manchmal schwierig, die ganze Historie nachzuvollziehen.
Dann lass doch den Anwender selbst entscheiden, welchen Modus er verwenden will: In den Einstellungen eine Auswahlbox mit
AP only / STA only / AP + STA. Dann braucht man auch die Compile-time Option AP_ONLY nicht mehr.
Beim boot / reboot wird mit AP gestartet, in der loop bei Bedarf entsprechend STA hinzugefügt / umgeschaltet. Hatte in dem zweiten Entwurf oben eine Funktion `scanStationNetwork` , wodurch die Umschaltung auf STA vermieden wurde, falls der Router nicht gefunden wird (fehlerhafte STA Wifi Angabe, ...). Und bei einem STA Disconnect Event kann man wieder mit dem AP Modus starten (als wäre es ein reboot).

beegee3, 2022-12-28:
@lumapu der Counter in der `tickWifiLoop` ist noch nicht an `everySec` angepasst. Bis zum `reconnect` dauert es 100s statt 10s.
Anpassung: `delay(100);` löschen (ist im Sekundenticker enthalten) und nach `mCnt++;` :
```
        DBGPRINT(F("reconnect in "));
        DBGPRINT(String(10-mCnt));
        DBGPRINTLN(F(" seconds"));
        if((mCnt % 10) == 0) { // try to reconnect after 10 sec without connection
            WiFi.reconnect();
            mCnt = 0;
        }

```

beegee3, 2022-12-29:
Die Countdown Anzeige `DBGPRINT(String((100-mCnt)/10));` ist noch falsch, richtig ist `DBGPRINT(String(10-mCnt));`
Und ich vergaß den Hinweis in `connectionEvent` `mCnt` von 50 auf 5 zu ändern.

beegee3, 2022-12-29:
Mir ist aufgefallen, dass der Reconnect manchmal mehr als 10s braucht, bis die Verbindung steht.
Wenn also in `tickWifiLoop `schon ein `Wifi.reconnect` gestartet wurde, sollte die Zeit bis zum nächsten Versuch länger sein.
Dabei ist mir aufgefallen, dass `mReconnect` und `mConnected` zusammengefasst werden können, dazu folgender Vorschlag:
ahoywifi.h
```
    private:
        typedef enum WiFiStatus
        {
            DISCONNECTED = 0,
            CONNECTING,
            CONNECTED,
            GOT_IP
        } WiFiStatus_t;
        ...
        void connectionEvent(WiFiStatus_t status);
        ...
        WiFiStatus_t mStaConn;
        uint8_t mCnt;
        ...
```
ahoywifi.cpp
```
void ahoywifi::setup(settings_t *config, uint32_t *utcTimestamp) {
    ...
    mStaConn    = DISCONNECTED;
    ...
}

void ahoywifi::tickWifiLoop() {
    #if !defined(AP_ONLY)
    if(mStaConn != GOT_IP) {
        if (WiFi.softAPgetStationNum() > 0) { // do not reconnect if any AP connection exists
            mDns.processNextRequest();
            if((WIFI_AP_STA == WiFi.getMode()) && !mScanActive) {
                DBGPRINTLN(F("AP client connected"));
                welcome(mApIp.toString());
                WiFi.mode(WIFI_AP);
            }
            return;
        }
        else if(WIFI_AP == WiFi.getMode()) {
            mCnt = 0;
            WiFi.mode(WIFI_AP_STA);
        }
        mCnt++;

        if (mStaConn == CONNECTED) {    // connected but no ip
            DBGPRINT(F("retry connecting in "));
            DBGPRINT(String(20-mCnt));
            DBGPRINTLN(F(" seconds"));
            if((mCnt % 20) == 0) {      // reset to reconnect after 20 sec without ip
                WiFi.reconnect();
                mCnt = 0;
            }
        }
        else {
            DBGPRINT(F("reconnect in "));
            DBGPRINT(String(10-mCnt));
            DBGPRINTLN(F(" seconds"));
            if((mCnt % 10) == 0) {      // try to reconnect after 10 sec without connection
                mStaConn = CONNECTING;
                WiFi.reconnect();
                mCnt = 0;
            }
        }
    }
    #endif
}

void ahoywifi::setupStation(void) {
    ...
    mStaConn = (WiFi.begin(mConfig->sys.stationSsid, mConfig->sys.stationPwd) != WL_CONNECTED) ? DISCONNECTED : CONNECTED;
    ...
}

void ahoywifi::connectionEvent(WiFiStatus_t status) {
    switch(status) {
        case CONNECTED:
            if(mStaConn != CONNECTED) {
                mStaConn = CONNECTED;
                DBGPRINTLN(F("\n[WiFi] Connected"));
                WiFi.mode(WIFI_STA);
                DBGPRINTLN(F("[WiFi] AP disabled"));
                mDns.stop();
            }
            break;

        case GOT_IP:
            mStaConn = GOT_IP;
            welcome(WiFi.localIP().toString() + F(" (Station)"));
            break;

        case DISCONNECTED:
            if(mStaConn != CONNECTING) {
                mStaConn = DISCONNECTED;
                mCnt       = 5;     // try to reconnect in 5 sec
                setupWifi();        // reconnect with AP / Station setup
                DPRINTLN(DBG_INFO, "[WiFi] Connection Lost");
            }
            break;

        default:
            break;
    }
}

//-----------------------------------------------------------------------------
#if defined(ESP8266)
    //-------------------------------------------------------------------------
    void ahoywifi::onConnect(const WiFiEventStationModeConnected& event) {
        connectionEvent(CONNECTED);
    }

    //-------------------------------------------------------------------------
    void ahoywifi::onGotIP(const WiFiEventStationModeGotIP& event) {
        connectionEvent(GOT_IP);
    }

    //-------------------------------------------------------------------------
    void ahoywifi::onDisconnect(const WiFiEventStationModeDisconnected& event) {
        connectionEvent(DISCONNECTED);
    }

#else
    //-------------------------------------------------------------------------
    void ahoywifi::onWiFiEvent(WiFiEvent_t event) {
        DBGPRINT(F("Wifi event: "));
        DBGPRINTLN(String(event));

        switch(event) {
            case SYSTEM_EVENT_STA_CONNECTED:
                connectionEvent(CONNECTED);
                break;

            case SYSTEM_EVENT_STA_GOT_IP:
                connectionEvent(GOT_IP);
                break;

            case SYSTEM_EVENT_STA_DISCONNECTED:
                connectionEvent(DISCONNECTED);
                break;

            default:
                break;
        }
    }
#endif
```

lumapu, 2022-12-30:
> ```
> if(WIFI_AP == WiFi.getMode())
>     WiFi.mode(WIFI_AP_STA);
> ```
> in scanAvailNetworks wird nicht m.E. benötigt, WiFi.scanNetworks schaltet das selbst um.

Doch, das wird schon benötigt und zwar wenn man im AP Mode ist und nach Netzwerken sucht. Dann muss er in den `WIFI_AP_STA` mode wechseln um nach Netzwerken suchen zu können.

Den Code habe ich 'fast' 1:1 übernommen, nur in `tickWifiLoop` habe ich noch doppelten Code reduziert.

closed 2022-12-30 by lumapu