lumapu/ahoy #510: inverter type can't be detected!

bombastic99, 2022-12-19:
Hallo,

habe einen Hoymiles 1000. Leider klappt das auslesen nicht mit folgender Fehlermeldung:

inverter type can't be detected!
E: unknown inverter type: 11E: 44

reserve85, 2022-12-19:
MI1000 oder HM1000?

DanielR92, 2022-12-19:
Die ganze Seriennummer muss hinterlegt sein.
In der Log steht es ja, er kann nichts mit "44" Anfangen.

bombastic99, 2022-12-19:
Hallo, die Seriennummer ist vollständig hinterlegt (habe die aus Sicherheitsgründen nicht gepostet) es steht auch Wechelrichtertyp dabei den er nicht kennt

HMS 1000 der hat Antenne dran

bombastic99, 2022-12-19:
[image]

reserve85, 2022-12-19:
srry, but HMS type is not supported yet.

bombastic99, 2022-12-19:
Hallo, hab's mir fast gedacht. 
Danke für Deine Antwort...hast Du Informationen ob das noch kommt oder kann ich das Thema begraben?

reserve85, 2022-12-19:
hab das hier gefunden: https://github.com/lumapu/ahoy/issues/501
denke das trifft dann auch auf den 1000er zu.

DanielR92, 2022-12-19:
HMS wird daran gearbeitet... solange kann man dieses issue closen meine ich. :)

bombastic99, 2022-12-19:
wird derzeit noch nicht unterstützt HMS

closed 2022-12-19 by bombastic99

stefan123t, 2023-01-14:
see #233 for implementation progress.