lumapu/ahoy #512: Verbesserungswünsche

Karlo49, 2022-12-19:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: __8266 Pro____
Retailer URL: ______

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x] external antenna

### Power Stabilization:
* [ x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0_.5_._56_
Github Hash: c8ec1c1_______

### Build & Flash Method:
* [ ] Arduino
* [ x] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ x] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ x] Setup settings (use our templates ... to be added)

Karlo49, 2022-12-19:
Ich benutze nur die Ahoy-DTU in der Webapp auf meinem Linux Desktop.
MQTT habe ich ein bischen getestet, aber bin noch  nicht sicher, ob ich das haben will.
Hier ein paar Dinge, die ich vermisse und gut gebrauchen könnte:
1. Anzeige des Device-Name im Kopf
2. Menue-Leiste ausblendbar
3. Grafic und Schrift auf ca. 75% verkleinerbar
4. Im Setup unter Inverter für jedes Modul ein Feld indem man die erzeugten YieldTotal eintragen kann und diese von den   bisherigen YieldTotal abgezogen werden. Sodas man z.B. zum 1.januar YieldTotal = 0 hat.
Vielen Dank für die bisher geleistete Arbeit.

knickohr, 2022-12-19:
Ähhh ?   Wo willst Du das haben ?  Auf dem Liveview vom Ahoy, also der Ahoy-Webseite oder im MQTT über Deinen Visualisierer (Node-Red) o. Ä. ?

Karlo49, 2022-12-19:
Wie geschrieben auf der Ahoy-Webseite

knickohr, 2022-12-19:
OK,

also bei mir ist das so (siehe Foto), und die Schriftgräße kann man über den Browser für jede Seite im Browser einstellen 🤔

Nummer 4 ist natürlich eine Herausforderung 😉

[image]

Karlo49, 2022-12-19:
Schriftgröße ist in der Webapp nicht einstellbar. Dafür ist aber alles drumrum weg.
[image]

1-am-r00t, 2022-12-19:
@Karlo49 versuch doch mal entweder [STRG] + "-" / "+" oder alternativ [STRG] + Mausrad rauf/runter. Damlt kannst du im Browser die Schriftgröße an deine Vorlieben anpassen. Das ist glaube ich was @knickohr meinte.

stefan123t, 2022-12-19:
Gerade Nummer 4. wird sollte eigentlich ganz interessant sein und m.E. durchaus möglich:
Die Differenz `deltaYieldTotal` zum vorherigen Wert (Prefix `_`) ist ja einfach `YieldTotal - _YieldTotal`
Eventuell könnte man daraus auch gleich den aktuellen Yield pro Zeiteinheit berechnen also `( YieldTotal - _YieldTotal ) / (Timestamp - _Timestamp)`.

Die Werte bzw. das Delta zu einem bestimmten Zeitpunkt nullen, könnte man noch einfacher durch eine virtuelle "Null"-Stelltaste in AhoyDTU verwirklichen. Dazu muss man sich nur den aktuellen YieldTotal zum Zeitpunkt X (z.B. den 1.1.2023) merken und zieht diesen künftig immer von allen YieldTotal Werten ab: Zack, fertig :D Man muss nur den Stand zum 1.1.2023 auch irgendwo notiert haben um ihn dann zurückzusetzen / -rechnen. Wohl dem der das per MQTT irgendwo gespeichert hat. Im Nachhinein ohne diese Daten / Information geht das natürlich nur noch schwierig.

jimknopf63, 2022-12-23:
> Gerade Nummer 4. sollte eigentlich ganz interessant sein und m.E. durchaus möglich sein: Der Unterschied `deltaYieldTotal` zum vorherigen Wert (Präfix `_`) ist ja einfach `YieldTotal - _YieldTotal`Eventuell könnte man auch gleich den aktuellen Yield pro Zeiteinheit berechnen `( YieldTotal - _YieldTotal ) / (Timestamp - _Timestamp)` > > Die Werte oder das Delta zu einem bestimmten Zeitpunkt null, könnte noch einfacher durch einen virtuellen "Null"-Stelltaste in AhoyDTU zu verwirklichen. Dazu muss man sich nur den aktuellen YieldTotal zum Zeitpunkt X (z.B. den 1.1.2023) merken und zieht diesen künftig immer von allen YieldTotal Werten ab: Zack, fertig :D Man muss nur den Stand zum 1.1.2023 auch irgendwo notiert haben, um ihn dann zurückzusetzen / -rechnen. Wohl dem, was das pro MQTT irgendwo gespeichert hat. Im Nachhinein ohne diese Daten / Information geht das natürlich nur noch schwierig. 

Ich logge die MQTT YieldTotal Daten im IOBroker mit dem SourceAnalytics Adapter. Da kannst Du den Wert für heute, gestern, Woche, Vorwoche,  Monat, Vormonat. Quartal, vorheriges Quartal,halbjahr, vorheriges Halbjahr, jahr, vorheriges Jahr nach Bedarf auswählen für die Statistik

lumapu, 2022-12-23:
Ich finde den Vorschlag von @jimknopf63 sehr gut. Denn wenn wir anfangen `Yield_Total` zurückzusetzen bzw. zu merken, dann kommen noch weitere Anfragen, ob man da nicht noch dies und das machen kann. 
Ich fände weiterhin sehr gut Ahoy als Bridge zwischen dem 2.4MHz Hoymiles Protokoll und MQTT / JSON zu sehen. 
@Karlo verwendest du eine Smart-Home System?

Karlo49, 2022-12-31:
@1-am-r00t Danke, mit dem Befehl bekomme ich auch meine Webapp kleiner. Habe lange danach gesucht. Habe auf 80% gestellt. Nun ist aber die Beschriftung unter den Zahlenwerten kaum noch lesbar.
@lumapu Habe ein Raspberrymatic als Heizungssteuerung. Da kommt aber nichts anderes drauf.
Ich halte einen Wert  "YieldYear" für sinnvoll, aber das liegt natürlich am Entwickler, ob er sowas umsetzen will. Werde mich mit "MQTT Mosquitto Telegraf influxDB Grafena" testweise auf einen alten Laptop beschäftigen. Dafür benötige ich dann  das Endprodukt einen Raspberry Pi mit SSD (momentan nicht zu bekommen oder sehr teuer).
Wünsche allen einen guten Rutsch ins neue Jahr.

stefan123t, 2023-01-12:
Feature Request: YieldTotal Correction siehe #570

closed 2023-01-12 by stefan123t