lumapu/ahoy #513: Passwort nur im Einstellungsbereich, die Startseite und die Liveseite ohne Passworteingabe

Domain-Markus, 2022-12-20:
## Hardware
  * [X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: __AZ-Delivery __
Retailer URL: ______

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [X] circuit board
* [ ] external antenna

### Power Stabilization:
* [X]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   **D2** |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  **D1** |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [X] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.52
Github Hash: ___67b9170____

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [X] Platform IO

### Desktop OS:
* [ ] Linux
* [X] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [X] Setup settings (use our templates ... to be added)

Domain-Markus, 2022-12-20:
Meine erste AhoyDTU habe ich jetzt bei meinen Eltern untergebracht. Der alte Mann will jetzt natürlich jetzt regelmäßig schauen was seine WR produzieren.

Gibt es die Möglichkeit, die Startseite und die Liveseite ohne PW freizugeben, somit kann er und ich sicher gehen das er nichts aus versehen verstellt!?

Gerri1, 2022-12-21:
Verpasse doch dem ESP einfach ein Display (SSD1306) 😁

Domain-Markus, 2022-12-21:
> Gerri1, das hilft mir nicht weiter. Die Ahoy ist in einem Aufputzverteiler untergebracht in der Nähe der drei WR. 
Erschwerend kommt seine Gehbehinderung dazu, so das ich auf die WLan Lösung hoffe.

Aber so wie ich das hier mitbekommen habe, ist dass nur noch eine Frage der Zeit :-), so schnell wie hier die Updates kommen, kann man die gar nicht flashen.

lumapu, 2022-12-21:
> Aber so wie ich das hier mitbekommen habe, ist dass nur noch eine Frage der Zeit :-), so schnell wie hier die Updates kommen, kann man die gar nicht flashen. 

Danke für das Lob, habe tatsächlich schon eine Idee dafür, aber noch nicht umgesetzt.

lumapu, 2022-12-21:
> Ich würde die Ahoy-DTU so wie sie jetzt ist, komplett PW geschützt, gerne weiter haben wollen.

zur Idee: Ich stelle mir eine Auflistung der Seiten in den Einstellungen vor mit Häkchen zur Auswahl was geschützt sein soll.

Domain-Markus, 2022-12-21:
> zur Idee: Ich stelle mir eine Auflistung der Seiten in den Einstellungen vor mit Häkchen zur Auswahl was geschützt sein soll.

Das hört sich gut an, wobei die einzigste Seite die man vielleicht wirklich ohne PW braucht, ist doch die Live-Seite alle anderen sind ja er für die, die was einstellen oder prüfen wollen die sollten dann auch die Admin-Rechte haben.

Aber ich lasse mich gern überraschen.

Karlo49, 2022-12-31:
Ich  verwende AhoyDTU ohne Password. Habe ein Update von 0.5.59 auf 0.5.66 gemacht.
Im Setup habe ich den Reiter "Protection". Alle Rubriken außer Live haben einen Haken.
Wenn ich die Haken entferne Und Save drücke sind die Haken wieder da.
Wenn in der alten Software kein Password vergeben ist sollten nach einem Update keine Haken da sein.
Danke für die bisher gute Arbeit. Die 0.5.59 lief bei mir 7 Tage ohne Störung.

Gerri1, 2022-12-31:
... stimmt, alle funktionieren, bis auf Index. Wenn man diesen auch abwählt, sind die anderen bis auf Live wieder ein!

Gerri1, 2023-01-10:
@lumapu  Wurde der Fehler eigentlich schon behoben?
Habe heute die 0.5.70 aufgespielt und es sieht so aus, als wenn das immer noch so ist.
Nimmt man alle raus und geht auf save, dann sind nach dem Reboot alle wieder drin, bis auf Live, man kommt aber dann auch überall drauf!

lumapu, 2023-01-11:
@Gerri1 nein, hier wurde noch nichts gefixt, ich habe den issue wieder aufgemacht - als Gedankenstütze

closed 2023-03-27 by lumapu