lumapu/ahoy #516: MQTT Username

Sinned42, 2022-12-22:
Das mit dem SSD1306 ist schon sehr geil und echt ein Vorteil gegenüber OpenDTU. Leider fehlt mir für meinen MQTT-Broker flespi.io noch ein Byte für den Username (dort werden Tokens mit 64 Byte verwendet). Hier ist es 64 Byte - 1 Byte, also für mich leider fail. Würde mich freuen, wenn es mit den 64 Byte irgendwann klappt, dann kann ich den ESP32 mit OpenDTU abschalten.

lumapu, 2022-12-22:
wird eingebaut - nächstes Dev-Build erlaubt dann 64 Zeichen.

Sinned42, 2022-12-22:
Thanks a lot, just in Time to test before the sun goes down, works perfekt with 64 byte username :)

Sinned42, 2022-12-22:
Not really important, but I just had to figure out, that that username for mqtt is only transmitted, if you put in something for the password too, which is in my case with flespi would not have been necessary. Same for radio, if you don't give latitude and longitude, the NRF didn't work, at least in versions before.

lumapu, 2022-12-22:
thank you for figuring that out! Second one is realy important.

closed 2022-12-30 by lumapu