lumapu/ahoy #517: MQTT -> Mosquitto -> Telegraf -> InfluxDB

Waldo56, 2022-12-22:
Hi, habe Probleme mit der Konfiguration von Telegraf betreffend JSON Objekte der Ahoy. Ich würde ja gerne das Plugin im Telegraf Consumer erstellen. Mir fehlt aber die Basis wie die JSON Strukturen am Input von Telegraf aussehen. Ich sehe zwar Strukturen im MQTTFx, weiss aber nicht wirklich ob diese passen. Hat jemand schon mal das telegraf.conf File entsprechend konfiguriert? Wäre für Hinweise, Tipps, oder gar einen Auszug aus einem telegraf.conf File dankbar.

c1328, 2022-12-23:
Hi! Schau dir mal [https://grafana.com/grafana/dashboards/16850-pv-power-ahoy/](url) an. Da findest du auch etwas zur MQTT/Telegraf Konfiguration.

Waldo56, 2022-12-23:
Danke Dir.

gone-for-coding, 2022-12-23:
Ich habe es wie folgt erfolgreich laufen:
```toml
[[inputs.mqtt_consumer]]
  servers = ["tcp://mosquitto:1883"]
  topics = ["ahoyTopic/#"]
  qos = 0
  connection_timeout = "60s"
  username = "xxx"
  password = "xxx"
  data_format = "value"
  data_type = "float"
  ## Route to pv1 bucket
  [inputs.mqtt_consumer.tags]
	target_bucket = "pv1"
```

Waldo56, 2022-12-23:
Herzlichen Dank.

LeSpocky, 2023-01-14:
Ich habe das etwas anders gelöst und lasse Telegraf da bisschen mehr machen, damit das Audröseln in der Datenbank bzw. im Dashboard einfacher ist:

```
[[inputs.mqtt_consumer]]
  servers = ["tcp://127.0.0.1:1883"]
  topics = [
    "ahoy-dtu/+/+/+",
    "ahoy-dtu/+/+",
  ]
  data_format = "value"
  data_type = "float"

[[inputs.mqtt_consumer.topic_parsing]]
  topic = "ahoy-dtu/+/+/+"
  measurement = "measurement/_/_/_"
  tags = "_/name/ch/field"

[[inputs.mqtt_consumer.topic_parsing]]
  topic = "ahoy-dtu/+/+"
  measurement = "measurement/_/_"
  tags = "_/name/field"

[[processors.pivot]]
  tag_key = "field"
  value_key = "value"
```

Das wirft mir nicht _alle_ Dinge in die DB, die AhoyDTU über MQTT ausgibt, aber die eigentlichen Daten zum Solarertrag, ich kann es nach den Tags *name* und *ch* sinnvoll filtern und hab die einzelnen Topics als fields passend zum Datenmodell von InfluxDB.

c1328, 2023-01-24:
Ich habe den Hinweis von @LeSpocky auf die Parsing Konfiguration mal ins How-To übernommen: https://github.com/c1328/ahoy-iobroker-influx-grafana

Und für den Fall, dass hier jemand mitliest, der wie ich auf ioBroker umgestiegen ist, poste ich hier auch gerne das passende Tutorial: https://github.com/c1328/ahoy-iobroker-influx-grafana

Waldo56, 2023-01-25:
Close

closed 2023-01-25 by Waldo56

giGa977, 2023-04-04:
Hallo zusammen, 

ich bin absoluter neuling mit hilfe einiger turt. habe ich auf meinen Server via Docker grafana, node-red und den mqtt Mosquito instaliert bekommen nur mit dem rest komm ich nicht so gut klar da mein Englisch mieserabel ist. Gibt es irgendwo eine ausführliche anleitung in deutsch ?

greetz giGa

lumapu, 2023-04-04:
@giGa977 Das hier ist ein geschlossener Issue - hier liest keiner mehr mit! Am besten einfach hier in den Issues nach den oben genannten Schlagworten suchen, es gbit verteilt Informationen dazu. Leider haben wir diese noch nicht an zentraler Stelle.