lumapu/ahoy #521: Ahoy verbindet sich zu Tagesbeginn nicht mit WR - trotz Produktion (V0.5.59 / V0.5.60)

jimknopf63, 2022-12-25:
### Platform

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.60

### Github Hash

6bb8a4e

### Build & Flash Method

Platform IO (build & flash)

### Desktop

Linux

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Debug Serial Log output

```shell
mac
C8:C9:A3:57:CF:74
sdk
2.2.2-dev(38a443e)
cpu_freq
80
heap_free
8008
sketch_used
485
version
0.5.60
wifi_rssi
-68
ts_uptime
1728
esp_type
ESP8266
core_version
3.0.2
flash_size
4096
heap_frag
15
max_free_blk
6800
reboot_reason
Software/System restart
Radio
nrf24l01+
is connected
Datarate
250 kbps
Power Level
LOW
RX success
344
RX fail
0
RX no answer
2
RX frames received
2314
TX count
477
Sun
Sunrise
25.12.2022, 08:36:04
Sunset
25.12.2022, 16:32:53
Communication start
25.12.2022, 08:36:04
Communication stop
25.12.2022, 16:32:53
Night Communication
disabled
```

### Error description

Die Ahoy DTU verbindet sich morgens nicht mit dem WR. Man muss erst über die Konsole einen Reboot starten. Wenn der Ahoy dann gestartet ist ist er auch direkt mit dem WR verbunden und gibt Daten aus. Das Problem fing mit Version 0.5.59 an und ist in der 0.5.60 immer noch vorhanden. Ich habe auch schon einen automatischen Reboot bei Sonnenaufgang ausgelöst. bringt aber keinen Erfolg. Das Problem ist reproduzierbar, habe hier 3 Ahoys an verschiedenen Anlagen/Haushalten laufen mit den gleichen Problemen. Die 0.5.57 zeigt diese Verhalten noch nicht.

DukeWolle, 2022-12-25:
Da hat sich leider ein Fehler eingeschlichen. 

@lumapu arbeitet quasi schon dran. Jetzt gönnen wir ihm aber erst mal Ruhe und Besinnung in der Weihnachtszeit.

knickohr, 2022-12-25:
Das „Feature“ (=Bug) funktioniert in beide Richtungen, je nachdem wann die DTU zum letzten Mal rebootet wurde 🤪

Aber wie DukeWolle schon geschrieben hat, lassen wir Lukas mal etwas Ruhe. als Workaround würde ich vorschlagen diese Disable Night Communikation komplett zu ignorieren und die DTU immer laufen zu lassen. Also den Haken da raus machen und einmal frisch rebooten.

commit c3fc01b9 (2022-12-26): fix #521 no reconnect at beginning of day
added immediate (each minute) report of inverter status MQTT #522
added protection mask to select which pages should be protected
closed 2022-12-30 by lumapu