lumapu/ahoy #525: regex - Eingabe eines Bindestrichs beim Inverternamen wird zurückgewiesen

roku133, 2022-12-27:
Aktuell kann mit v0.5.61 beim Inverternamen kein Bindestrich eingegeben werden.
Der Bindestrich im regex-Ausdruck wird als Bereichsangabe zwischen "." und "_" interpretiert.

Gerri1, 2022-12-27:
Auch bei den Eingaben >Modulname, >MQTT - Topic. Es erscheint auch ein Meldetext "Invalid input". 
Das war aber auch schon bei der  v0.5.60 vorhanden!

beegee3, 2022-12-27:
Leerzeichen sind im regex auch nicht vorgesehen (genauso wenig wie Umlaute und sonstige internationale Zeichen).
Wenn es darum geht, bestimmte Zeichen nicht zu erlauben wäre eine regex Negativliste vielleicht besser, ansonsten öffnet sich die Büchse der Pandora.

lumapu, 2022-12-27:
ich würde so wenig wie möglich erlauben, dann haben wir die geringesten Probleme. Mit der aktuellen Dev Version `0.5.62` sind folgende Zeichen erlaubt:

* `A-Z`
* `a-z`
* `0-9`
* `.`
* `-`
* `+`
* `_`

roku133, 2022-12-28:
super, schneller Fix 👍

closed 2022-12-28 by roku133

beegee3, 2022-12-28:
zumindest noch das Leerzeichen wäre schön, wenn nichts dagegen spricht

lumapu, 2022-12-28:
@beegee3 ich dachte genau das ist eines der Zeichen, die bei MQTT nicht erlaubt sind

beegee3, 2022-12-29:
als 'topic' klar. Sorry, hatte es nicht gesagt, mir ging es um die Modul Namen, aber das ist ja in der neusten Version ok.