lumapu/ahoy #528: AZDelivery D1 Mini V3 NodeMCU with ahoy is not stable?

Hansi5544, 2022-12-27:
## Hardware
  * [ ] ESP8266
 
AZDelivery D1 Mini V3 NodeMCU

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
### Antenna:
* [ ] circuit board

### Power Stabilization:
* [ ]  470uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
I will test it with a smaller cap too.

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO0  CE   D3 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO2  IRQ  D4 |

Attention D3 and D4 must be changed.
With the connections of the "Getting Started AhoyDTU.pdf" it doesn't work.  

## Software
* [ ] AhoyDTU
221119_ahoy_0.5.41_esp8266_dec333f.bin this bin does not work.
The software makes a AP but the web interface is not complete.
Many menus are not visible. 

221119_ahoy_0.5.41_esp8266_1m_dec333f.bin only this bin works.
The following states i have seen:
1. WEB interface is not available but the MQTT interface and ping works fine.
2. WEB interface, MQTT interface and ping does not work. I must reboot.
3. If i reboot i must do it for 3-10 times and then the ping shows the connection.
4. In the Ahoy WEB Interface->Live the Q_AC value with the Parameter discription are sometimes visible sometimes not. (is it a value 0.0 format problem?)    

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
Command line:
/home/user/.arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3 -I /home/user/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/upload.py --chip esp8266 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash 0x0 /home/user/Downloads/AZ_Delivery/Ahoy/Ahoy/221119_ahoy_0.5.41_esp8266_1m_dec333f.bin

### Desktop OS:
* [ ] Linux

Gerri1, 2022-12-27:
@Hansi5544 
Hast du den ESP vor dem Flashen mal gelöscht?
Für den ESP8266 verwende ich den Tasmotizer 1.2, funktionierte bis jetzt immer tadellos!
Nach dem Flashen den ESP unbedingt mal Stromlos machen!

Hansi5544, 2022-12-28:
Hallo Gerri1,
danke für deine Antwort und deinen Hinweis.

Ich habe ein zweites System aufgebaut und die Konfig und das Setup nochmal getestet.
Dabei traten die gleichen Probleme auf. Stellenweise konnte ich mich nicht mit dem AP verbinden weder über Handy noch über Laptop. Das Setup-Menü war unvollständig. WEB Oberfläche nicht erreichbar. System muste resetet werden.

Nein ich habe nicht jedes mal das Flash komplett gelöscht, mache ich ab jetzt aber.
Neue Arduino Command Zeile:

/home/user/.arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3 -I /home/user/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/upload.py --chip esp8266 --port /dev/ttyUSB0 --baud 921600 erase_flash --before default_reset --after hard_reset
write_flash 0x0 /home/user/Downloads/AZ_Delivery/Ahoy/Ahoy/221119_ahoy_0.5.41_esp8266_dec333f.bin

Ich habe bei AZDelivery auch nochmal nachgesehen was die für Einstellungen vorschlagen um den ESP8266 mit Arduino zu brennen. (Daraus ist die neue Command Zeile entstanden.)

Damit habe ich das zweite System nochmal komplett neu gebrannt und konfiguriert. Das lief auch ganz schlüssig ab.
Danach habe ich das zweite System mehrmals vom Power getrennt und wieder verbunden um zu sehen ob
eine Verbindung zur WEB Oberfläche zustande kommt.
Bis jetzt war das jedes mal erfolgreich.

Ich habe zur Zeit zwei Systeme am Laufen:
1.  Altes System mit folgendem Bin-File: 221119_ahoy_0.5.41_esp8266_1m_dec333f.bin
     Darauf läuft die WEB Oberfläche (AhoyDTU) und das MQTT-Interface und der NRF24L01-
     Chip mit HM-600 und 2 Panels.
2.  Neues System mit folgendem Bin-File: 221119_ahoy_0.5.41_esp8266_dec333f.bin
     Darauf läuft nur die WEB Oberfläche (AhoyDTU)

Im Moment laufen beide Systeme. (Ich kann aber nicht sagen weshalb???)
Ich werde beide Systeme jetzt mal beobachten.

Gerri1, 2022-12-28:
Die Arduino IDE habe ich auch anfangs mal genommen, hatte damit keinen Erfolg gehabt. 
PlatformIO ist eigentlich schon Nr.1 in solchen Sachen und funktionierte bis jetzt auch sehr gut.
Versuche es doch mal mit dem Tasmotizer 1.2, der ist schehr schnell konfiguriert (COM-Port, File, Haken bei self Resetting) und dan Feuer frei! 😁

Hansi5544, 2022-12-28:
Hallo Gerri1,
ich arbeite im privaten Bereich nur mit Linux bestückten Rechnern.
Da ist die installation solcher Tools wie Tasmotizer oder ähnliches nicht ganz so einfach wie unter Windows.
Da die Arduino IDE für mich wie ein Werkzeugkasten ist, wo ich mir nach belieben
was zusammenbauen  kann und deshalb diese IDE auf meinem Rechner als standard installiert ist, ist dieser weg für mich einfacher.  
Aber es sieht so aus als ob du mit deiner Vermutung recht gehabt hast.
Mein altes System hat heute Abend den WEB Dienst wieder eingestellt nur MQTT und ping gehen noch.
Das neue System läuft noch einwandfrei. Ich werde es weiter beobachten und morgen das alte System neu flashen.
Nochmals Danke für den Tip.

Hansi5544, 2022-12-29:
So jetzt habe ich neue Informationen.
Altes System neu geflasht, gleiches Problem. AP wird angezeigt ->Wird eingerichtet->IP Holen-> keine Verbindung.
Entweder abbruch über Power-Reset oder Reset-Taster.
Das ca. 1 Stunde lang bis zufällig eine Verbindung zustande kommt. Dann wird meistens auf 192.168.4.1 nur Ahoy-DTU Menü angezeigt. Die Untermenüs sind nur erreichbar wenn ich in dem Browser 192.168.4.1/setup usw. eingebe.

Ich habe noch ein drittes System aufgebaut, (mit Windows10 in einer VirtualBox und Tasmotizer 1.2) gelöscht und geflasht. Genau gleiches Verhalten und Problem.

Wenn ich dann nach ca. 4-6 Stunden ein System soweit gebracht habe,dass es läuft,, dann scheint es mit dem 221119_ahoy_0.5.41_esp8266_dec333f.bin File stabil durchzulaufen. Das probier ich aber noch aus.

Eine Sache ist mir noch aufgefallen.
Wenn ich folgendes einstelle, werden die Werte einwandfrei übernommen: 
Static IP (optional)

Leave fields blank for DHCP
The following fields are parsed in this format: 192.168.1.1
IP Address   192.168.1.247
Submask     255.255.255.0
DNS 1         192.168.1.1
DNS 2 
Gateway      192.168.1.1

Stelle ich aber folgende Werte ein:
Static IP (optional)

Leave fields blank for DHCP
The following fields are parsed in this format: 192.168.1.1
IP Address   192.168.1.60
Submask     255.255.255.0
DNS 1         192.168.1.1
DNS 2 
Gateway      192.168.1.1

dann macht die WEB Oberfläche nach einem Save und reboot folgendes daraus:
Static IP (optional)

Leave fields blank for DHCP
The following fields are parsed in this format: 192.168.1.1
IP Address   192.168.1.60
Submask     255.255.255.0
DNS 1         192.168.1.10
DNS 2 
Gateway      192.168.1.10

Und damit ist das Gateway und der DNS nicht erreichbar.

Das der Wert und die Bezeichnung "Q_AC" in der Live Ansicht angezeigt wird und wieder verschwindet hatte ich schon gesagt und das ist auch immer noch so. 

So jetzt brauche ich erstmal eine Pause.

stefan123t, 2022-12-29:
@Hansi5544 ich nutze auch nur Linux und fahre mit dem VSCode + PlatformIO Bundle recht gut. Man muß sich die Pakete / Downloads halt am Besten ohne den Paketmanager installieren aber dann läuft es sehr gut.

Du verwendest die 0.5.41 die ist zwar als stable markiert, aber die Entwickler arbeiten aktuell an einem Release fürs Neue Jahr :D D.h. wir sind im development03 branch bereits bei 0.5.64 es gibt darin u.a. auch einige Verbesserungen zu Static IP/DNS Angaben als auch zum MQTT PubSubClient. Also entweder mit git das aktuelle Repo clonen und per PlatformIO selber bauen und flashen oder evtl. ein Development Release aus den GitHub actions herunterladen und das mit den ESP Web Tools flashen. Vorsicht für die ESP Web Tools muß man den Google Chrome Browser stoppen (im Tasktray!) und mit einem Parameter starten:
`google-chrome --allow-file-access-from-files` sonst darf Dein Google Chrome nicht auf die Serielle Schnittstelle und die Binaries zum Flashen zugreifen.

Hansi5544, 2022-12-30:
Hallo Stefan123t,
im Moment scheinen alle drei Systeme zu laufen.
Mit dem Konfigurieren war etwas schwerfällig, aber die Version 221119_ahoy_0.5.41_esp8266_dec333f.bin scheint zu arbeiten.
MQTT lief eigentlich immer am besten.
Ich lasse die System jetzt erstmal in Ruhe und halte alles unter Beobachtung.
Wenn eine neue Version stable herauskommt probiere ich die mal aus.

Hansi5544, 2023-01-22:
Hallo stefan123t,
das ist kein Wemos Board, sondern von AZDelivery ein D1 Mini V3 NodeMCU siehe Link:
[https://www.az-delivery.de/en/products/d1-mini-v3](url)
Ich hatte das auch zuerst mit dem Wemos Board verwechselt, aber es ist ein ganz anderes Board was nur ähnlich aussieht.
Ich könnte mir vorstellen, dass das dem Ein oder Anderem auch so ergangen ist.

Auf das Board habe ich jetzt wieder den 10uF eingelötet. Das Board lief aber auch ohne Elko ohne Probleme.
Das der 470uF Elko Probleme verursachen könnte, würde ich nicht so sehen und hat sich auch nicht so dargestellt.

Die Fehler mit dem Q_AC Wert und dem IP-Adressen vom DNS und Gateway sind behoben.

Das Update über die Browser Oberfläche funktionierte einwandfrei.

Wieso manchmal der AHOY-DTU AP aufgebaut wird und dann wider verschwindet, kann ich noch nicht nach vollziehen.

Und wenn man vom Teufel spricht. Gerade noch ein paar Sachen ausprobiert und plötzlich läuft MQTT sowie die Browseroberfläche nicht mehr und auch mit ping ist nichts mehr erreichbar. Dann mehrmals Reset machen bis es wieder läuft.

Es ist noch nicht perfekt, aber schon besser als vorher.

closed 2023-01-22 by Hansi5544