lumapu/ahoy #529: Feature Request :  MQTT neuer Topic „comm_disabled“

knickohr, 2022-12-29:
Jajaa, ich weiß, wieder MQTT und kein Einde 🤪

Bei. Node-Red puzzeln ist mir aufgefallen das ich nichts über den Status der Kommunikation zum NRF-Mudul bzw. den Invertern i. Erfahrung bringen kann.

Ich vermisse einen Topic <topicname>/comm_disabled true/false

Damit meine ich nicht das Setzen des Häkchens bei „disable night communication“, das hat ja einen eigenen Topic, sondern den wirklichen Status ob die DTU gerade kommuniziert oder nicht. Der Status sollte gesetzt sein (true) wenn einerseits der obige Fall eintritt (Nachts) oder andererseits wenn man bei den Invertern die Kommunikation manuell deaktiviert. Also quasi immer wenn keine Kommunikation statt findet, oder unterbricht. Oder salopp gesagt, wenn nicht gepollt wird.

DukeWolle, 2022-12-29:
Glaub mir du merkst wenn nichts mehr kommt. Also mir reichen aktuell Status und available..

knickohr, 2022-12-29:
Darum geht’s nicht 😅

Ich würde gerne eine Rückmeldung haben was die DTU treibt oder eben nicht 🤓

DukeWolle, 2022-12-29:
Stable Release ist mE erst mal wichtiger wie schnick schnack. 😀

beegee3, 2022-12-30:
>Ich vermisse einen Topic /comm_disabled true/false

Evtl. noch ein /comm_dis_ts', für den Zeitpunkt der letzten Anpassung. Dazu braucht man nur
in pubMqtt.h
```
        void tickerComm(bool disabled) {
            publish("comm_disabled", ((disabled) ? "true" : "false"), true);
            publish("comm_dis_ts", String(*mUtcTimestamp).c_str(), true);
        }
```
und eine Anpassung in app.cpp
```
void app::tickIVCommunication(void) {
    mIVCommunicationOn = !mConfig->sun.disNightCom; // if sun.disNightCom is false, communication is always on
    if (!mIVCommunicationOn) {  // inverter communication only during the day
        ...
            if (mTimestamp > (mSunset + mConfig->sun.offsetSec)) { // current time is past communication stop, nothing to do. Next update will be done by tickCalcSunrise
                nxtTrig = 0;
        ...
        if (nxtTrig > 0)
            onceAt(std::bind(&app::tickIVCommunication, this), nxtTrig);
    }
    if (mConfig->mqtt.broker[0] > 0) {
        mMqtt.tickerComm(mIVCommunicationOn);
    }
}
```

knickohr, 2022-12-30:
@beegee3 
Ich verstehe C(++) nur zum Teil, das ist klingonisch für mich einfacher. Ist auch der Grund warum ich mich da so schlecht einbringen kann. ansonsten wäre da schon längst ein ePaper Display an der DTU 😉

Flag müßte eigentlich doch schon vorhanden sein, ansonsten wüßte die DTU doch gar nicht ob sie senden soll oder nicht. Muß doch salopp gesagt nur noch ein publish zum Broker gemacht werden.

Qapla‘

beegee3, 2022-12-30:
@knickohr nuqneH
Der Code ist mehr als Anregung für @lumapu gedacht. Der wird zur Zeit nicht nur von mir mit jeder Menge anderer Anfragen genervt 😒 und muss ja nicht alles allein machen 😏.
>Flag müßte eigentlich doch schon vorhanden sein

ngem Sarghmey tlha'laH, to'waQ vIyIvtaH (tlhIngan Soj 'oH -- not bIyaj 😁):

`mIVCommunicationOn` ist die Flag, in `tickIVCommunication` wird sie gesetzt. Im Code oben ist die neue Funktion `tickerComm`, die den publish macht und (auch neu) am Ende von `tickIVCommunication` aufgerufen wird. Dazu noch eine kleine Korrektur, damit der Aufruf auch immer gemacht wird (vorher wurde die Funktion im Fall von 'current time is past communication stop' sofort beendet).

In diesem Sinne
Qapla‘

knickohr, 2023-01-06:
Die Stable ist da 😎 Wie geht’s weiter ?  *drängel* *quängel* 😅

knickohr, 2023-01-07:
Das comm_disabled ist verkehrt herum 😲

Also entweder umbenennen in comm_enabled oder true/false tauschen 😅

knickohr, 2023-01-07:
Danke 😎😘

Jetzt funktionieren alle 5 Stati out of the box 👍

- online (alle Inverter online und produzieren)
- partial (min. ein Inverter produziert nicht)
- missing (min. ein Inverter fehlt unter Tags)
- offline (alle Inverter sind offline)
- communicatin disabled (es ist Nacht 🤪)

🥳

closed 2023-01-07 by knickohr